davidrapan/ha-solarman #269: Setup wizard, can not enter serial number containing letters.

RadioactiveMan56, 2024-11-28:
### Description

I am trying to configure the integration with the wizard, but my inverter serial number contains a letter which is not accepted. I also noted the wizard does not accept leading zeroes is the serial number.

### Reproduction Steps

Try to enter text string with letters and numbers for inverter serial number.

### Log

There is no debug log

### Version

24.11.02

davidrapan, 2024-11-28:
Hi @RadioactiveMan56,

That is correct, loggers with alphanumeric serial numbers are an older version of the hardware which is not (and won't be) supported by pysolarmanv5.

davidrapan, 2024-12-19:
Hey @RadioactiveMan56, can you tell me which device you have exactly?

RadioactiveMan56, 2024-12-19:
Hi David,

It is an S2-WL-ST.

Using this integration now with some success:

https://github.com/wills106/homeassistant-solax-modbus

Eric

> Hey @RadioactiveMan56 <https://github.com/RadioactiveMan56>, can you tell
me which device you have exactly?

davidrapan, 2024-12-19:
So this integration should work too if you force `passthrough = True` mode. 😉 But you have to change it directly in the api.py file.

RadioactiveMan56, 2024-12-19:
Thanks, I will have a look at it in the new year.

davidrapan, 2025-01-20:
When comes to Ethernet Loggers latest version of the code contains this change:
```
self._passthrough = serial <= 0
```
which means that setting serial number to 0 will activate passthrough mode. 😉

closed 2025-03-12 by davidrapan

easyest, 2025-07-24:
S2-WL-ST is the latest wi-fi datalogger from Solis. It is for sure not "older version of the hardware" and therefor should be supported.