davidrapan/ha-solarman #271: Power losses

WernerKr, 2024-11-29:
### Is your feature request related to a problem?

Power losses! Why is a negative value not taken into account?
" if "uint" in definition and value < 0:
value = 0
"
My Deye 2mppt microinverter (Deye Sun-M80G3) therefore never showed power losses, although power loss does occur.

### Describe the solution you'd like

 I therefore changed it to "value = abs(value)".
This means that this microinverter also shows power losses

### Describe alternatives you've considered

In my opinion there is no alternative

davidrapan, 2024-12-01:
Hi @WernerKr! Thank you for your input. Do you really think that power losses calculated essentially in reverse will produce valid result (output > input)? I'm guessing the main issue is probably with low update interval of this type of microinverter, right?

WernerKr, 2024-12-01:
As I wrote, without this change, power losses were never recorded on the DEYE SUN-M80-G3. Only after my change on November 28th were power losses recorded. For comparison, the graph of the DEYE SUN-M160-G4 when starting in the morning (example November 25th) - these values ​​are also missing), the inverter needs more power than is supplied by the panels.
[image]
[image]

davidrapan, 2024-12-01:
I do understand your POV but the issue is proly w/ source of input power value (precision or out of sync w/ source of output power value because it's calculated in HA from input voltages and currents) rather than w/ "incorrect" calculation of said losses. So  even when abs produces **value > 0** it's not what you think it is and **consumption of the microinverter has to manifest** either w/ **output power < calculated input power** or w/ negative output power value which would indicate that solar panels don't produce enough energy to power the microinverter.

WernerKr, 2024-12-02:
> >it's not what you think it is and consumption of the microinverter has to manifest 
> > either w/ output power < calculated input power or w/ negative output power value 

I am aware of that. Even if the value is not correct for my deye_2mppt micro inverter, 
I personally think it is better to display a value than no loss value at all. 
Every micro inverter has its own consumption. 
I record the actual value with an SDM230 independently of the data from the micro inverter.

closed 2024-12-02 by WernerKr