davidrapan/ha-solarman #272: Error displayed when changing Time Entity

Pyinthesky99, 2024-11-30:
### Description

If you change the value of any of the hour or minute entities, an error message is displayed.
The Entity value does change as requested.

### Reproduction Steps

Change any hour or Minute timer on Dashboard and an error is displayed but the timer does change to the correct setting in the Inverter.

[image]

 It seems to work without error if I use entity: number.solis_timed_<charge_start/end>etc where the dashboard displays the times in hours and minutes in separate sliders but not if I use entity: time.solis_timed_charge_<start/end>etc which is displayed on a Dashboard as above.

Hi David, remember us :-)

We (couldn't have done this without @stevebuk1) have just about assimilated this repo of Solarman into Francis Boundy's PV_OPT (https://github.com/fboundy/pv_opt) 

I raised this as an issue 310 on that repo and Steve responded with the section of the home assistant log below which apparently indicates is not within pv_opt...
Any ideas?

 

### Log

```
Exception: [4065705146] Unexpected response: Invalid length! (Length: 1, Expected: 2)
2024-11-27 01:50:20.564 ERROR (MainThread) [aiohttp.server] Error handling request
Traceback (most recent call last):
  File "/usr/local/lib/python3.12/site-packages/aiohttp/web_protocol.py", line 462, in _handle_request
    resp = await request_handler(request)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/aiohttp/web_app.py", line 537, in _handle
    resp = await handler(request)
           ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/aiohttp/web_middlewares.py", line 114, in impl
    return await handler(request)
           ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/http/security_filter.py", line 92, in security_filter_middleware
    return await handler(request)
           ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/http/forwarded.py", line 83, in forwarded_middleware
    return await handler(request)
           ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/http/request_context.py", line 26, in request_context_middleware
    return await handler(request)
           ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/http/ban.py", line 85, in ban_middleware
    return await handler(request)
           ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/http/auth.py", line 242, in auth_middleware
    return await handler(request)
           ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/http/headers.py", line 32, in headers_middleware
    response = await handler(request)
               ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/http.py", line 73, in handle
    result = await handler(request, **request.match_info)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/api/__init__.py", line 430, in post
    response = await shield(
               ^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/core.py", line 2761, in async_call
    response_data = await coro
                    ^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/core.py", line 2804, in _execute_service
    return await target(service_call)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/service.py", line 996, in entity_service_call
    single_response = await _handle_entity_call(
                      ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/service.py", line 1068, in _handle_entity_call
    result = await task
             ^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/time/__init__.py", line 35, in _async_set_value
    return await entity.async_set_value(service_call.data[ATTR_TIME])
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/solarman/time.py", line 58, in async_set_value
    await self.write(get_t_as_list_int(value, self._multiple_registers), value.strftime(TIME_FORMAT))
  File "/config/custom_components/solarman/entity.py", line 140, in write
    if await self.coordinator.inverter.call(self.code, self.register, value, ACTION_ATTEMPTS_MAX) > 0:
       ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/solarman/api.py", line 263, in call
    response = await self.safe_read_write(code, start, arg)
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/solarman/api.py", line 167, in safe_read_write
    raise Exception(f"[{self.serial}] Unexpected response: Invalid length! (Length: {length}, Expected: {expected})")```

The full log submitted was...
[home-assistant_2024-11-28T14-51-35.385Z.log (1).txt](https://github.com/user-attachments/files/17967259/home-assistant_2024-11-28T14-51-35.385Z.log.1.txt)

```

### Version

V24.10.16

Pyinthesky99, 2024-11-30:
[home-assistant_2024-11-28T14-51-35.385Z.log (1).txt](https://github.com/user-attachments/files/17967283/home-assistant_2024-11-28T14-51-35.385Z.log.1.txt)

davidrapan, 2024-12-01:
Hi @Pyinthesky99, yes there was a bug and successful response from the device was treated as unsuccessful.

It should be fixed in the latest version.

Pyinthesky99, 2024-12-01:
Hi David
Thanks for the information.
I may have done something stupid which would not be a first!
I have copied the latest version of the solarman directory into my custom_components, re-started HA and I now have a duplicate set of entities all appended with _2

[image]

Is this a feature?

Regards
Alan

davidrapan, 2024-12-01:
You downloaded main?... not 24.11.02? I mean if yes the damage is done (don't worry, nothing serious) and you will fix it by removing and re-adding device back (hope you don't have any tweaking on top of used profile, hah)!

Pyinthesky99, 2024-12-01:
Yes, case proved, I did do something stupid. Have removed from 'Settings' and 'HACS' then downloaded properly, intead of just copying solarman directory.

>  (hope you don't have any tweaking on top of used profile, hah)!
> 

Now doing some tweaking (ugh) before I can test!

Thanks

Pyinthesky99, 2024-12-01:
Tweaking accomplished, my extra entities for time were....
time.solis_timed_charge_start_2
yours are....
time.solis_timed_charge_2_start
Whoops!
Tested and working fine now with no error message. (@SteveBuk1)
Thanks again for your help
Kind regards
Alan

closed 2024-12-01 by Pyinthesky99

davidrapan, 2024-12-01:
Yeah It made more sense to me that way and I also think that similar entities already existed somewhere else so I'm trying to stick w/ some regularity across profiles.

When you are done downgrading could you maybe check for me format of `unique_id` and `original_name` in 24.11.02? You can see it in `/homeassistant/.storage/core.entity_registry` when you search for `"platform":"solarman"`.

Pyinthesky99, 2024-12-02:
Hi David
This is a copy of core.entity_registry
[core.entity_registry.txt](https://github.com/user-attachments/files/17978690/core.entity_registry.txt)