davidrapan/ha-solarman #274: Everything unavailable

realkibou, 2024-12-02:
### Description

The setup with my Deye Inverter SUN-5K-SG03LP1-EU works perfectly. All sensor data etc gets pulled. But after maybe 5 minutes everything switches to 'unavailable'. 

### Reproduction Steps

I can remove the device, add it again, then I get all the data, but again after 5 minutes or so, it is all 'unavailable' again.

### Log

No log

### Version

2024.11.3

sofkaski, 2024-12-02:
I updated my system to the the latest (b5b9ada595) and see also problems:
> 2024-12-02 17:39:44.335 ERROR (MainThread) [homeassistant.components.switch] Error adding entity switch.inverter_2 for domain switch with platform solarman
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/entity_platform.py", line 599, in _async_add_entities
    await coro
  File "/usr/src/homeassistant/homeassistant/helpers/entity_platform.py", line 918, in _async_add_entity
    await entity.add_to_platform_finish()
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 1368, in add_to_platform_finish
    self.async_write_ha_state()
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 1007, in async_write_ha_state
    self._async_write_ha_state()
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 1132, in _async_write_ha_state
    self.__async_calculate_state()
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 1069, in __async_calculate_state
    state = self._stringify_state(available)
            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 1013, in _stringify_state
    if (state := self.state) is None:
                 ^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 1672, in state
    if (is_on := self.is_on) is None:
                 ^^^^^^^^^^
  File "/config/custom_components/solarman/switch.py", line 70, in is_on
    return self._native_value() != self._value_off
           ^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/solarman/switch.py", line 65, in _native_value
    return self._attr_native_value
           ^^^^^^^^^^^^^^^^^^^^^^^
AttributeError: 'SolarmanSwitchEntity' object has no attribute '_attr_native_value'. Did you mean: '_to_native_value'?
2024-12-02 17:39:44.359 ERROR (MainThread) [homeassistant.components.switch] Error adding entity switch.inverter_battery_wake_up for domain switch with platform solarman
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/entity_platform.py", line 599, in _async_add_entities
    await coro
  File "/usr/src/homeassistant/homeassistant/helpers/entity_platform.py", line 918, in _async_add_entity
    await entity.add_to_platform_finish()
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 1368, in add_to_platform_finish
    self.async_write_ha_state()
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 1007, in async_write_ha_state
    self._async_write_ha_state()
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 1132, in _async_write_ha_state
    self.__async_calculate_state()
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 1069, in __async_calculate_state
    state = self._stringify_state(available)
            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 1013, in _stringify_state
    if (state := self.state) is None:
                 ^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 1672, in state
    if (is_on := self.is_on) is None:
                 ^^^^^^^^^^
 File "/config/custom_components/solarman/switch.py", line 70, in is_on
    return self._native_value() != self._value_off
           ^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/solarman/switch.py", line 65, in _native_value
    return self._attr_native_value
           ^^^^^^^^^^^^^^^^^^^^^^^
AttributeError: 'SolarmanSwitchEntity' object has no attribute '_attr_native_value'. Did you mean: '_to_native_value'?

sofkaski, 2024-12-02:
I reverted back to d30089e which was the previous version that I was successfully running. Quick glance shows that now the integration created new entities (with _2 postfix). Need to investigate further.

davidrapan, 2024-12-02:
Hi @sofkaski, would you be willing to do the update to the latest, enabled debug logging, and restart HA w/ debug logging enabled and provide the log file? It would be very helpful!

sofkaski, 2024-12-02:
I'll do that

sofkaski, 2024-12-02:
> > I'll do that.
> 
> Here is the log for the first minute after restart. I hope this reveals already useful information. [home-assistant_solarman_2024-12-02T17-01-48.210Z.log](https://github.com/user-attachments/files/17981459/home-assistant_solarman_2024-12-02T17-01-48.210Z.log)

@davidrapan : Note that this is now the second time the new version is used since I reverted back after the first attempt. And the revert did not go well either.

davidrapan, 2024-12-02:
You should have remove and re-add back device so it clears out those duplicates.

davidrapan, 2024-12-02:
@realkibou, please I need you to enable debug logging then remove and re-add your device back. Wait until your sensors are 'unavailable', stop the debugging and attach the file here.

Thank you.

realkibou, 2024-12-02:
[error_log-1.txt](https://github.com/user-attachments/files/17985114/error_log-1.txt)
Here you go

davidrapan, 2024-12-03:
Whoa... 🤯

So you are saying that re-adding the device helps? And what about reload under that 3 dots menu? Could you also record the log w/ device remove and re-add back after that connections loss?

realkibou, 2024-12-03:
Reloading it also helps. I have a suspicion that is becomes unavailable once I log into my inverter with the Deye app. But also that is inconsistent.

davidrapan, 2024-12-03:
What's the logger's firmware?

realkibou, 2024-12-03:
Inverter: V0.2.0.1 3385-1515 0000-036B
Logger: LSW3_15_FFFF_1.0.9E V1.1.00.10

sofkaski, 2024-12-03:
> Reloading it also helps. I have a suspicion that is becomes unavailable once I log into my inverter with the Deye app. But also that is inconsistent.

I have noticed that I get about once or twice a day no-socket-available  errors and then timeout in connection, but  solarman-ha recovers from that in the next update round nicely. My setup is such that the inverter is hybrid Deye SUN-8K-SG04P3-EU with a logger with firmware version LSW3_15_FFFF_1.0.9E. I have connected it also with Deye cloud. (which I would like to get rid of finally.)

My understading is that Deye app does not talk directly to the logger, but the logger pushes data to the Deye cloud and app uses data from there. (Remote control operations might though talk with the logger directly :thinking: ).
My perception is that at some point the periods of Deye and ha-solarman co-inside so that the logger runs out of resources and fails the connection .

sofkaski, 2024-12-03:
I now updated ha-solarman to 39732bd and everything went fine. Migrations were run. 
[home-assistant_solarman_2024-12-03T08-35-49.621Z.log](https://github.com/user-attachments/files/17989872/home-assistant_solarman_2024-12-03T08-35-49.621Z.log)

davidrapan, 2024-12-03:
Yes that's what's proly happening but as you said it should recover. Why it does not work in @realkibou's case is really weird...

Anyway, I'm glad it's good for you, @sofkaski. Thanks for the report, really appreciate it. 😉

realkibou, 2024-12-03:
I updated to the newest version of solarman v24.12.02 and also my Deye inverter got an update to HMI: 0000-C372. The bug however persists.

davidrapan, 2024-12-03:
Hello @githubDante! Me again! Any ideas?

githubDante, 2024-12-03:
Hi @davidrapan , 

Something remains hidden here:
```
2024-12-03 06:43:39.006 INFO (MainThread) [custom_components.solarman.api] [2786122897] Connecting to 192.168.101.3:8899
2024-12-03 06:44:09.011 DEBUG (MainThread) [custom_components.solarman.api] [2786122897] Fetching failed. [Previous State: Disconnected (-1)]
2024-12-03 06:44:09.011 INFO (MainThread) [custom_components.solarman.api] [2786122897] Disconnecting from 192.168.101.3:8899
```
but it looks like the logger is unavailable. Considering the timestamps probably a _NoSocketAvailable_ error is raised in the call to `connect()`. This however is just a wild guess, as the `connect` call is nested deep under the [`safe_read_write` method.](
https://github.com/davidrapan/ha-solarman/blob/39732bd2fd818cf6a48f9a734ec932f70bf00adf/custom_components/solarman/api.py#L172) A hint about the missing/unreachable logger is also the fact that this starts happening soon after the first read timeout.

@realkibou can you try to reach the logger IP address (ping/telnet to port 8899) in such moment, preferrably from the device where ha-solarman is running.

davidrapan, 2024-12-03:
Yes, No Sockat Available does occur once per ~ 5m or so.

This hang visible in the log is most likely in Asyncio’s open_connection as I’ve seen it happen but never in this unrecoverable form. There there is 30s timeout triggered.

githubDante, 2024-12-03:
With the latest version 24.12.02 & Python 3.12 the _NoSocketAvailable_ exception is definitely propagated to the log. But ... I managed to achieve the same result 

```
2024-12-03 18:34:29.433 DEBUG (MainThread) [custom_components.solarman.api] [2654321987] Querying 03 ~ 0000 - 0124 | 0x0000 - 0x007C # 125 ...
2024-12-03 18:34:29.433 INFO (MainThread) [custom_components.solarman.api] [2654321987] Connecting to 192.168.168.5:8890
2024-12-03 18:34:59.436 DEBUG (MainThread) [custom_components.solarman.api] [2654321987] Fetching failed. [Previous State: Disconnected (-1)]
2024-12-03 18:34:59.436 INFO (MainThread) [custom_components.solarman.api] [2654321987] Disconnecting from 192.168.168.5:8890
2024-12-03 18:34:59.436 DEBUG (MainThread) [custom_components.solarman.coordinator] Finished fetching  data in 30.003 seconds (success: False)

```
It's caused by a firewall dropping the packets to the destination address:

```
iptables -I OUTPUT -d 192.168.168.5 -j DROP
```

in the **ha** container/machine/device is enough.

realkibou, 2024-12-04:
> I updated to the newest version of Solarman v24.12.02 and also my Deye inverter got an update to HMI: 0000-C372. The bug however persists. But one click on reload and it works for a bit

Still the same bug :(

davidrapan, 2024-12-04:
I mean sure if you drop the packets I'd say it should timeout... but how could that be happen in his case? Some HA's quirks?

@realkibou, could you run a fresh HA installation aside and try it there?

githubDante, 2024-12-04:
> but how could that happen in his case? Some HA's quirks?

It could be a router, switch or anything that stands between the datalogger and the client.

davidrapan, 2024-12-04:
Yeah.. that's why I wanted to know if there is any chance that @realkibou heard anything about things like tcpdump and wireshark. 😆

realkibou, 2024-12-04:
The inverter is pingable. I just now made an automation that HA just reload the integration. Dirty, but it works.

realkibou, 2024-12-04:
[image]

davidrapan, 2024-12-05:
If you could run tcpdump in HA or on the device which hosts HA that would reveal way more.

realkibou, 2024-12-06:
At 11:47:35 the inverter became unavailable. 

Here you go:

`
11:46:24.662597 ARP, Request who-has 192.168.101.11 tell 0.0.0.0, length 46
11:46:24.663585 IP6 :: > ff02::1:ffa6:7c38: ICMP6, neighbor solicitation, who has fe80::82ce:8d79:faa6:7c38, length 24
11:46:24.665859 IP6 fe80::82ce:8d79:faa6:7c38 > ip6-allrouters: ICMP6, router solicitation, length 8
11:46:24.667326 IP6 fe80::1 > ip6-allnodes: ICMP6, router advertisement, length 56
11:46:24.714852 ARP, Request who-has 192.168.101.1 tell 192.168.101.11, length 46
11:46:24.715840 IP6 fe80::82ce:8d79:faa6:7c38 > ff02::1:ff00:1: ICMP6, neighbor solicitation, who has fe80::1, length 32
11:46:24.800401 IP6 fe80::82ce:8d79:faa6:7c38.49388 > ff02::1:3.5355: UDP, length 24
11:46:24.884551 IP 192.168.101.6.59731 > 255.255.255.255.6667: UDP, length 172
11:46:25.054898 IP6 fe80::82ce:8d79:faa6:7c38 > ff02::1:ff00:1: ICMP6, neighbor solicitation, who has fe80::1, length 32
11:46:25.229043 IP 192.168.101.11.5353 > mdns.mcast.net.5353: 0 ANY (QM)? Boomba.local. (30)
11:46:25.229371 IP6 fe80::82ce:8d79:faa6:7c38.5353 > ff02::fb.5353: 0 ANY (QM)? Boomba.local. (30)
11:46:25.229501 IP 192.168.101.11.5353 > mdns.mcast.net.5353: 0*- [0q] 2/0/0 AAAA fe80::82ce:8d79:faa6:7c38, A 192.168.101.11 (68)
11:46:25.229741 IP6 fe80::82ce:8d79:faa6:7c38.5353 > ff02::fb.5353: 0*- [0q] 2/0/0 AAAA fe80::82ce:8d79:faa6:7c38, A 192.168.101.11 (68)
11:46:25.230051 IP6 fe80::82ce:8d79:faa6:7c38.56813 > ff02::1:3.5355: UDP, length 24
11:46:25.230290 IP 192.168.101.11.56813 > 224.0.0.252.5355: UDP, length 24
11:46:25.269054 IP 192.168.101.11.137 > 192.168.101.255.137: UDP, length 68
11:46:25.269198 IP 192.168.101.11.137 > 192.168.101.255.137: UDP, length 68
11:46:25.269320 IP 192.168.101.11.137 > 192.168.101.255.137: UDP, length 68
11:46:25.294847 ARP, Request who-has 192.168.101.4 tell 192.168.101.1, length 46
11:46:25.321277 IP 192.168.101.14.60726 > 192.168.101.3.8899: Flags [P.], seq 5437:5473, ack 10013, win 64024, length 36
11:46:25.340715 ARP, Request who-has 192.168.101.14 tell 192.168.101.1, length 46
11:46:25.340794 ARP, Reply 192.168.101.14 is-at 7c:d3:0a:1d:18:37 (oui Unknown), length 28
11:46:25.420015 IP 192.168.101.3.8899 > 192.168.101.14.60726: Flags [P.], seq 10013:10053, ack 5473, win 3336, length 40
11:46:25.420229 IP 192.168.101.14.60726 > 192.168.101.3.8899: Flags [.], ack 10053, win 64024, length 0
11:46:25.422501 IP 192.168.101.14.60726 > 192.168.101.3.8899: Flags [P.], seq 5473:5509, ack 10053, win 64024, length 36
11:46:25.519873 IP 192.168.101.3.8899 > 192.168.101.14.60726: Flags [P.], seq 10053:10089, ack 5509, win 3300, length 36
11:46:25.535812 IP6 fe80::82ce:8d79:faa6:7c38.546 > ff02::1:2.547: dhcp6 inf-req
11:46:25.560269 IP 192.168.101.14.60726 > 192.168.101.3.8899: Flags [.], ack 10089, win 64024, length 0
11:46:25.569412 ARP, Request who-has 192.168.101.3 tell 192.168.101.1, length 46
11:46:25.569412 ARP, Request who-has 192.168.101.12 tell 192.168.101.1, length 46
11:46:25.591766 ARP, Request who-has 192.168.101.10 tell 192.168.101.1, length 46
11:46:25.666987 ARP, Request who-has 192.168.101.11 tell 0.0.0.0, length 46
11:46:25.667345 IP6 fe80::82ce:8d79:faa6:7c38 > ip6-allnodes: ICMP6, neighbor advertisement, tgt is fe80::82ce:8d79:faa6:7c38, length 32
11:46:25.681781 IP 192.168.101.11.5353 > mdns.mcast.net.5353: 0 ANY (QM)? Boomba.local. (30)
11:46:25.682075 IP6 fe80::82ce:8d79:faa6:7c38.5353 > ff02::fb.5353: 0 ANY (QM)? Boomba.local. (30)
11:46:25.682259 IP6 fe80::82ce:8d79:faa6:7c38.5353 > ff02::fb.5353: 0*- [0q] 2/0/0 AAAA fe80::82ce:8d79:faa6:7c38, A 192.168.101.11 (68)
11:46:25.682491 IP 192.168.101.11.5353 > mdns.mcast.net.5353: 0*- [0q] 2/0/0 AAAA fe80::82ce:8d79:faa6:7c38, A 192.168.101.11 (68)
11:46:25.682773 IP6 fe80::82ce:8d79:faa6:7c38.50944 > ff02::1:3.5355: UDP, length 24
11:46:25.683052 IP 192.168.101.11.50944 > 224.0.0.252.5355: UDP, length 24
11:46:25.704251 IP 192.168.101.8.5353 > mdns.mcast.net.5353: 0 [5q] PTR (QM)? _ftp._tcp.local. PTR (QM)? _smb._tcp.local. PTR (QM)? _nfs._tcp.local. PTR (QM)? _sftp-ssh._tcp.local. PTR (QM)? _rtsp._tcp.local. (123)
11:46:26.001007 IP 192.168.101.11.137 > 192.168.101.255.137: UDP, length 68
11:46:26.001155 IP 192.168.101.11.137 > 192.168.101.255.137: UDP, length 68
11:46:26.001272 IP 192.168.101.11.137 > 192.168.101.255.137: UDP, length 68
11:46:26.368767 IP 192.168.101.11.27036 > 192.168.101.255.27036: UDP, length 41
11:46:26.471331 IP 192.168.101.7.49156 > 192.168.101.14.38900: UDP, length 141
11:46:26.668154 ARP, Request who-has 192.168.101.11 tell 0.0.0.0, length 46
11:46:26.744844 IP 192.168.101.11.137 > 192.168.101.255.137: UDP, length 68
11:46:26.744990 IP 192.168.101.11.137 > 192.168.101.255.137: UDP, length 68
11:46:26.745077 IP 192.168.101.11.137 > 192.168.101.255.137: UDP, length 68
11:46:27.009671 IP 192.168.101.14.44049 > 192.168.101.16.38899: UDP, length 105
11:46:27.009960 IP 192.168.101.14.47624 > 192.168.101.7.38899: UDP, length 105
11:46:27.285157 IP 192.168.101.7.38899 > 192.168.101.14.47624: UDP, length 84
11:46:27.289747 IP 192.168.101.16.38899 > 192.168.101.14.44049: UDP, length 84
11:46:27.290943 IP 192.168.101.16.51326 > 192.168.101.14.38900: UDP, length 178
11:46:27.290943 IP 192.168.101.7.49156 > 192.168.101.14.38900: UDP, length 115
11:46:27.374322 IP 192.168.101.11.27036 > 192.168.101.255.27036: UDP, length 139
11:46:27.513932 IP 192.168.101.11.5353 > mdns.mcast.net.5353: 0 ANY (QM)? Boomba.local. (30)
[... cut ...]
11:47:47.677063 IP 192.168.101.14.40320 > dns2.mtmbali.net.id.53: 58125+ [1au] PTR? 0.c.1.2.1.9.a.b.5.c.a.7.6.8.0.1.0.0.0.0.0.0.0.0.0.0.0.0.0.8.e.f.ip6.arpa. (101)
11:47:47.677125 IP 192.168.101.14.60119 > dns1.mtmbali.net.id.53: 4010+ NS? . (17)
11:47:47.687554 IP dns1.mtmbali.net.id.53 > 192.168.101.14.60119: 4010 13/0/15 NS h.root-servers.net., NS k.root-servers.net., NS d.root-servers.net., NS j.root-servers.net., NS e.root-servers.net., NS g.root-servers.net., NS i.root-servers.net., NS f.root-servers.net., NS c.root-servers.net., NS l.root-servers.net., NS m.root-servers.net., NS b.root-servers.net., NS a.root-servers.net. (492)

`

githubDante, 2024-12-06:
192.168.101.3 is definitely not responding to ARP requests, making it unavailable.

davidrapan, 2024-12-06:
Yes it's clear it's not integration's fault. Something on your network is here to blame. Almost reminds me one issue I was experiencing w/ OpenWRT on AX3600, ARP table was not flushed when client reappeared on different network path and packets were still send to their old location. Which is really not the case here as I bet your inverter does not walk around the house, right? 😆

And I proly even know why reload of the integration helps, it's thanks to the discovery which sends broadcast packets at the start so I guess because inverter replies there is then small window for which it works.

Anyway w/ the recent changes I implemented feature of rediscovery after every timeout. 😉

closed 2024-12-17 by davidrapan