davidrapan/ha-solarman #281: Deye SUN-12K-SG02LP1-EU not showing data for PV3 (MPPT3)

peyanski, 2024-12-04:
### Description

Hello I'm using the Deye SUN-12K-SG02LP1-EU inverter which is having 3 MPPTs where 1 & 2 are working just fine with this great integration, but 3 is not showing at all and I have no entities for it. I'm using the deye_hybrid.yaml I also tried with auto and I selected 3 MPPT during setup but no luck. Everything else is working perfectly apart from the missing MPPT 3 (PV3) What can I do ?

### Reproduction Steps

add solarman integration for SUN-12K-SG02LP1-EU inverter with the deye_hybrid.yaml config

### Log

`Logger: homeassistant.components.number
Source: helpers/entity_platform.py:764
integration: Number (documentation, issues)
First occurred: 8:42:00 PM (2 occurrences)
Last logged: 8:42:00 PM

Platform solarman does not generate unique IDs. ID Deye_2990981151_Battery Generator Charging Current already exists - ignoring number.deye_battery_generator_charging_current
Platform solarman does not generate unique IDs. ID Deye_2990981151_Battery Grid Charging Current already exists - ignoring number.deye_battery_grid_charging_current

Logger: homeassistant.components.switch
Source: helpers/entity_platform.py:764
integration: Switch (documentation, issues)
First occurred: 8:42:00 PM (1 occurrences)
Last logged: 8:42:00 PM

Platform solarman does not generate unique IDs. ID Deye_2990981151_Generator already exists - ignoring switch.deye_generator
`

### Version

24.11.02

davidrapan, 2024-12-04:
Hi @peyanski, can you show me attributes of Device sensor?

davidrapan, 2024-12-04:
[feat: Add MPPT 3 & 4 for Deye Hybrid](https://github.com/davidrapan/ha-solarman/commit/8c817fbf82c246c375dc08934e32b4665fbc7af8)

peyanski, 2024-12-04:
> Hi @peyanski, can you show me attributes of Device sensor?

sure, I'm attaching screenshots but if you prefer something else just let me know
[image]
[image]
[image]

peyanski, 2024-12-04:
it seems that is working now! Too bad that I have no Sun at the moment to confirm it. Many, many, many thanks!

davidrapan, 2024-12-05:
Can you click on sensor named "Device" to show it's details and then click on "Attributes" at the bottom of that popup window?

vinste, 2024-12-05:
Hi,
I have the same inverter ( for Deye SUN-12K-SG02LP1-EU-AM3) , and I see that the value multipliers are wrong for some power measurements (some powers appear divided by 10)
I propose attached definition that works fine for me (including MPPT3 data)
is it possible to add this definition file?
[deye_sg02lp1am3.zip](https://github.com/user-attachments/files/18022120/deye_sg02lp1am3.zip)

peyanski, 2024-12-05:
> Can you click on sensor named "Device" to show it's details and then click on "Attributes" at the bottom of that popup window?

Sure, now I got it 
[image]

peyanski, 2024-12-05:
> Hi, I have the same inverter (Deye SUN-12K-SG02LP1-EU-AM3) , and I see that the value multipliers are wrong for some power measurements (some powers appear divided by 10) I propose attached definition that works fine for me (including MPPT3 data) is it possible to add this definition file? [deye_sg02lp1am3.zip](https://github.com/user-attachments/files/18022120/deye_sg02lp1am3.zip)

indeed some power values (like grid power and load power are divided by 10). To sort this out I created a sensor templates like this {{ int(states('sensor.deye_grid_power'))*10 }} but if this is added in the definition like @vinste suggested will be better

davidrapan, 2024-12-05:
Okay so which sensors are incorrect?

vinste, 2024-12-05:
`      - name: "Grid L1 Power"
        l: 1
        class: "power"
        state_class: "measurement"
        uom: "W"
        rule: 2
        registers: [0x00A7]
        icon: "mdi:transmission-tower"

      - name: "Grid Power"
        l: 1
        class: "power"
        state_class: "measurement"
        uom: "W"
        rule: 2
        registers: [0x00A9]
        icon: "mdi:transmission-tower"
        attributes: [inverse]

      - name: "External CT1 Power"
        class: "power"
        state_class: "measurement"
        uom: "W"
        rule: 2
        registers: [0x00AA]
        icon: "mdi:transmission-tower"

      - name: "External CT2 Power"
        class: "power"
        state_class: "measurement"
        uom: "W"
        rule: 2
        registers: [0x00AB]
        icon: "mdi:transmission-tower"

      - name: "External Power"
        class: "power"
        state_class: "measurement"
        uom: "W"
        rule: 2
        registers: [0x00AC]
        icon: "mdi:transmission-tower"

      - name: "Load L1 Power"
        l: 1
        class: "power"
        state_class: "measurement"
        uom: "W"
        rule: 2
        registers: [0x00B0]

      - name: "Load L2 Power"
        l: 2
        class: "power"
        state_class: "measurement"
        uom: "W"
        rule: 2
        registers: [0x00B1]

      - name: "Load Power"
        l: 1
        class: "power"
        state_class: "measurement"
        uom: "W"
        rule: 2
        registers: [0x00B2]
`

Each of these require a scale 10

vinste, 2024-12-05:
Great, thanks.
Still, I do not know if these modifications concern all other deye hybrid inverters. 
Deye SG04lp3 and SG01hp3 are deye hybrid inverters too, but with different registers definition than SG02lp1 or what is named "hybrid" in your deifinitions.
To be on the safe side for the ones using the deye_hybrid definition, you might consider defining a SG02lp1 dedicated definition file.

davidrapan, 2024-12-05:
Yes they have their own profiles.

There is now autodetection in place using register number 0:
- String Inverter: `deye_string.yaml`
- Single-Phase Hybrid Inverter: `deye_hybrid.yaml`
- Microinverter: `deye_micro.yaml`
- LV 3-Phase Hybrid Inverter: `deye_sg04lp3.yaml`
- HV 3-Phase Hybrid Inverter: `deye_sg01hp3.yaml`
- HV 3-Phase Inverter 6-15kw: `deye_sg01hp3.yaml`
- HV 3-Phase Inverter 20-50kw: `deye_sg01hp3.yaml`

davidrapan, 2024-12-05:
[fix: Platform solarman does not generate unique IDs](https://github.com/davidrapan/ha-solarman/commit/02a7d8a666210367499fae49e66dbd497fd0a3eb)

davidrapan, 2024-12-06:
Hey guys (@peyanski and @vinste)! You have both this exact device model including power rating?

peyanski, 2024-12-06:
> Hey guys (@peyanski and @vinste)! You have both this exact device model including power rating?
> 
> Edit: It looks like that not every deye_hybrid.yaml compatible device needs that * 10 scaling... bruh.

Hey @davidrapan thanks for your efforts it is working flawlessly now for our models but that seems to hurt other the other hybrid inverters. What if we leave the SG02LP1-EU with a dedicated definition file? That will sort this issue, actually all issues. What do you think?

davidrapan, 2024-12-06:
I'd rather not give up on that autodetection. I'll try to figure out if there is any way of differentiating the devices first.

davidrapan, 2024-12-06:
I wonder if it depends on Rated Power? Solarman App has to be able to detect it too so there needs to be some way.

vinste, 2024-12-06:
It could depend on generations (SG01, SG02,... SG05), or on rated power, or on firmware version, or on family (SUN-7.6/8/10/12K-SG02LP1-EU-AM2/AM3 for instance)
It might depend on rated power, but then not only, as sg01hp3 also exist in 12k rated power, but does not have scale 10...

davidrapan, 2024-12-06:
sg01hp3 in fact does have different scaling than sg04lp3 that's why there are different files for them but in their case they also differentiate in have separate Device type so for them it's easy.

SG02 is in range above 6kW w/ 3 MPPTs and SG03 equal or lower than 6kW w/ 2 MPPTs. I guess it will be good starting point even so SG05 rated power has wider range but also w/ 2MPPTs.

davidrapan, 2024-12-06:
**v24.12.06_4**:
* [feat: Add scaling modifier + Reverted Deye Hybrid scaling](https://github.com/davidrapan/ha-solarman/commit/b0ae389e16288ef2f2cdfb4b82232004536c3ecd)

And check that new checkbox in inverter configuration to switch to `scale: 10`. This will very probably change in the future but I had to fix it for now somehow. 😉

vinste, 2024-12-08:
Hi,
With 24.12.08_5, when I check the « mod » switch, power is not scaled by ten. 
Moreover, some values are not available anymore, such as grid voltage and output voltage.

davidrapan, 2024-12-08:
Which ones exactly are gonne?

davidrapan, 2024-12-08:
[fix: Entity descs mod overwritten during autodetection](https://github.com/davidrapan/ha-solarman/commit/edcc91f1a0c925ad65f9e460e0e540941dfd8dad)

vinste, 2024-12-08:
> Which ones exactly are gonne?

Grid voltage and output voltage

davidrapan, 2024-12-08:
They are relevant only when you have more than 1 phase.

vinste, 2024-12-08:
Then where can I get grid power and voltage?

davidrapan, 2024-12-08:
You have:
* Grid Power
* Grid L1 Voltage
* Power
* Output Voltage

vinste, 2024-12-08:
I now have Grid power. 
Still, from what I understand, all L1 and L2 values are normally the values for the two other phases in a 3 phase system (otherwise you would have L3 values), or for split phase Systems. 
Grid voltage /current and power, as well as output voltage/current and power are the standards in a single-phase system. 

now there is a mix of L1 voltage and L1 current but no L1 power, and power but no voltage or current
Here is a picture of what is populated in Deye application for a single-phase system. 
[image]

davidrapan, 2024-12-08:
Yes, and because in Single Phase you have e.g. `Grid L1 Power` = `Grid Power` I left just `Grid Power`.

vinste, 2024-12-08:
Yes, but then why keep Grid power and not Grid voltage?
Now, there is a mix of split phase values (L1/L2) together with single phase ones (no L1 nor L2 mentioned):
grid power=grid L1 voltage x grid L1 current, but grid L1 power, grid voltage and grid current doesn't exist

In a single phase system, grid power= grid voltage x grid current
In a split phase system, grid L1 power = grid L1 voltage x grid L1 current and grid L2 power = grid L2 voltage x grid L2 current

davidrapan, 2024-12-08:
Because Grid Voltage is voltage between L1 and L2.

davidrapan, 2024-12-10:
Full list (structure) of phase sensors is:
* L1 Voltage, Current and Power
* L2 Voltage, Current and Power
* Then there is Power w/o L mentioned which is SUM of L1 and L2 Power 

Single/Split-Phase sensor visibility reasoning:
* Case A) Single Phase:
  * L1 Voltage and Current
  * Power as sum
* Case B) Split Phase:
  * L1 Voltage, Current and Power 
  * L2 Voltage, Current and Power
  * Power as sum

This way no matter single/split phase you always have sensor which always refers to sum power (and it doesn't matter if you also use L2 or not) - change from single to split phase only adds new sensors and doesn't change anything.

So @vinste, if you think that there is better way tell me and we can change it. 😉

vinste, 2024-12-16:
Thanks David for your concern. 
Actually, I do not really understand what these L1 values measure as it does not seem coherent:  if grid power=grid L1 current x grid L1 voltage,  L1 current is wrong on my system, as well as in my exemple hereunder, Load power should be 0 as nothing is connected to load. 
[image]

There is one remaining issue too: battery current scale is wrong too, it should be 10 times more. 
[image]

davidrapan, 2024-12-16:
* Single Phase: Grid Power = Grid L1 Current * Grid L1 Voltage
* Split Phase: Grid Power = Grid L1 Current * Grid L1 Voltage + Grid L2 Current * Grid L2 Voltage

The original request for * 10 was just for Load and Grid Power and I don't have device which uses this profile so @peyanski? Can you confirm that for battery?

vinste, 2024-12-16:
> * **Single Phase**: Grid Power = Grid L1 Current * Grid L1 Voltage

As you can see in my screen dump, grid L1 current is unfortunately wrong, as well as load power

davidrapan, 2024-12-17:
Do you have the scale modifier enabled in the options?