davidrapan/ha-solarman #285: How can we read bits from registers

PitinoB, 2024-12-05:
[image]

I need help with writing yaml code.
How can I read bits from registers?
Register 0x3104 return integer like 12610 or 12866.
I need to get values in table for that number.

Thanx

davidrapan, 2024-12-05:
You can select specific bit with:
```
bit: {index}
```
added into entity description.

PitinoB, 2024-12-05:
I dont know how to read bits from 0 to 4 and then get integer from 0 - 17 like in a table.
I tray with 
lookup:
  - bit; [0,1,2,3,4]

but not working

davidrapan, 2024-12-05:
Show me full entity description and please don't forge to wrap the code block in \`\`\`{code}\`\`\`.

PitinoB, 2024-12-05:
`  - group: Alarm-State
    items:
      - name: "DSP State"
        update_interval: 5
        class: "enum"
        rule: 1 # unsigned 16
        registers: [0x3104]
        bit: 2`

PitinoB, 2024-12-05:
this reads single register and return value 0 or 1
but I want to read five registaers at once

davidrapan, 2024-12-05:
Ah sure, I get it now! 😉 For that you need to use mask:
```
mask: 0x1F
```

davidrapan, 2024-12-05:
And there is also another route you can take:
```
      - name: "Device Alarm"
        class: "enum"
        rule: 3
        registers: [0x0229, 0x022A]
        icon: "mdi:alert-outline"
        lookup:
          - key: 0
            value: "OK"
          - bit: [0, 1, 2, 3, 4]
            value: "{state_description}"
          - key: "default"
            value: "Error"
```