davidrapan/ha-solarman #287: Deye SUN600 can not be added

SirBen88, 2024-12-06:
### Description

Try to add my Deye SUN600 micro inverter, but it shows an error (I replaced the Inverter ID by xxx): 
[XXXXXXXXXX] Device autodetection failed. [UpdateFailed: [XXXXXXXXXX] ValueError: The truth value of an array with more than one element is ambiguous. Use a.any() or a.all()]

I can access the inverter via IP adress and I used the Device serial number and not the Inverter serial number.
The port is set to 8899, yaml set to auto or micro (doesn't matter), MPPTs set to 2 and Phases set to 1.

### Reproduction Steps

Go back to v24.10.12 and it is working. Go to v24.10.16 or later -> not working.
Always trying to delete existing device and add a new one.
HAOS version is: 2024.12.0

### Log

`[XXXXXXXXXX] Device autodetection failed. [UpdateFailed: [XXXXXXXXXX] ValueError: The truth value of an array with more than one element is ambiguous. Use a.any() or a.all()]`

### Version

v24.12.04

davidrapan, 2024-12-06:
Hi @SirBen88, thank you for your report! Please can you try latest version and see if it solves the problem?

😉

SirBen88, 2024-12-07:
Hey,
I updated to v24.12.06_4 but that does not fix the problem. Still the same failure message as before.
I try to check and uncheck the "mod" checkbox, in "auto" mode and "deye_micro.yaml".

davidrapan, 2024-12-07:
Could you try to add device w/ debug log enabled, and attach the log here?

SirBen88, 2024-12-10:
Here the debug log:
```
2024-12-10 10:29:35.965 DEBUG (MainThread) [custom_components.solarman] async_setup_entry({'created_at': '2024-12-10T09:29:00.068933+00:00', 'data': {'name': 'Inverter', 'serial': OBFUSCATED_SERIAL}, 'discovery_keys': {}, 'disabled_by': None, 'domain': 'solarman', 'entry_id': '01JEQYD0X4WC6MG2KGNK1SG4YK', 'minor_version': 4, 'modified_at': '2024-12-10T09:29:00.068939+00:00', 'options': {'host': '192.168.0.60', 'additional_options': {'mppt': 2, 'phase': 1}}, 'pref_disable_new_entities': False, 'pref_disable_polling': False, 'source': 'user', 'title': 'Inverter', 'unique_id': 'solarman_OBFUSCATED_SERIAL', 'version': 1})
2024-12-10 10:29:35.965 DEBUG (MainThread) [custom_components.solarman] async_setup: coordinator.async_config_entry_first_refresh
2024-12-10 10:29:35.965 DEBUG (MainThread) [custom_components.solarman.discovery] discover
2024-12-10 10:29:35.965 DEBUG (MainThread) [custom_components.solarman.discovery] _discover_all: Broadcasting on 192.168.0.60
2024-12-10 10:29:35.965 DEBUG (MainThread) [custom_components.solarman.discovery] _discover
2024-12-10 10:29:35.972 DEBUG (MainThread) [custom_components.solarman.discovery] _discover: [192.168.0.60, E8FDF85F10C0, OBFUSCATED_SERIAL]
2024-12-10 10:29:35.973 DEBUG (MainThread) [custom_components.solarman.provider] Device autodetection is enabled
2024-12-10 10:29:35.973 DEBUG (MainThread) [custom_components.solarman.api] [OBFUSCATED_SERIAL] Scheduling 1 query request. #-1
2024-12-10 10:29:35.973 DEBUG (MainThread) [custom_components.solarman.api] [OBFUSCATED_SERIAL] Querying 03 ~ 0000 - 0022 | 0x0000 - 0x0016 # 023 ...
2024-12-10 10:29:35.973 INFO (MainThread) [custom_components.solarman.api] [OBFUSCATED_SERIAL] Connecting to 192.168.0.60:8899
2024-12-10 10:29:35.982 DEBUG (MainThread) [custom_components.solarman.api] [OBFUSCATED_SERIAL] SENT: a5 17 00 10 45 bb 00 b1 f1 71 f7 02 00 00 00 00 00 00 00 00 00 00 00 00 00 00 01 03 00 00 00 17 05 c4 17 15
2024-12-10 10:29:36.086 DEBUG (MainThread) [custom_components.solarman.api] [OBFUSCATED_SERIAL] V5_MISMATCH: 41 54 2b 59 5a 43 4d 50 56 45 52 3d 4d 57 33 5f 31 36 55 5f 35 34 30 36 5f 32 2e 33 32 2d 44 31 0d 0a 0d 0a
2024-12-10 10:29:36.291 DEBUG (MainThread) [custom_components.solarman.api] [OBFUSCATED_SERIAL] RECD: a5 41 00 10 15 bb 73 b1 f1 71 f7 02 01 1f 08 dd 01 b8 14 00 00 00 00 00 00 01 03 2e 00 04 01 00 02 01 32 32 30 39 30 36 37 31 31 32 00 01 00 00 12 0c 07 00 00 00 01 14 12 12 00 00 17 70 00 00 02 01 00 4b 00 00 00 3c 30 00 14 3d 9b 15
2024-12-10 10:29:36.292 DEBUG (MainThread) [custom_components.solarman.api] [OBFUSCATED_SERIAL] Querying 03 ~ 0000 - 0022 | 0x0000 - 0x0016 # 023 failed, attempts left: 4 [ValueError: The truth value of an array with more than one element is ambiguous. Use a.any() or a.all()]
2024-12-10 10:29:36.482 DEBUG (MainThread) [custom_components.solarman.api] [OBFUSCATED_SERIAL] V5_MISMATCH: aa 03 26 00 04 01 00 02 01 32 32 30 39 30 36 37 31 31 32 00 01 00 00 12 0c 07 00 00 00 01 14 12 12 00 00 17 70 00 00 02 01 7c af
2024-12-10 10:29:36.493 DEBUG (MainThread) [custom_components.solarman.api] [OBFUSCATED_SERIAL] Fetching failed. [Previous State: Disconnected (-1)]
2024-12-10 10:29:36.493 INFO (MainThread) [custom_components.solarman.api] [OBFUSCATED_SERIAL] Disconnecting from 192.168.0.60:8899
2024-12-10 10:29:36.493 ERROR (MainThread) [custom_components.solarman.coordinator] Unexpected error fetching Inverter data
Traceback (most recent call last):
  File "/config/custom_components/solarman/api.py", line 242, in get
    responses[(code, start)] = await self.try_read_write(code, start, quantity, f"Querying {code:02X} ~ {start:04} - {end:04} | 0x{start:04X} - 0x{end:04X} # {quantity:03}", True)
                               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/solarman/api.py", line 202, in try_read_write
    while attempts_left > 0 and not response:
                                    ^^^^^^^^
ValueError: The truth value of an array with more than one element is ambiguous. Use a.any() or a.all()

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/config/custom_components/solarman/api.py", line 140, in load
    self.device_info = await self.profile.resolve(self.get)
                       ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/solarman/provider.py", line 116, in resolve
    f = lookup_profile(await request(-1, [set_request(*AUTODETECTION_REQUEST_DEYE)]), self.attributes) if self.auto and request else self.filename
                       ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/solarman/api.py", line 256, in get
    raise Exception(f"[{self.config.serial}] {format_exception(e)}") from e
Exception: [OBFUSCATED_SERIAL] ValueError: The truth value of an array with more than one element is ambiguous. Use a.any() or a.all()

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/update_coordinator.py", line 318, in __wrap_async_setup
    await self._async_setup()
  File "/usr/src/homeassistant/homeassistant/helpers/update_coordinator.py", line 351, in _async_setup
    return await self.setup_method()
           ^^^^^^^^^^^^^^^^^^^^^^^^^
[... cut ...]
    raise Exception(f"[{self.config.serial}] Device setup failed. [{format_exception(e)}]") from e
Exception: [OBFUSCATED_SERIAL] Device setup failed. [Exception: [OBFUSCATED_SERIAL] ValueError: The truth value of an array with more than one element is ambiguous. Use a.any() or a.all()]
2024-12-10 10:30:39.294 DEBUG (MainThread) [custom_components.solarman.coordinator] async_shutdown
2024-12-10 10:30:39.295 INFO (MainThread) [custom_components.solarman.api] [OBFUSCATED_SERIAL] Disconnecting from 192.168.0.60:8899

```

SirBen88, 2024-12-10:
I manually updated to v24.12.08_11 and now it is working again!
THANK YOU!

SirBen88, 2024-12-10:
v24.12.08_11 fixed the problem.

closed 2024-12-10 by SirBen88