davidrapan/ha-solarman #298: After update to 24.12.06_2 the values of Load Power and Grid Power are incorrect.

tinntbg, 2024-12-06:
### Description

After updating to the latest release I needed re-add the integration according to the resolution stated in bug: [https://github.com/davidrapan/ha-solarman/issues/283](#283)

When I did that the values of Load Power and Grid Power where multiplied by 10. In order to fix this I have updated the configuration of the my inverter HYBRID by Deye and decrease the scale from 10 to 1. 

Can this be changed in future releases? 

### Reproduction Steps

Update to latest version and re-add integration. 

### Log

Not applicable. 

### Version

24.12.06_2

davidrapan, 2024-12-06:
Yes I'm aware of that but thank you for your report. I will do something about it.

**Could you please tell me exact model number of your device?** (e.g. SUN-12K-SG04LP3)

tinntbg, 2024-12-06:
Thanks for the quick response! My device is: Deye SUN-8K-SG01LP1-EU

davidrapan, 2024-12-06:
Could you also do for me a screenshot like this one? https://github.com/davidrapan/ha-solarman/issues/281#issuecomment-2520017035

Fennsy, 2024-12-06:
Hello, I've the same inverter : Deye SUN-8K-SG01LP1-EU with the same problem than @tinntbg and this is my screenshot.
In the screenshot the param Phases is 3 but I have only 1 Phase :/

[image]

davidrapan, 2024-12-06:
> In the screenshot the param Phases is 3 but I have only 1 Phase :/

Guess your inverter has it incorrectly written in the registers (Nothing to be really worried about as your device was correctly identified as Single Phase Hybrid Inverter, numeric value: 3).

v24.12.06_4:
* [feat: Add scaling modifier + Reverted Deye Hybrid scaling](https://github.com/davidrapan/ha-solarman/commit/b0ae389e16288ef2f2cdfb4b82232004536c3ecd)

Fennsy, 2024-12-07:
s this phase problem due to the configuration of the inverter in the integration or is it directly in the inverter ?
In any case I tested the scale problem for consumption and everything is ok
Thanks

davidrapan, 2024-12-07:
> Is this phase problem due to the configuration of the inverter in the integration or is it directly in the inverter ?

That value is read directly from your device.

tinntbg, 2024-12-07:
Thanks for the quick fix! I have tested as well and it all looks to be working now! 

Should I close the ticket?

davidrapan, 2024-12-07:
You can leave it for a while. 😉

davidrapan, 2024-12-09:
Can please someone confirm if is working single phase detection is working in latest release? And I don't mean displayed value in the device sensor's attributes but in the amount of added sensors to the list (= sensors w/ L2 are not present).

tinntbg, 2024-12-09:
Hi @davidrapan, 

I have just deleted and re-added my inverter in home assistant and from what I can see there are a lot of sensors with L2 names. 

So the short answer is that it's not working. 

Thanks!

tinntbg, 2024-12-09:
If I manually change the number of MPPTs and the number of phases then the values are not created, but the auto detection is not working with release 24.12.08_9

davidrapan, 2024-12-09:
Could you please show me read of register number 8 using services? Thank you!

tinntbg, 2024-12-09:
Is this what you are looking for or something else? 
[image]

davidrapan, 2024-12-09:
Mh I guess that's the culprit. So it's either not used or 0 also means single phase...

davidrapan, 2024-12-10:
So what's the current state @tinntbg?

tinntbg, 2024-12-10:
Hey @davidrapan, 

Just updated to the latest version, deleted and re-added my inverter, but unfortunately, nothing has changed. I still see L2 values and it's still been detected as a 3 phase inverter. 

Thanks!

tinntbg, 2024-12-10:
Here are all the Auto discovered values: 
[image]

ikeare, 2024-12-13:
Same problem on 12k sg01lp1.
Defined as 1 phase, but essentially 3 phase. Also, “scale” for “GRID Power” and “LOAD Power” does not work correctly, I had to manually change the values

davidrapan, 2024-12-13:
> I had to manually change the values

What do you mean by that? Where?

ikeare, 2024-12-14:
> > Same problem on 12k sg01lp1
> 
> This is not enough information.
> 
> > I had to manually change the values
> 
> What do you mean by that? Where?

[image]
After a recent update of the Deye inverter, integer numbers in watts stopped being displayed, that is, it was 1043 watts, it became 104, judging by Deye Cloud - they removed this figure and multiplied the value by 10, their watts are always equal to 10.
In order to fix this I changed the value of "scale: 10" to "Grid Power" and "Load Power" in deye_hybrid.yaml

davidrapan, 2024-12-14:
There is a switch for that in configuration. 😉

nazeg, 2025-02-12:
The PV power sensor values are also incorrect. They appear to be high after the update.

davidrapan, 2025-02-12:
You will have to provide way more details if you want me to look into it.

nazeg, 2025-02-13:
> You will have to provide way more details if you want me to look into it.

As you can see, I see a maximum of 1.41 kW on the Solarman app, but the Home Assistant addon shows 1.794 kW at the same time. Also, in the other photo, it shows about 5.5 kW, but my PV1 power cannot exceed 5000 watts. Additionally, there wasn't enough sunlight yesterday. Something is wrong. There was no issue 3–4 days ago.

[image]
[image]
[image]

davidrapan, 2025-02-13:
What device it is? I need you to enable debug logging for couple of hours throughout the day and attach the file here.

davidrapan, 2025-02-13:
> How can I enable debugging for this addon?

It's integration, not addon.

> Can u help?

[image]

nazeg, 2025-02-13:
> > How can I enable debugging for this addon?
> 
> It's integration, not addon.
> 
> > Can u help?
> 
> [image]

Sorry, i found it and enabled. How can i download it? I couldn't see any link for that.

nazeg, 2025-02-13:
> > How can I enable debugging for this addon?
> 
> It's integration, not addon.
> 
> > Can u help?
> 
> [image]

Sorry, i found it enabled it. I will u send it tomorrow.

nazeg, 2025-02-13:
> What device it is? I need you to enable debug logging for couple of hours throughout the day and attach the file here.

I attached it

[home-assistant_solarman_2025-02-13T13-35-29.705Z.log](https://github.com/user-attachments/files/18784619/home-assistant_solarman_2025-02-13T13-35-29.705Z.log)

davidrapan, 2025-02-13:
I can see that there is proly broken scaling. Values should be 10 times lower.

Can you do following steps please:
* Enabled debug logging (of Solarman)
* Restart HA
* Disabled debug logging
* Attach the file

This will show me setup flow during start and I will able to determine where is the issue exactly.

Thank you!

nazeg, 2025-02-13:
> I can see that there is proly broken scaling. Values should be 10 times lower.
> 
> Can you do following steps please:
> 
> * Enabled debug logging (of Solarman)
> * Restart HA
> * Disabled debug logging (of Solarman)
> * Attach the file here
> 
> This will show me setup flow during start and I will able to determine where is the issue exactly.
> 
> Thank you!

Here it's:

[home-assistant_solarman_2025-02-13T13-59-01.620Z.log](https://github.com/user-attachments/files/18784887/home-assistant_solarman_2025-02-13T13-59-01.620Z.log)

davidrapan, 2025-02-13:
What were the values for **PV1 Power** and **PV Power** ~+- during the latest debug window?

nazeg, 2025-02-13:
> What were the values for **PV1 Power** and **PV Power** ~+- during the latest debug window?

PV1 was about 500watt and PV was about 1000watt

davidrapan, 2025-02-13:
But it does match values provided by your device:
* **PV1 Voltage**: 583.5 V
* **PV1 Current**: 1 A
* **PV2 Voltage**: 405.4 V
* **PV2 Current**: 0.7 V

Maybe the scaling is different altogether? Can you try to compare Voltage and Current values against Solarman App?

Because when I test it (Added **PV Power - calculated** which calculates power from voltage and current) it seems alright:
[image]

davidrapan, 2025-02-13:
Are you also able to compare PV voltages and currents against Solarman App/inverter?

nazeg, 2025-02-13:
> Are you also able to compare PV voltages and currents against Solarman App/inverter?

I will try tomorrow morning.

nazeg, 2025-02-14:
> What were the values for **PV1 Power** and **PV Power** ~+- during the latest debug window?

I think the values are ok now. They're going same with solarman. Thank you for your interest and help.

nazeg, 2025-02-14:
> Are you also able to compare PV voltages and currents against Solarman App/inverter?

I think the issue has gone. The values are the same with solarman now. Thank you for your interest and help.

davidrapan, 2025-02-14:
Hmmm, interesting.. but I'm glad it's working again!

gabrydiamonds, 2025-03-17:
hello, thanks for the work done for this very useful integration. I don't know if this is the right place, but I would like to report that even in my integration of my SUN-12K-SG01LP1-EU inverter I have some sensors with the wrong scale. They are multiplied by 10. I attach some images:

[image]
[image]

gabrydiamonds, 2025-07-18:
Hi, are there any updates to fix this issue? In my case, the grid power and the load power are multiplied by 10 instead of 1 and for me this is a wrong scale.

ikeare, 2025-07-18:
Hi @gabrydiamonds 
Have you tried updating the logger firmware with Deye support?
After the update everything worked fine for me.

davidrapan, 2025-08-09:
Hi @gabrydiamonds, check modifier in configuration under additional options.