davidrapan/ha-solarman #301: SOFAR LSW3 setup/connection problem

maniac2003, 2024-12-07:
### Description

Hi all,

I've got the following error during installation:
[image]

Any ideas?

Regards
 Eike

### Reproduction Steps

After install

### Log

see description

### Version

24.12.08_1

davidrapan, 2024-12-07:
Hi @maniac2003, that means that this profile is not for your device (contains registers which are not used in your device).

maniac2003, 2024-12-07:
Hi @davidrapan, thanks for the fast reply.

I have a Sofar inverter HYD 10KTL-3PH-PRO with logger LSW3.
The profile I have chosen is sofar_lsw3.yaml.

Is the inverter / logger compatible with your integration?

Regards
 Eike

davidrapan, 2024-12-08:
You should proly try `sofar_g3hyd.yaml`.

maniac2003, 2024-12-08:
Perfect, it's working now, thanks.

closed 2024-12-08 by maniac2003