davidrapan/ha-solarman #302: Error: struct.error: unpack requires a buffer of 4 bytes

jkolo, 2024-12-08:
### Description

When trying to change the configuration of the Deye SUN-12K-SG04-LP3 inverter a Home Assistant WebSocket API error occurs. 

### Reproduction Steps

1. Add the Deye SUN-12K-SG04-LP3 inverter to Home Assistant using the Solarman custom component and deye_sg04lp3.yaml profile.
2. Attempt to change an adjustable parameter from the Home Assistant UI (e.g. Battery Grid Charging).
3. Observe the WebSocket error in the logs.

### Log

```
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/components/websocket_api/commands.py", line 245, in handle_call_service
    response = await hass.services.async_call(
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
    ...<7 lines>...
    )
    ^
  File "/usr/src/homeassistant/homeassistant/core.py", line 2802, in async_call
    response_data = await coro
                    ^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/core.py", line 2845, in _execute_service
    return await target(service_call)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/service.py", line 1007, in entity_service_call
    single_response = await _handle_entity_call(
                      ^^^^^^^^^^^^^^^^^^^^^^^^^^
        hass, entity, func, data, call.context
        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
    )
    ^
  File "/usr/src/homeassistant/homeassistant/helpers/service.py", line 1079, in _handle_entity_call
    result = await task
             ^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/number/__init__.py", line 122, in async_set_value
    await entity.async_set_native_value(native_value)
  File "/config/custom_components/solarman/number.py", line 75, in async_set_native_value
    await self.write(value_int if value_int < 0xFFFF else 0xFFFF, get_number(value))
  File "/config/custom_components/solarman/entity.py", line 150, in write
    if await self.coordinator.inverter.call(self.code, self.register, value, ACTION_ATTEMPTS_MAX) > 0 and state:
       ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/solarman/api.py", line 281, in call
    response = await self.safe_read_write(code, start, arg)
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/solarman/api.py", line 176, in safe_read_write
    if (response := await self.read_write(code, start, arg)) and (length := ilen(response)) and (expected := arg if code < CODE.WRITE_SINGLE_COIL else 1) and length != expected:
                    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/solarman/api.py", line 171, in read_write
    return await self.write_multiple_holding_registers(start, ensure_list(arg))
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/solarman/api.py", line 109, in write_multiple_holding_registers
    return await self._tcp_parse_response_adu(write_multiple_registers(self.mb_slave_id, register_addr, values))
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/solarman/api.py", line 33, in _tcp_parse_response_adu
    return parse_response_adu(await self._tcp_send_receive_frame(mb_request_frame), mb_request_frame)
  File "/usr/local/lib/python3.13/site-packages/umodbus/client/tcp.py", line 235, in parse_response_adu
    function = create_function_from_response_pdu(resp_pdu, req_adu)
  File "/usr/local/lib/python3.13/site-packages/umodbus/functions.py", line 140, in create_function_from_response_pdu
    return function.create_from_response_pdu(resp_pdu)
           ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^
  File "/usr/local/lib/python3.13/site-packages/umodbus/functions.py", line 1572, in create_from_response_pdu
    starting_address, data = struct.unpack('>HH', resp_pdu[1:5])
                             ~~~~~~~~~~~~~^^^^^^^^^^^^^^^^^^^^^^
struct.error: unpack requires a buffer of 4 bytes
```

### Version

24.12.06_3

### Home Assistant Version

2024.12.1

davidrapan, 2024-12-08:
Hi @jkolo, thank your for your report.

This issue was fixed in [v24.12.08_2](https://github.com/davidrapan/ha-solarman/releases/tag/v24.12.08_2) and latest release is [v24.12.08_4](https://github.com/davidrapan/ha-solarman/releases/tag/v24.12.08_4).

jkolo, 2024-12-08:
After updating to version 24.12.08_4 and, of course, restarting, the error still occurs.

Also, I switched the profile to `auto`.
```
[546800674000] Unexpected exception

Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/components/websocket_api/commands.py", line 245, in handle_call_service
    response = await hass.services.async_call(
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
    ...<7 lines>...
    )
    ^
  File "/usr/src/homeassistant/homeassistant/core.py", line 2802, in async_call
    response_data = await coro
                    ^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/core.py", line 2845, in _execute_service
    return await target(service_call)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/service.py", line 1007, in entity_service_call
    single_response = await _handle_entity_call(
                      ^^^^^^^^^^^^^^^^^^^^^^^^^^
        hass, entity, func, data, call.context
        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
    )
    ^
  File "/usr/src/homeassistant/homeassistant/helpers/service.py", line 1079, in _handle_entity_call
    result = await task
             ^^^^^^^^^^
  File "/config/custom_components/solarman/switch.py", line 82, in async_turn_off
    await self.write(value, value)
  File "/config/custom_components/solarman/entity.py", line 150, in write
    if await self.coordinator.inverter.call(self.code, self.register, value, ACTION_ATTEMPTS_MAX) > 0 and state:
       ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/solarman/api.py", line 281, in call
    response = await self.safe_read_write(code, start, arg)
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/solarman/api.py", line 176, in safe_read_write
    if (response := await self.read_write(code, start, arg)) and (length := ilen(response)) and (expected := arg if code < CODE.WRITE_SINGLE_COIL else 1) and length != expected:
                    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/solarman/api.py", line 171, in read_write
    return await self.write_multiple_holding_registers(start, ensure_list(arg))
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/solarman/api.py", line 109, in write_multiple_holding_registers
    return await self._tcp_parse_response_adu(write_multiple_registers(self.mb_slave_id, register_addr, values))
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/solarman/api.py", line 33, in _tcp_parse_response_adu
    return parse_response_adu(await self._tcp_send_receive_frame(mb_request_frame), mb_request_frame)
  File "/usr/local/lib/python3.13/site-packages/umodbus/client/tcp.py", line 235, in parse_response_adu
    function = create_function_from_response_pdu(resp_pdu, req_adu)
  File "/usr/local/lib/python3.13/site-packages/umodbus/functions.py", line 140, in create_function_from_response_pdu
    return function.create_from_response_pdu(resp_pdu)
           ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^
  File "/usr/local/lib/python3.13/site-packages/umodbus/functions.py", line 1572, in create_from_response_pdu
    starting_address, data = struct.unpack('>HH', resp_pdu[1:5])
                             ~~~~~~~~~~~~~^^^^^^^^^^^^^^^^^^^^^^
struct.error: unpack requires a buffer of 4 bytes
```
There was also a second error: `Error fetching SUN-12K-SG04-LP3 data: [2110645731] Busy: Currently writing to the device! `

jkolo, 2024-12-08:
It may be useful to know that I use an Ethernet logger LSE3. Intuition tells me that the fix in v24.12.06 version may not have covered this case :)

davidrapan, 2024-12-08:
The fix was exactly for ethernet loggers. 😉

jkolo, 2024-12-08:
Damn, intuition to the trash.... Below log.
```log
[36m2024-12-08 15:52:03.650 DEBUG (MainThread) [custom_components.solarman.api] [2110645731] Scheduling 1 query request. #39905[0m
[36m2024-12-08 15:52:03.650 DEBUG (MainThread) [custom_components.solarman.api] [2110645731] Querying 03 ~ 0586 - 0708 | 0x024A - 0x02C4 # 123 ...[0m
[36m2024-12-08 15:52:03.650 DEBUG (MainThread) [custom_components.solarman.api] [2110645731] SENT: 53 87 00 00 00 06 01 03 02 4a 00 7b[0m
[36m2024-12-08 15:52:04.106 DEBUG (MainThread) [custom_components.solarman.api] [2110645731] RECD: 53 87 00 00 00 f9 01 03 f6 04 7e 14 e5 00 30 00 00 ff e8 ff d2 01 2c 00 00 00 00 00 00 00 00 00 00 09 2f 09 2b 09 22 00 00 00 00 00 00 01 cf 00 a2 ff e5 02 56 00 00 13 88 00 c2 00 4e 00 2b 00 e3 00 a5 00 b9 01 c0 01 a3 01 6e 04 d1 00 00 00 00 01 c0 01 a3 01 6e 04 d1 00 00 09 37 09 2f 09 25 00 00 ff ba 00 00 00 24 ff 65 00 23 ff ac ff ac 13 88 00 00 01 f3 00 00 00 00 02 02 09 37 09 2f 09 24 00 00 00 00 00 00 01 e4 01 08 01 91 04 7d 04 7d 13 88 00 00 00 00 00 00 00 00 00 00 00 04 00 01 00 04 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 04 ed 00 00 06 06 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 ff ff 00 00 ff ff ff ff 00 00 00 00 00 00 00 00 00 00 00 00 ff ff 00 00 00 00 00 00 00 00 00 00 00 00[0m
[36m2024-12-08 15:52:04.107 DEBUG (MainThread) [custom_components.solarman.api] [2110645731] Querying 03 ~ 0586 - 0708 | 0x024A - 0x02C4 # 123 succeeded.[0m
[36m2024-12-08 15:52:04.109 DEBUG (MainThread) [custom_components.solarman.api] [2110645731] Returning 67 new values to the Coordinator. [Previous State: Connected (1)][0m
[36m2024-12-08 15:52:04.109 DEBUG (MainThread) [custom_components.solarman.coordinator] Finished fetching SUN-12K-SG04-LP3 data in 0.462 seconds (success: True)[0m
[36m2024-12-08 15:52:09.693 DEBUG (MainThread) [custom_components.solarman.api] [2110645731] Scheduling 1 query request. #39910[0m
[36m2024-12-08 15:52:09.694 DEBUG (MainThread) [custom_components.solarman.api] [2110645731] Querying 03 ~ 0586 - 0708 | 0x024A - 0x02C4 # 123 ...[0m
[36m2024-12-08 15:52:09.694 DEBUG (MainThread) [custom_components.solarman.api] [2110645731] SENT: aa 73 00 00 00 06 01 03 02 4a 00 7b[0m
[36m2024-12-08 15:52:10.173 DEBUG (MainThread) [custom_components.solarman.api] [2110645731] RECD: aa 73 00 00 00 f9 01 03 f6 04 7e 14 e5 00 30 00 00 ff ea ff d6 01 2c 00 00 00 00 00 00 00 00 00 00 09 30 09 2d 09 24 00 00 00 00 00 00 01 b1 00 a3 ff e5 02 39 00 00 13 88 00 b5 00 4f 00 2b 00 d8 00 ad 00 b9 01 bc 01 a4 01 6d 04 cd 00 00 00 00 01 bc 01 a4 01 6d 04 cd 00 00 09 37 09 2f 09 25 00 00 ff b0 00 00 00 24 ff 64 00 23 ff ab ff ab 13 88 00 00 01 d5 00 00 00 00 01 e4 09 37 09 31 09 26 00 00 00 00 00 00 01 e0 01 08 01 90 04 78 04 78 13 88 00 00 00 00 00 00 00 00 00 00 00 04 00 01 00 04 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 03 65 00 00 06 01 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 ff ff 00 00 ff ff ff ff 00 00 00 00 00 00 00 00 00 00 00 00 ff ff 00 00 00 00 00 00 00 00 00 00 00 00[0m
[36m2024-12-08 15:52:10.174 DEBUG (MainThread) [custom_components.solarman.api] [2110645731] Querying 03 ~ 0586 - 0708 | 0x024A - 0x02C4 # 123 succeeded.[0m
[36m2024-12-08 15:52:10.176 DEBUG (MainThread) [custom_components.solarman.api] [2110645731] Returning 67 new values to the Coordinator. [Previous State: Connected (1)][0m
[36m2024-12-08 15:52:10.176 DEBUG (MainThread) [custom_components.solarman.coordinator] Finished fetching SUN-12K-SG04-LP3 data in 0.485 seconds (success: True)[0m
[36m2024-12-08 15:52:12.777 DEBUG (MainThread) [custom_components.solarman.api] [2110645731] Call 10 ~ 145 | 0x0091: 1 ...[0m
[36m2024-12-08 15:52:12.777 DEBUG (MainThread) [custom_components.solarman.api] [2110645731] SENT: e5 35 00 00 00 09 01 10 00 91 00 01 02 00 01[0m
[36m2024-12-08 15:52:12.960 DEBUG (MainThread) [custom_components.solarman.api] [2110645731] RECD: e5 35 00 00 00 03 01 10 00[0m
[36m2024-12-08 15:52:12.961 DEBUG (MainThread) [custom_components.solarman.api] [2110645731] Call 10 ~ 145 | 0x0091: 1 failed, attempts left: 4 [error: unpack requires a buffer of 4 bytes][0m
[36m2024-12-08 15:52:13.162 DEBUG (MainThread) [custom_components.solarman.api] [2110645731] SENT: 98 62 00 00 00 09 01 10 00 91 00 01 02 00 01[0m
[36m2024-12-08 15:52:13.424 DEBUG (MainThread) [custom_components.solarman.api] [2110645731] RECD: 98 62 00 00 00 03 01 10 00[0m
[36m2024-12-08 15:52:13.425 DEBUG (MainThread) [custom_components.solarman.api] [2110645731] Call 10 ~ 145 | 0x0091: 1 failed, attempts left: 3 [error: unpack requires a buffer of 4 bytes][0m
[36m2024-12-08 15:52:13.626 DEBUG (MainThread) [custom_components.solarman.api] [2110645731] SENT: b3 47 00 00 00 09 01 10 00 91 00 01 02 00 01[0m
[36m2024-12-08 15:52:13.878 DEBUG (MainThread) [custom_components.solarman.api] [2110645731] RECD: b3 47 00 00 00 03 01 10 00[0m
[36m2024-12-08 15:52:13.879 DEBUG (MainThread) [custom_components.solarman.api] [2110645731] Call 10 ~ 145 | 0x0091: 1 failed, attempts left: 2 [error: unpack requires a buffer of 4 bytes][0m
[36m2024-12-08 15:52:14.080 DEBUG (MainThread) [custom_components.solarman.api] [2110645731] SENT: 9e f0 00 00 00 09 01 10 00 91 00 01 02 00 01[0m
[36m2024-12-08 15:52:14.343 DEBUG (MainThread) [custom_components.solarman.api] [2110645731] RECD: 9e f0 00 00 00 03 01 10 00[0m
[... cut ...]
[36m2024-12-08 15:53:36.542 DEBUG (MainThread) [custom_components.solarman.api] [2110645731] Querying 03 ~ 0586 - 0708 | 0x024A - 0x02C4 # 123 succeeded.[0m
[36m2024-12-08 15:53:36.547 DEBUG (MainThread) [custom_components.solarman.api] [2110645731] Returning 90 new values to the Coordinator. [Previous State: Connected (1)][0m
[36m2024-12-08 15:53:36.548 DEBUG (MainThread) [custom_components.solarman.coordinator] Finished fetching SUN-12K-SG04-LP3 data in 0.900 seconds (success: True)[0m
```

davidrapan, 2024-12-08:
What is that messed up formatting? `�[0m`

Anyway it looks like your sticks replies are even worse... they are missing not just quantity block but also register address...

davidrapan, 2024-12-08:
**The error is just from parsing of the response so the question is does the value change in the inverter or not?**

davidrapan, 2024-12-09:
If yes then try [fix: Workaround for modbus tcp responses w/o address](https://github.com/davidrapan/ha-solarman/commit/9c802f39c510027406893e09434fa9d85f8b0564) and let me know. 😉

jkolo, 2024-12-10:
Well, it's "worse" now. In the previous version an error would pop up, but the switch would remain in the state it was set in. Now the error does not pop up, but.... Turning the switch on works ok: the switch switches, a logbook entry appears and the inverter responds. When trying to switch off, the switch toggles, but no entry in the logbook and after a while it returns to the on state. Interestingly the inverter responds correctly to the option being switched off and does not switch the option back on even though on the interface the switch returns to the on state. Below is an overview video and log.

[Nagranie ekranu_20241211_000152.webm](https://github.com/user-attachments/assets/ae1e56d2-76bc-4925-a4fa-fc8431077bc7)

```
2024-12-11 00:01:28.668 DEBUG (MainThread) [custom_components.solarman] async_setup_entry({'created_at': '2024-10-31T23:58:26.896638+00:00', 'data': {'battery_life_cycle_rating': 6000, 'battery_nominal_voltage': 48, 'inverter_host': '10.38.107.18', 'inverter_mb_slave_id': 1, 'inverter_port': 8899, 'inverter_serial': 2110645731, 'lookup_file': 'deye_sg04lp3.yaml', 'name': 'SUN-12K-SG04-LP3'}, 'discovery_keys': {}, 'disabled_by': None, 'domain': 'solarman', 'entry_id': '01JBJG892GNQX5T5R0RYX1VZ2Q', 'minor_version': 5, 'modified_at': '2024-12-10T22:23:48.284280+00:00', 'options': {'additional_options': {'battery_nominal_voltage': 51, 'mppt': 2}, 'host': '10.38.107.18'}, 'pref_disable_new_entities': False, 'pref_disable_polling': False, 'source': 'user', 'title': 'SUN-12K-SG04-LP3', 'unique_id': 'solarman_2110645731', 'version': 1})
2024-12-11 00:01:28.669 DEBUG (MainThread) [custom_components.solarman] async_setup: coordinator.async_config_entry_first_refresh
2024-12-11 00:01:28.669 DEBUG (MainThread) [custom_components.solarman.discovery] discover
2024-12-11 00:01:28.670 DEBUG (MainThread) [custom_components.solarman.discovery] _discover_all: Broadcasting on 10.38.107.18
2024-12-11 00:01:28.670 DEBUG (MainThread) [custom_components.solarman.discovery] _discover
2024-12-11 00:01:28.674 DEBUG (MainThread) [custom_components.solarman.discovery] _discover: [10.38.107.18, 402A8F2CD64B, 2110645731]
2024-12-11 00:01:28.677 DEBUG (MainThread) [custom_components.solarman.provider] Device autodetection is enabled
2024-12-11 00:01:28.678 DEBUG (MainThread) [custom_components.solarman.api] [2110645731] Scheduling 0 query requests. ^-1
2024-12-11 00:01:28.678 DEBUG (MainThread) [custom_components.solarman.api] [2110645731] Querying 03 ~ 0000 - 0022 | 0x0000 - 0x0016 #023 ...
2024-12-11 00:01:28.679 INFO (MainThread) [custom_components.solarman.api] [2110645731] Connecting to 10.38.107.18:8899
2024-12-11 00:01:28.685 DEBUG (MainThread) [custom_components.solarman.api] [2110645731] SENT: a5 17 00 10 45 9c 00 e3 e5 cd 7d 02 00 00 00 00 00 00 00 00 00 00 00 00 00 00 01 03 00 00 00 17 05 c4 00 15
2024-12-11 00:01:29.790 DEBUG (MainThread) [custom_components.solarman.api] [2110645731] V5_ETHERNET_DETECTED: a5 17 00 10 45 03 00 e3 02
2024-12-11 00:01:29.790 DEBUG (MainThread) [custom_components.solarman.api] [2110645731] RECD: a5 17 00 10 45 03 00 e3 02
2024-12-11 00:01:29.791 DEBUG (MainThread) [custom_components.solarman.api] frame_len does not match payload_len.
2024-12-11 00:01:29.791 DEBUG (MainThread) [custom_components.solarman.api] [2110645731] Querying 03 ~ 0000 - 0022 | 0x0000 - 0x0016 #023 failed, attempts left: 4 [V5FrameError: V5 frame contains invalid start or end values]
2024-12-11 00:01:29.992 DEBUG (MainThread) [custom_components.solarman.api] [2110645731] SENT: 92 9d 00 00 00 06 01 03 00 00 00 17
2024-12-11 00:01:30.295 DEBUG (MainThread) [custom_components.solarman.api] [2110645731] RECD: 92 9d 00 00 00 31 01 03 2e 00 05 00 01 01 04 32 34 30 35 32 33 36 37 31 33 00 00 00 00 00 00 18 07 d2 07 00 00 20 06 11 44 00 00 10 01 c0 47 00 00 d4 c0 00 01 ff ff
2024-12-11 00:01:30.295 DEBUG (MainThread) [custom_components.solarman.api] [2110645731] Querying 03 ~ 0000 - 0022 | 0x0000 - 0x0016 #023 succeeded, response: [5, 1, 260, 12852, 12341, 12851, 13879, 12595, 0, 0, 0, 6151, 53767, 0, 8198, 4420, 0, 4097, 49223, 0, 54464, 1, 65535]
2024-12-11 00:01:30.295 DEBUG (MainThread) [custom_components.solarman.api] [2110645731] Returning 1 new values to the Coordinator. [Previous State: Disconnected (-1)]
2024-12-11 00:01:31.308 DEBUG (MainThread) [custom_components.solarman.parser] Defaults for update_interval: 60, code: 3, min_span: 25, max_size: 125, digits: 6
2024-12-11 00:01:31.314 DEBUG (MainThread) [custom_components.solarman.api] {'connections': {('mac', '40:2a:8f:2c:d6:4b')}, 'identifiers': {('solarman', 2110645731)}, 'serial_number': 2110645731, 'manufacturer': 'Deye', 'model': 'SG04LP3', 'name': 'SUN-12K-SG04-LP3'}
2024-12-11 00:01:31.321 DEBUG (MainThread) [custom_components.solarman.api] [2110645731] Scheduling 5 query requests. ^0
2024-12-11 00:01:31.322 DEBUG (MainThread) [custom_components.solarman.api] [2110645731] Querying 03 ~ 0000 - 0124 | 0x0000 - 0x007C #125 ...
2024-12-11 00:01:31.322 DEBUG (MainThread) [custom_components.solarman.api] [2110645731] SENT: 7d 19 00 00 00 06 01 03 00 00 00 7d
2024-12-11 00:01:31.769 DEBUG (MainThread) [custom_components.solarman.api] [2110645731] RECD: 7d 19 00 00 00 fd 01 03 fa 00 05 00 01 01 04 32 34 30 35 32 33 36 37 31 33 00 00 00 00 00 00 18 07 d2 07 00 00 20 06 11 44 00 00 10 01 c0 47 00 00 d4 c0 00 01 ff ff 00 00 00 06 00 00 00 00 00 00 ff ff ff ff 00 00 00 00 00 00 00 ff 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 0 [... cut ...] , 0, 0, 4, 1, 4, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 290, 0, 91, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 65535, 65535, 65535, 65535, 65535, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0]
2024-12-11 00:02:58.317 DEBUG (MainThread) [custom_components.solarman.api] [2110645731] Returning 67 new values to the Coordinator. [Previous State: Connected (1)]
2024-12-11 00:02:58.317 DEBUG (MainThread) [custom_components.solarman.coordinator] Finished fetching SUN-12K-SG04-LP3 data in 0.524 seconds (success: True)
```

davidrapan, 2024-12-11:
There was a bug!

[fix: Set of configurables to 0 did not update state in HA](https://github.com/davidrapan/ha-solarman/commit/5d4c3374069355553b14c77882974311a981b3c6)

Thanks for catching that out!

closed 2024-12-17 by davidrapan