davidrapan/ha-solarman #305: Log: sensor.solarman_today_production is using state class 'measurement'

philmassie, 2024-12-08:
### Description

Hello

This is popping up in my logs, couldnt find it mentioned, but might have missed it.

`Entity sensor.solarman_today_production (<class 'custom_components.solarman.sensor.SolarmanSensor'>) is using state class 'measurement' which is impossible considering device class ('energy') it is using; expected None or one of 'total_increasing', 'total'; Please update your configuration if your entity is manually configured, otherwise create a bug report at https://github.com/davidrapan/ha-solarman/issues`

### Reproduction Steps

Home assistant version:
Core 2024.12.1
Supervisor 2024.11.4

Solarman: 24.12.08_5

Not aware of anything I've done specifically to cause this

### Log

2024-12-08 12:33:59.240 WARNING (MainThread) [homeassistant.components.sensor] Entity sensor.solarman_today_production (<class 'custom_components.solarman.sensor.SolarmanSensor'>) is using state class 'measurement' which is impossible considering device class ('energy') it is using; expected None or one of 'total_increasing', 'total'; Please update your configuration if your entity is manually configured, otherwise create a bug report at https://github.com/davidrapan/ha-solarman/issues

### Version

24.12.08_5

### Home Assistant Version

2024.12.1

CrazyUs3r, 2024-12-08:
Witch yaml and what inverter

philmassie, 2024-12-08:
Right - Sorry
deye_hybrid.yaml
Deye 5KW Hybrid

CrazyUs3r, 2024-12-08:
Try to Chance in the yaml line 532 to total_increasing

https://github.com/davidrapan/ha-solarman/blob/b57b988af22bc628470b98ff7afbd2190900264c/custom_components/solarman/inverter_definitions/deye_hybrid.yaml#L532

philmassie, 2024-12-08:
Thanks very much, I will do.
Out of interest, would that survive an update, or would I make that change each time I update?

CrazyUs3r, 2024-12-08:
If it works, you can write it here so that @davidrapan can take it over if no one else has the problem. Then it would be included in the next update

philmassie, 2024-12-08:
ok cool, yes that seems to have worked! Thanks very much :)

davidrapan, 2024-12-08:
👍 I'll fix it.

closed 2024-12-10 by davidrapan