davidrapan/ha-solarman #308: Update from v24.11.02 - 4mppt missing

Wubarius, 2024-12-09:
### Description

Hello,

Got the update info today and installed it.
From 24.11.02 to 24.12.08_09

My inverter config was:
deye_4mppt.yaml

This file was deleted after the update.
Manually recreating it did not help.
I no longer get any inverter data.

Need help please....

### Reproduction Steps

Update from 24.11.02 - to 24.12.08_09

### Log

Logger: homeassistant.config_entries
Quelle: config_entries.py:640
Erstmals aufgetreten: 13:09:58 (2 Vorkommnisse)
Zuletzt protokolliert: 13:09:58

Error setting up entry West-PV for solarman
Error setting up entry Garage-PV for solarman
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/config_entries.py", line 640, in __async_setup_with_context
    result = await component.async_setup_entry(hass, self)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/solarman/__init__.py", line 56, in async_setup_entry
    await async_migrate_entries(hass, config_entry.entry_id, partial(migrate_unique_ids, config.name, config.serial))
  File "/usr/src/homeassistant/homeassistant/helpers/entity_registry.py", line 1636, in async_migrate_entries
    ent_reg.async_update_entity(entry.entity_id, **updates)
    ~~~~~~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/entity_registry.py", line 1179, in async_update_entity
    return self._async_update_entity(
           ~~~~~~~~~~~~~~~~~~~~~~~~~^
        entity_id,
        ^^^^^^^^^^
    ...<21 lines>...
        unit_of_measurement=unit_of_measurement,
        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
    )
    ^
  File "/usr/src/homeassistant/homeassistant/helpers/entity_registry.py", line 1118, in _async_update_entity
    raise ValueError(
    ...<2 lines>...
    )
ValueError: Unique id 'west_pv_4064093016_over_frequency_load_reduction_switch' is already in use by 'switch.west_pv_over_frequency_load_reduction'

### Version

24.12.08_09

### Home Assistant Version

2024.12.1

CrazyUs3r, 2024-12-09:
You have change to deye_micro.yaml?
And Set all setting by gui?

davidrapan, 2024-12-09:
One should not have to change anything in theory. But that duplicate id is weird.

Wubarius, 2024-12-09:
Hi,

Ok, I deleted everything and reconfigured them. 
Values arrive again, but everything has to be rebuilt as entities are called differently.

That takes time again.... 

Thanks so far....

closed 2024-12-09 by Wubarius

davidrapan, 2024-12-09:
There is migration in place which should take care of everything but can't figure out where that second `west_pv_4064093016_over_frequency_load_reduction_switch` came from... 🤯

davidrapan, 2024-12-09:
Mh, @Wubarius, It would be very helpful if you could do partial buckup restore to before you did the update and repeat the process but then (before restart of HA) enabled debug logging, restarted HA and share the debug log w/ migrations. But only really if you have the time, no pressure. 😉