davidrapan/ha-solarman #310: "Wired" connection support?

voed, 2024-12-09:
### Is your feature request related to a problem?

I have some issues with grabbing data from the stick, so i decided to connect the inverter via rs485 port. I am not sure if HA provides access to the serial port for the integrations, but maybe at least we can make it work via [mbusd](https://github.com/kellerza/sunsynk/tree/main/hass-addon-mbusd)?
For now i am getting this error, seems like the integration completely ignores ip and obtaining logger's ip by serial number.
[image]
[image]

### Describe the solution you'd like

Provide ability to connect by TCP to Modbus Gateway

### Describe alternatives you've considered

Dunno

voed, 2024-12-09:
Did not helped much, now i am getting "Device loading failed [TimeoutError]".
I see that it tries to connect to the gateway, so looks like there are some protocol issues. Maybe you can learn further how kellerza's addon works? As it's using pysolarman too i think there's should be a way to make it work.

davidrapan, 2024-12-09:
Show me the debug log. 😉

> As it's using pysolarman too

No, it does not use pysolarmanv5.

voed, 2024-12-09:
Oh, my bad, i guess it is using pysolarman only for connection via stick, not modbus.

Logger: homeassistant
Source: runner.py:147
First occurred: 22:37:27 (13 occurrences)
Last logged: 22:57:38

Error doing job: Task was destroyed but it is pending! (<Task pending name='ConnKeeper' coro=<PySolarmanV5Async._conn_keeper() running at /usr/local/lib/python3.12/site-packages/pysolarmanv5/pysolarmanv5_async.py:172> wait_for=<Future pending cb=[Task.task_wakeup()]>>)
That's all what i have in the logs.

davidrapan, 2024-12-09:
You need to enable debug logging.

voed, 2024-12-09:
```
2024-12-09 23:17:27.233 DEBUG (MainThread) [custom_components.solarman.api] [<id>] Auto reconnect enabled. Will try to restart the socket reader
2024-12-09 23:17:27.234 DEBUG (MainThread) [custom_components.solarman.api] [<id>] Successful reconnect
2024-12-09 23:17:27.234 DEBUG (MainThread) [custom_components.solarman.api] [<id>] Data expected. Will retry the last request
2024-12-09 23:17:27.235 DEBUG (MainThread) [custom_components.solarman.api] [<id>] Connection reset. Closing the socket reader.
Traceback (most recent call last):
  File "/usr/local/lib/python3.12/site-packages/pysolarmanv5/pysolarmanv5_async.py", line 172, in _conn_keeper
    data = await self.reader.read(1024)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/asyncio/streams.py", line 713, in read
    await self._wait_for_data('read')
  File "/usr/local/lib/python3.12/asyncio/streams.py", line 545, in _wait_for_data
    await self._waiter
  File "/usr/local/lib/python3.12/asyncio/selector_events.py", line 1003, in _read_ready__data_received
    data = self._sock.recv(self.max_size)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
ConnectionResetError: [Errno 104] Connection reset by peer
```
Oh, and if forgot to mention that addon is using inverter SN for connection, not the stick SN

davidrapan, 2024-12-09:
That's all? If so then you are not listening on the port. 😉

> Oh, and if forgot to mention that addon is using inverter SN for connection, not the stick SN

Does not matter if it talks modbus tcp... it does not use serial at all.

voed, 2024-12-09:
But i am, it connects fine from the addon. 
Here's what i see in the mbusd logs:
```
09 Dec 2024 23:43:12 conn_open(): accepting connection from 192.168.1.101
09 Dec 2024 23:43:17 conn[192.168.1.101]: timeout, closing connection
09 Dec 2024 23:43:17 conn_close(): closing connection from 192.168.1.101
```

davidrapan, 2024-12-09:
Try:
https://github.com/davidrapan/ha-solarman/blob/4f20b5c27d07469e103186093974dd1449f3e322/custom_components/solarman/api.py#L47
switch it to True.

And give me whole log file, because if it listen it has to at least connect and send some data. And I can't see that it is doing just from log snippets. 😉

voed, 2024-12-09:
Here, check please.
[home-assistant_solarman_2024-12-09T22-19-48.020Z.log](https://github.com/user-attachments/files/18068787/home-assistant_solarman_2024-12-09T22-19-48.020Z.log)

davidrapan, 2024-12-09:
So there is communication... you did not see it? It even successfully reads registers 0 - 22. It fails afterwards when it requests registers 0 - 112. Doesn't have that gateway or anything along the way some limitation how big the request can be?

davidrapan, 2024-12-09:
* Don't put serial number of the stick into config, when you don't want to talk with the stick.
  * This will also disable that ip enforce, because no one will respond.
* Show me if it also works w/ (set to False)
  * https://github.com/davidrapan/ha-solarman/blob/4f20b5c27d07469e103186093974dd1449f3e322/custom_components/solarman/api.py#L47

Then find out if the requests are not too big. Copy deye_string.yaml under custom folder and add to default section:
```
default:
    max_size: 20
```

davidrapan, 2024-12-09:
More info about the exception from mbusd:
**GATEWAY TARGET DEVICE FAILED TO RESPOND**:
* Specialized use in conjunction with gateways indicates that no response was obtained from the target device. This usually means that the device is not present on the network.

voed, 2024-12-09:
[image]
But the SN field is required. I am used serial number of the inverter.

voed, 2024-12-09:
Oh, yes, initially i forgot about that and used SN of stick.
I will check the rest tomorrow, thanks for help.

davidrapan, 2024-12-09:
Anyway, it looks like there won't be any changes necessary for it to work.

davidrapan, 2024-12-10:
Anyway can you show me debug log file with passthrough = False? Maybe I could make it work through autodetection. Depends on how it responds.

voed, 2024-12-10:
> Maybe because you have string inverter?
No, i have SG05LP01 6kw 😁
With connection by stick it shows correctly. But via modbus it shows string inverter. Also battery temperature is incorrect(-85 °C)
[image]

voed, 2024-12-10:
[Uploading home-assistant_solarman_2024-12-10T19-10-54.701Z.log…]()
Here's the log with passthrough=false

davidrapan, 2024-12-10:
Can you then show me your log when you do connection via stick normally so I can compare it?

voed, 2024-12-10:
I will check it again tomorrow.

davidrapan, 2024-12-10:
> Here's the log with passthrough=false

Will you upload the file?

voed, 2024-12-10:
Yes, i will do that tomorrow, too sleepy already:)

rtripon, 2024-12-11:
Hello, sorry for busting here :), but I`m interested in this topic also. 
I have an ESPHome board that talks with inverter via RS485, your work here means its possible to integrate that IP into solarman?.
Currently Solarman is getting data from wifi dongle sticked into inverter, but it would be nice if it could get data from my ESPHome board also.

voed, 2024-12-11:
Hi. As far as i know ESPHome does not have RTU to TCP proxy. But you should be able to do that by something like [this](/harihanv/esp32-modbus-gateway)

neofral, 2024-12-11:
@voed 
are you talking about this type of device?
https://esphome.io/components/modbus_controller.html

rtripon, 2024-12-11:
Yes, that's it.
I already have the data comming in via modbus_controller from my ESPHome board into HA, but in Solarman Application, there are custom sensors that I like, and those are missing from my ESPHome board.

neofral, 2024-12-11:
you can add custom sensors in your configuration.yaml.
can use the sensors provided by the board as input sensor.
or, you can also use HA UI to create helpers to help you create many types of custom sensors

voed, 2024-12-11:
No, i mean a gateway that will relay modbus to the lan network so solarman can read it.

neofral, 2024-12-11:
the simplest way i know is to make that sensor in configuration.yaml.
something like this:

template:
  - sensor:
      - name: Total PV Power
        unit_of_measurement: "W"
        state_class: measurement
        device_class: energy
        icon: mdi:solar-power
        state: >-
          {{ (states('${device_type}-PV1 Power')) |float |round(2) + (states('${device_type}-PV1 Power')) |float |round(2) }}

in this case i don't really know how the substitutes will work, you want to use actually sensor name.
give it a try and you'll figure it out. check for logs for any errors in your configuration.

voed, 2024-12-11:
Guys it's not related to this isue:)

voed, 2024-12-11:
@davidrapan mbusd gateway connection with passthrough = False
https://drive.google.com/file/d/18hIEncrcsFu34PlnbKFCkNGBSuEVAef7/view

Stick connection with passthrough = False
[home-assistant_solarman_2024-12-11T12-00-38.589Z.log](https://github.com/user-attachments/files/18095019/home-assistant_solarman_2024-12-11T12-00-38.589Z.log)

davidrapan, 2024-12-11:
> Stick connection with passthrough = False

Unfortunately there isn't successful connection logged.

voed, 2024-12-11:
Not sure what do you mean, the first one was unable to connect because it needs passthrough i guess, and the second one is a successful connection and still receiving data from the inverter.

davidrapan, 2024-12-11:
> because it needs passthrough i guess

No, this can't affect initialization of the connection. This switch affects later sending of the data.

davidrapan, 2024-12-11:
Anyway, this is really strange, reading of register 0 in standard mode results in device type detection as 0x3, which is hybrid inverter. And reding over that gateway results in 0x0200, which is string inverter. 😆 Someone is lying here! 😆

voed, 2024-12-11:
> You did not wait long enough for the file to upload. 😉

Just now realized it lol. I thought you are saying that i am colleted not much enough data. Already deleted that log but i am sure it was the same that i have posted today: "Connection reset by peer" and nothing else

davidrapan, 2024-12-11:
With "Connection reset by peer" it means that it did not even get to the point where it should use that passthrough switch.

voed, 2024-12-11:
I got it. No idea how passthrough affects that, but it is what it is:)

davidrapan, 2024-12-11:
Just copy over deye_hybrid.yaml under custom folder and configure device using that copy of the file (when connecting through that gateway).

voed, 2024-12-11:
Um, but i am using copy of deye_hybrid profile now🤔

davidrapan, 2024-12-11:
And where is that file located?

voed, 2024-12-11:
Looks like 39 is a maximum.

voed, 2024-12-13:
Also seems like the battery temperature register provides valid data, but without offset. Removed `offset:1000` and values looks valid:
[image]

davidrapan, 2024-12-13:
You should ask the author of that gateway bc it looks like it is changing a lot of things. 😉

voed, 2024-12-13:
It's fine, just saying.
So, are you going to implement that passthrough option or auto-detect for this purpose?

davidrapan, 2024-12-13:
I would argue that changing device type from hybrid to string is not really fine. 😆

voed, 2024-12-13:
In kellerza's addon this value refers to "inverter", so maybe it just means any inverter, not string inverter:)
[image]

Everything else including SN, rated power etc is okay. So yes, i think it's fine
[image]

davidrapan, 2024-12-13:
That means string inverter, every non-hybrid inverter is string inverter.

But the most important part is that it does not even report 2 but 512.

voed, 2024-12-13:
Which is 0x200 in hexadecimal
[image]

davidrapan, 2024-12-13:
Yes, and? I know why I have it there but I was referring to your example from that addon. 😆

as-cmi, 2025-01-03:
Hi guys, following this thread i was able to connect fine to my deye 3ksg03lp1 via a waveshare modbus-tcp interface.

Now I updated to 24.12.22 and api.py is deeply modified, so is not possible anymore to tweek "self._passthrough = True" as suggested:

> Try:
>
> https://github.com/davidrapan/ha-solarman/blob/4f20b5c27d07469e103186093974dd1449f3e322/custom_components/solarman/api.py#L47
>
>
> switch it to True.
> And give me whole log file, because if it [...]

Any help?
Thanks

voed, 2025-01-03:
I guess it's moved to pysolarman dependency
https://github.com/davidrapan/pysolarmanv5/blob/93f21e59f84be98e164b10b8855750db6a30c2f5/pysolarmanv5/pysolarman.py#L42

davidrapan, 2025-01-04:
It's now in `include/pysolarmanv5/pysolarman.py`

as-cmi, 2025-01-04:
It works, thank you!

davidrapan, 2025-01-13:
Latest version of the code contains this change:
```
self._passthrough = serial <= 0
```
which means that setting serial number to 0 will activate passthrough mode. 😉

voed, 2025-02-09:
@davidrapan here we go again:) Can you reopen this or should i create a new issue?
```
2025-02-10 00:42:05.161 DEBUG (MainThread) [custom_components.solarman] async_setup_entry({'created_at': '2024-12-15T16:48:01.693763+00:00', 'data': {'name': 'Inverter', 'serial': <inverter SN>}, 'discovery_keys': {}, 'disabled_by': None, 'domain': 'solarman', 'entry_id': '01JF5KGG4XSBJN73GYWMHNT824', 'minor_version': 6, 'modified_at': '2025-02-09T22:41:48.892425+00:00', 'options': {'host': '192.168.1.101', 'port': 502, 'lookup_file': 'deye_hybrid.yaml', 'additional_options': {'mppt': 2, 'phase': 1}}, 'pref_disable_new_entities': False, 'pref_disable_polling': False, 'source': 'user', 'title': 'Inverter', 'unique_id': 'solarman_<inverter SN>', 'version': 1})
2025-02-10 00:42:05.161 DEBUG (MainThread) [custom_components.solarman] async_setup_entry: config_entry.runtime_data.async_config_entry_first_refresh
2025-02-10 00:42:05.162 DEBUG (MainThread) [custom_components.solarman.discovery] _discover: Broadcasting on 192.168.1.101
2025-02-10 00:42:05.163 DEBUG (MainThread) [custom_components.solarman.discovery] DiscoveryProtocol: Send to 192.168.1.101
2025-02-10 00:42:05.665 DEBUG (MainThread) [custom_components.solarman.discovery] _discover_all: Broadcasting on [IPv4Network('192.168.1.0/24'), IPv4Network('172.30.32.0/23'), IPv4Network('172.30.232.0/23')]
2025-02-10 00:42:05.666 DEBUG (MainThread) [custom_components.solarman.discovery] DiscoveryProtocol: Connection closed
2025-02-10 00:42:05.667 DEBUG (MainThread) [custom_components.solarman.discovery] DiscoveryProtocol: Send to ['192.168.1.255', '172.30.33.255', '172.30.233.255']
2025-02-10 00:42:05.784 DEBUG (MainThread) [custom_components.solarman.discovery] DiscoveryProtocol: [192.168.1.65, D4278711BBDA, <>]
2025-02-10 00:42:06.287 DEBUG (MainThread) [custom_components.solarman.provider] Device autodetection is enabled
2025-02-10 00:42:06.288 DEBUG (MainThread) [custom_components.solarman.device] [<inverter SN>] Scheduling 1 query request. ^-1
2025-02-10 00:42:06.288 DEBUG (MainThread) [custom_components.solarman.device] [<inverter SN>] Querying 03 ❘ 0x03 ~ 0000 - 0022 ❘ 0x0000 - 0x0016 #023 ...
2025-02-10 00:42:06.288 DEBUG (MainThread) [custom_components.solarman.discovery] DiscoveryProtocol: Connection closed
2025-02-10 00:42:06.292 DEBUG (MainThread) [custom_components.solarman.pysolarman.pysolarman] [<inverter SN>] SENT: a5 17 00 10 45 ef 00 24 ac 6a 8f 02 00 00 00 00 00 00 00 00 00 00 00 00 00 00 01 03 00 00 00 17 05 c4 0a 15
2025-02-10 00:42:06.293 DEBUG (MainThread) [custom_components.solarman.pysolarman.pysolarman] [<inverter SN>] Connection is reset by the peer. Will try to restart the connection
2025-02-10 00:42:06.296 DEBUG (MainThread) [custom_components.solarman.pysolarman.pysolarman] [<inverter SN>] Successful reconnection! Data expected. Will retry the last request
2025-02-10 00:42:06.297 DEBUG (MainThread) [custom_components.solarman.pysolarman.pysolarman] [<inverter SN>] Connection is reset by the peer. Will try to restart the connection
2025-02-10 00:42:06.300 DEBUG (MainThread) [custom_components.solarman.pysolarman.pysolarman] [<inverter SN>] Successful reconnection! Data expected. Will retry the last request
2025-02-10 00:42:06.301 DEBUG (MainThread) [custom_components.solarman.pysolarman.pysolarman] [<inverter SN>] Connection is reset by the peer. Will try to restart the connection
2025-02-10 00:42:06.303 DEBUG (MainThread) [custom_components.solarman.pysolarman.pysolarman] [<inverter SN>] Successful reconnection! Data expected. Will retry the last request
2025-02-10 00:42:06.304 DEBUG (MainThread) [custom_components.solarman.pysolarman.pysolarman] [<inverter SN>] Connection is reset by the peer. Will try to restart the connection
2025-02-10 00:42:06.306 DEBUG (MainThread) [custom_components.solarman.pysolarman.pysolarman] [<inverter SN>] Successful reconnection! Data expected. Will retry the last request
2025-02-10 00:42:06.307 DEBUG (MainThread) [custom_components.solarman.pysolarman.pysolarman] [<inverter SN>] Connection is reset by the peer. Will try to restart the connection
2025-02-10 00:42:06.309 DEBUG (MainThread) [custom_components.solarman.pysolarman.pysolarman] [<inverter SN>] Successful reconnection! Data expected. Will retry the last request
2025-02-10 00:42:06.311 DEBUG (MainThread) [custom_components.solarman.pysolarman.pysolarman] [<inverter SN>] Connection is reset by the peer. Will try to restart the connection
2025-02-10 00:42:06.313 DEBUG (MainThread) [custom_components.solarman.pysolarman.pysolarman] [<inverter SN>] Successful reconnection! Data expected. Will retry the last request
2025-02-10 00:42:06.314 DEBUG (MainThread) [custom_components.solarman.pysolarman.pysolarman] [<inverter SN>] Connection is reset by the peer. Will try to restart the connection
2025-02-10 00:42:06.316 DEBUG (MainThread) [custom_components.solarman.pysolarman.pysolarman] [<inverter SN>] Successful reconnection! Data expected. Will retry the last request
2025-02-10 00:42:06.317 DEBUG (MainThread) [custom_components.solarman.pysolarman.pysolarman] [<inverter SN>] Connection is reset by the peer. Will try to restart the connection
2025-02-10 00:42:06.319 DEBUG (MainThread) [custom_components.solarman.pysolarman.pysolarman] [<inverter SN>] Successful reconnection! Data expected. Will retry the last request
```

> which means that setting serial number to 0 will activate passthrough mode. 😉

Reminding that i cannot set serial number to 0 because it needs inverter(not loger) SN to connect.

davidrapan, 2025-02-09:
Oh yeah, I forgot to allow 0 as serial in the config!

davidrapan, 2025-02-09:
> Reminding that i cannot set serial number to 0 because it needs inverter(not loger) SN to connect.

Are you sure about that?

voed, 2025-02-10:
Yep, without SN i was not able to connect. Same as kellerza's addon:
```
SERIAL_NR

The serial number of your inverter. When you start the add-on the connected serial will be displayed in the log.

The add-on will not run if the expected/configured serial number is not found.
```

davidrapan, 2025-02-10:
But, it’s not using serial number while connecting using modbus TCP..

[1 more reply]