davidrapan/ha-solarman #312: Unable to set up integration for Deye SUN-12K-SG04LP3-EU

rene-neimann, 2024-12-09:
### Description

When trying to set up the integration for my Deye SUN-12K-SG04LP3-EU, the integration fails to load and returns 2 different types of errors:

```
Exception: [OBFUSCATED_SERIAL] Device setup failed. [Exception: [OBFUSCATED_SERIAL] ValueError: The truth value of an array with more than one element is ambiguous. Use a.any() or a.all()]

Config entry 'Deye Hybrid' from integration solarman has an invalid unique_id 'OBFUSCATED_SERIAL' of type int when a string is expected, please create a bug report at https://github.com/davidrapan/ha-solarman/issues
```

I tried different profile YAMLs, same result (Auto, deye_hybrid, deye_p3).

The exact same configuration (IP, port, serial, slave id) works with https://github.com/StephanJoubert/home_assistant_solarman integration.

### Reproduction Steps

1) Install integration
2) Add device
3) Fill in inverter configuration (was already correctly discovered automatically in my case)
4) Integration fails to load

### Log

This error originated from a custom integration.

Logger: custom_components.solarman.coordinator
Source: helpers/update_coordinator.py:318
integration: Solarman (documentation, issues)
First occurred: 23:35:33 (17 occurrences)
Last logged: 23:42:08

Unexpected error fetching Deye Hybrid data
Traceback (most recent call last):
  File "/config/custom_components/solarman/api.py", line 244, in get
    responses[(code, start)] = await self.try_read_write(code, start, quantity, f"Querying {code:02X} ~ {start:04} - {end:04} | 0x{start:04X} - 0x{end:04X} # {quantity:03}", True)
                               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/solarman/api.py", line 203, in try_read_write
    while attempts_left > 0 and not response:
                                    ^^^^^^^^
ValueError: The truth value of an array with more than one element is ambiguous. Use a.any() or a.all()

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/config/custom_components/solarman/api.py", line 140, in load
    self.device_info = await self.profile.resolve(self.get)
                       ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/solarman/provider.py", line 120, in resolve
    f = lookup_profile(await request(-1, [set_request(*AUTODETECTION_REQUEST_DEYE)]), self.attributes) if self.auto and request else self.filename
                       ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/solarman/api.py", line 258, in get
    raise Exception(f"[{self.config.serial}] {format_exception(e)}") from e
Exception: [2985518877] ValueError: The truth value of an array with more than one element is ambiguous. Use a.any() or a.all()

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/update_coordinator.py", line 318, in __wrap_async_setup
    await self._async_setup()
  File "/usr/src/homeassistant/homeassistant/helpers/update_coordinator.py", line 351, in _async_setup
    return await self.setup_method()
           ^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/solarman/api.py", line 143, in load
    raise Exception(f"[{self.config.serial}] Device setup failed. [{format_exception(e)}]") from e
Exception: [OBFUSCATED_SERIAL] Device setup failed. [Exception: [OBFUSCATED_SERIAL] ValueError: The truth value of an array with more than one element is ambiguous. Use a.any() or a.all()]

#########################################

Logger: homeassistant.config_entries
Source: config_entries.py:1663
First occurred: 23:35:33 (2 occurrences)
Last logged: 23:35:33

Config entry 'Deye Hybrid' from integration solarman has an invalid unique_id 'OBFUSCATED_SERIAL' of type int when a string is expected, please create a bug report at https://github.com/davidrapan/ha-solarman/issues

### Version

24.12.08_10

### Home Assistant Version

2024.12.1

davidrapan, 2024-12-09:
Hi, @rene-neimann, strange I have the same device and had no issue but try [v24.12.08_11](https://github.com/davidrapan/ha-solarman/releases/tag/v24.12.08_11) and let me know.

rene-neimann, 2024-12-10:
> Hi, @rene-neimann, strange I have the same device and had no issue but try [v24.12.08_11](https://github.com/davidrapan/ha-solarman/releases/tag/v24.12.08_11) and let me know.

Thanks a lot, this fix solved my problem and I'm now able to integrate my inverter. However, your integration seems to be suffering from the same issue as the other Solarman integration -- some values showing incorrect and unrealistic values:

[image]

Has this issue been reported already for should I create a new issue thread?

davidrapan, 2024-12-10:
Can you show me screenshot of Device attributes like  can be seen here? https://github.com/davidrapan/ha-solarman/issues/281#issuecomment-2520017035

rene-neimann, 2024-12-10:
> Can you show me screenshot of Device attributes like can be seen here? [#281 (comment)](https://github.com/davidrapan/ha-solarman/issues/281#issuecomment-2520017035)

[image]

davidrapan, 2024-12-10:
Could you also give me a reading using Developer tools >> ACTIONS >> Solarman: Read Holding Registers (Modbus Function Code 3) w/ register: 0, quantity: 22.

rene-neimann, 2024-12-10:
> Could you also give me a reading using Developer tools >> ACTIONS >> Solarman: Read Holding Registers (Modbus Function Code 3) w/ register: 0, quantity: 22.

Returns:
1) UI error "Failed to perform the action solarman.read_holding_registers. Unknown error"
2) Logs - 2 different errors, see below

[image]

`
```
Logger: homeassistant.components.websocket_api.http.connection
Source: custom_components/solarman/services.py:66
integration: Home Assistant WebSocket API (documentation, issues)
First occurred: 20:15:42 (11 occurrences)
Last logged: 21:24:40

[139926583213888] Error handling message: Unknown error (unknown_error) René from 192.168.0.126 (Mozilla/5.0 (Linux; Android 14; CPH2449 Build/TP1A.220905.001; wv) AppleWebKit/537.36 (KHTML, like Gecko) Version/4.0 Chrome/131.0.6778.104 Mobile Safari/537.36 Home Assistant/2024.12.1-14604 (Android 14; CPH2449))
[139926417040880] Error handling message: Unknown error (unknown_error) René from 104.28.233.138 (Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/131.0.0.0 Safari/537.36 Edg/131.0.0.0)
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/components/websocket_api/decorators.py", line 28, in _handle_async_response
    await func(hass, connection, msg)
  File "/usr/src/homeassistant/homeassistant/components/websocket_api/commands.py", line 816, in handle_execute_script
    script_result = await script_obj.async_run(
                    ^^^^^^^^^^^^^^^^^^^^^^^^^^^
        msg.get("variables"), context=context
        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
    )
    ^
  File "/usr/src/homeassistant/homeassistant/helpers/script.py", line 1801, in async_run
    return await asyncio.shield(create_eager_task(run.async_run()))
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/script.py", line 464, in async_run
    await self._async_step(log_exceptions=False)
  File "/usr/src/homeassistant/homeassistant/helpers/script.py", line 528, in _async_step
    self._handle_exception(
    ~~~~~~~~~~~~~~~~~~~~~~^
        ex, continue_on_error, self._log_exceptions or log_exceptions
        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
    )
    ^
  File "/usr/src/homeassistant/homeassistant/helpers/script.py", line 558, in _handle_exception
    raise exception
  File "/usr/src/homeassistant/homeassistant/helpers/script.py", line 526, in _async_step
    await getattr(self, handler)()
  File "/usr/src/homeassistant/homeassistant/helpers/script.py", line 764, in _async_call_service_step
    response_data = await self._async_run_long_action(
                    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
    ...<9 lines>...
    )
    ^
  File "/usr/src/homeassistant/homeassistant/helpers/script.py", line 727, in _async_run_long_action
    return await long_task
           ^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/core.py", line 2802, in async_call
    response_data = await coro
                    ^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/core.py", line 2845, in _execute_service
    return await target(service_call)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/solarman/services.py", line 66, in read_holding_registers
    if response:
       ^^^^^^^^
ValueError: The truth value of an array with more than one element is ambiguous. Use a.any() or a.all()
`

`
Logger: homeassistant.helpers.script.websocket_api_script
Source: helpers/script.py:526
First occurred: 20:15:42 (11 occurrences)
Last logged: 21:24:40

websocket_api script: Error executing script. Unexpected error for call_service at pos 1: The truth value of an array with more than one element is ambiguous. Use a.any() or a.all()
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/script.py", line 526, in _async_step
    await getattr(self, handler)()
  File "/usr/src/homeassistant/homeassistant/helpers/script.py", line 764, in _async_call_service_step
    response_data = await self._async_run_long_action(
                    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
    ...<9 lines>...
    )
    ^
  File "/usr/src/homeassistant/homeassistant/helpers/script.py", line 727, in _async_run_long_action
    return await long_task
           ^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/core.py", line 2802, in async_call
    response_data = await coro
                    ^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/core.py", line 2845, in _execute_service
    return await target(service_call)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/solarman/services.py", line 66, in read_holding_registers
    if response:
       ^^^^^^^^
ValueError: The truth value of an array with more than one element is ambiguous. Use a.any() or a.all()
```
`

davidrapan, 2024-12-10:
Fuuu, I wonder what is the deal w/ your environment... anyway: [fix: The truth value of an array is ambiguous](https://github.com/davidrapan/ha-solarman/commit/eb2a1411477f6d0701546dc5b08b30c6d375722b)

Will you be able to replace content of `services.py` w/ [this](https://raw.githubusercontent.com/davidrapan/ha-solarman/refs/heads/main/custom_components/solarman/services.py)?

rene-neimann, 2024-12-10:
> Fuuu, I wonder what is the deal w/ your environment... 

I think the deal is that Docker/container HA Core installations are messed up after Core 2024.12 upgrade:
https://community.home-assistant.io/t/wth-is-going-on-with-docker-container-installs-and-third-party-integrations-that-need-python-modules/806430

I appreciate your patience with this! So far my Docker/container installation has worked pretty much flawlessly but it seems they've created a mess now. I've created a completely fresh new container today for testing purposes and it has the same issues with Solarman integration as my few-years-old container.

> anyway: [fix: The truth value of an array is ambiguous](https://github.com/davidrapan/ha-solarman/commit/eb2a1411477f6d0701546dc5b08b30c6d375722b)
> Will you be able to replace content of [...]

Replaced. New error when running register 0, quantity 23:
`Failed to perform the action solarman.read_holding_registers. Invalid JSON in response
`
```
Logger: homeassistant.components.websocket_api.messages
Source: components/websocket_api/messages.py:246
integration: Home Assistant WebSocket API (documentation, issues)
First occurred: 22:38:06 (3 occurrences)
Last logged: 22:40:59

Unable to serialize to JSON. Bad data found at $.result.response.0=5(<class 'numpy.uint16'>, $.result.response.1=1(<class 'numpy.uint16'>, $.result.response.2=260(<class 'numpy.uint16'>, $.result.response.3=12852(<class 'numpy.uint16'>, $.result.response.4=12343(<class 'numpy.uint16'>, $.result.response.5=12596(<class 'numpy.uint16'>, $.result.response.6=13362(<class 'numpy.uint16'>, $.result.response.7=13876(<class 'numpy.uint16'>, $.result.response.8=0(<class 'numpy.uint16'>, $.result.response.9=0(<class 'numpy.uint16'>, $.result.response.10=0(<class 'numpy.uint16'>, $.result.response.11=6151(<class 'numpy.uint16'>, $.result.response.12=53767(<class 'numpy.uint16'>, $.result.response.13=0(<class 'numpy.uint16'>, $.result.response.14=8198(<class 'numpy.uint16'>, $.result.response.15=4423(<class 'numpy.uint16'>, $.result.response.16=0(<class 'numpy.uint16'>, $.result.response.17=4097(<class 'numpy.uint16'>, $.result.response.18=49225(<class 'numpy.uint16'>, $.result.response.19=0(<class 'numpy.uint16'>, $.result.response.20=54464(<class 'numpy.uint16'>, $.result.response.21=1(<class 'numpy.uint16'>, $.result.response.22=65535(<class 'numpy.uint16'>

```

davidrapan, 2024-12-10:
I'm gonna need to see that error w/ debug log enabled so I can locate the origin of it.

rene-neimann, 2024-12-10:
> I'm gonna need to see that error w/ debug log enabled so I can locate the origin of it.

Hmm, debug log enabled (and HA restarted just in case) but oddly, the level of details seems exactly the same:

```
Logger: homeassistant.components.websocket_api.messages
Source: components/websocket_api/messages.py:246
integration: Home Assistant WebSocket API (documentation, issues)
First occurred: 22:38:06 (6 occurrences)
Last logged: 23:00:28

Unable to serialize to JSON. Bad data found at $.result.response.0=5(<class 'numpy.uint16'>, $.result.response.1=1(<class 'numpy.uint16'>, $.result.response.2=260(<class 'numpy.uint16'>, $.result.response.3=12852(<class 'numpy.uint16'>, $.result.response.4=12343(<class 'numpy.uint16'>, $.result.response.5=12596(<class 'numpy.uint16'>, $.result.response.6=13362(<class 'numpy.uint16'>, $.result.response.7=13876(<class 'numpy.uint16'>, $.result.response.8=0(<class 'numpy.uint16'>, $.result.response.9=0(<class 'numpy.uint16'>, $.result.response.10=0(<class 'numpy.uint16'>, $.result.response.11=6151(<class 'numpy.uint16'>, $.result.response.12=53767(<class 'numpy.uint16'>, $.result.response.13=0(<class 'numpy.uint16'>, $.result.response.14=8198(<class 'numpy.uint16'>, $.result.response.15=4423(<class 'numpy.uint16'>, $.result.response.16=0(<class 'numpy.uint16'>, $.result.response.17=4097(<class 'numpy.uint16'>, $.result.response.18=49225(<class 'numpy.uint16'>, $.result.response.19=0(<class 'numpy.uint16'>, $.result.response.20=54464(<class 'numpy.uint16'>, $.result.response.21=1(<class 'numpy.uint16'>, $.result.response.22=65535(<class 'numpy.uint16'>
```

rene-neimann, 2024-12-10:
Not sure if it helps in any way but I spotted a pattern. The incorrect unrealistic values occur with sensors that can have both, positive and negative values. For example, with my battery power sensor. In the example below, I was:
1) **discharging** at ~11kW around 17:00 -- correct positive value
2) **charging** at ~11kW around 23:00 -- incorrect positive value, should have been -11kW instead of the actual +54kW (seems like 65535 minus 11k)

[image]

[image]

rene-neimann, 2024-12-11:
> @rene-neimann you need to upload a file which is produced during debug disable. HA log won't show debug messages. 😉

Here's the debug log:
[debug.log](https://github.com/user-attachments/files/18093731/debug.log)

> And your observation seems to be correct, however I have the same device so I'm 100% sure that the problem is not w/ the integration... I have to say that I'm really clueless of what could be the culprit but it seems even more strange for it to be HA Core's fault...
> 
> Can't you just try to run some older version maybe?

Yes, I'm considering it. It also seems like the clean Docker Container installation I created behaves better and less Solarman sensors are messed up. So it could be my older container has accumulated some clutter which is causing issues.

closed 2025-01-26 by github-actions[bot]