davidrapan/ha-solarman #315: Setup failed for custom integration 'solarman': Unable to import component: No module named 'pysolarmanv5'

Benni1982, 2024-12-11:
### Description

Just did HA core update to 2024.12.2
After HA restart got message:
Setup failed for custom integration 'solarman': Unable to import component: No module named 'pysolarmanv5'

### Reproduction Steps

-

### Log

[home-assistant_2024-12-11T09-27-15.419Z.log](https://github.com/user-attachments/files/18092677/home-assistant_2024-12-11T09-27-15.419Z.log)

### Version

24.12.08_9

### Home Assistant Version

2024.12.2

rp2019433, 2024-12-11:
Hi.
Same problem here. Ha-solarman is not working after updating HA core to 2024.12.2.

Benni1982, 2024-12-11:
tried restore of automatic backup before core update, error still present!?

rtripon, 2024-12-11:
Weird, I have 2 HA instances(one on proxmox VM and one local on rPI4).
Both setup's are basically the same, same HACS, same Solarman app, etc.
But on one server, the one on proxmox VM I have :
Solarman version : 24.12.10_2 - updated 9 hours ago
And on the rPI4 HA server I have ::
Solarman version : 24.12.08_9 - updated 2 days ago.

Benni1982, 2024-12-11:
On Pi if i click on HACS - Solarman it only displays me 24.12.08_09
[image]
How to force download of newer one?!

rtripon, 2024-12-11:
Hit the 3 dots on the right and "Redownload".

closed 2024-12-11 by Benni1982