davidrapan/ha-solarman #316: Invalid value active power after update HA core 2024.12

Pstart, 2024-12-11:
### Description

I had integration Solarman StephanJoubert but update core 2024.12 entered incorrect values many entity. I changed for new integration ha-solarman. Most of the values is ok but active_energe and battery power is incorect for more time. More time returning value is 65kW, like 
ActivePower_Output_R 64,526 W
ActivePower_Output_S 910 W
ActivePower_Output_T 64,876 W
ActivePower_Output_Total 64,766 W
ActivePower_PCC_R 64,406 W
ActivePower_PCC_S 64,456 W
ActivePower_PCC_T 64,586 W
ActivePower_PCC_Total 62,356 W
Details is chart in attachment.
My max use energy is about 8kW
My profile is sofar_g3hyd.yaml
[image]

### Reproduction Steps

I only use integration. Value active power is invalid.

### Log

[33m2024-12-11 20:38:15.422 WARNING (MainThread) [py.warnings] /config/custom_components/solarman/parser.py:226: RuntimeWarning: overflow encountered in scalar subtract
  value = (value - maxint) if not magnitude else -(value & (maxint >> 1))
[0m
[33m2024-12-11 20:38:15.423 WARNING (MainThread) [py.warnings] /config/custom_components/solarman/parser.py:229: RuntimeWarning: overflow encountered in scalar multiply
  value *= scale
[0m
[33m2024-12-11 20:38:41.460 WARNING (MainThread) [py.warnings] /config/custom_components/solarman/parser.py:226: RuntimeWarning: overflow encountered in scalar subtract
  value = (value - maxint) if not magnitude else -(value & (maxint >> 1))
[0m
[33m2024-12-11 20:38:41.460 WARNING (MainThread) [py.warnings] /config/custom_components/solarman/parser.py:229: RuntimeWarning: overflow encountered in scalar multiply
  value *= scale

### Version

24.12.10

### Home Assistant Version

2024.12.2

davidrapan, 2024-12-11:
Hi @Pstart, you are running containerized home assistant?

Pstart, 2024-12-12:
Hi, no i use Assistant OS on x86-64 hardware

davidrapan, 2024-12-12:
Did you try downgrade of the core (to some known to be working) to see of the problem prevails?

Pstart, 2024-12-12:
@davidrapan Yes, I downgraded version to 2024.11.3 that's all working good.  But when ha-sofarman well work good on 2024.12?

davidrapan, 2024-12-13:
Once I figure out what's wrong, because right now I don't have even slight idea...

werfpsa, 2024-12-14:
Same problem with inverter file sofar_g3hyd.yaml

DawidJan, 2024-12-16:
I also have the same issue with sofar_g3hyd.yaml file

davidrapan, 2024-12-16:
Can someone record 10 minute debug log?

davidreher, 2024-12-17:
@davidrapan this is the log from HA:
```
2024-12-16 20:31:55.636 ERROR (MainThread) [homeassistant.config_entries] Error setting up entry Inverter for solarman
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/config_entries.py", line 640, in __async_setup_with_context
    result = await component.async_setup_entry(hass, self)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/solarman/__init__.py", line 59, in async_setup_entry
    await async_migrate_entries(hass, config_entry.entry_id, partial(migrate_unique_ids, config.name, config.serial))
  File "/usr/src/homeassistant/homeassistant/helpers/entity_registry.py", line 1636, in async_migrate_entries
    ent_reg.async_update_entity(entry.entity_id, **updates)
    ~~~~~~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/entity_registry.py", line 1179, in async_update_entity
    return self._async_update_entity(
           ~~~~~~~~~~~~~~~~~~~~~~~~~^
        entity_id,
        ^^^^^^^^^^
    ...<21 lines>...
        unit_of_measurement=unit_of_measurement,
        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
    )
    ^
  File "/usr/src/homeassistant/homeassistant/helpers/entity_registry.py", line 1118, in _async_update_entity
    raise ValueError(
    ...<2 lines>...
    )
```

How can I create the debug log? Happy to provide it ...

davidrapan, 2024-12-17:
https://www.home-assistant.io/docs/configuration/troubleshooting/#enabling-debug-logging

davidreher, 2024-12-18:
Sent it to you via mail (hope that is fine) ...

mariomurrent-softwaresolutions, 2024-12-19:
> @davidrapan this is the log from HA:
>
> ```
> 2024-12-16 20:31:55.636 ERROR (MainThread) [homeassistant.config_entries] Error setting up entry Inverter for solarman
> Traceback (most recent call last): [...]

Having the same issue :(

davidrapan, 2024-12-23:
I did not receive any debug logs. Please upload it here in the thread. 😉

maconly34, 2024-12-30:
Hello David,
I have similar error with Deye Sun inverter 6k SG03LP1.
Grid Power, Load Power etc. have a x10 factor shown.
Values is ok on iPhone Solarman App.

See screen copy.

[image]

[image]

[image]

maconly34, 2025-01-01:
Hi David,
I've managed to correct the problem by modifying the code in the "deye_hybrid.yaml" file
There are 9 lines (# out of date docs) where the scale function is "scale: [1, 10]" I've replaced it with "scale: 1".

part of the code corrected :
- name: "External CT1 Power"
        l: 2
        class: "power"
        state_class: "measurement"
        uom: "W"
        rule: 2
        scale: 1 # out of date docs
        registers: [0x00AA]
        icon: "mdi:transmission-tower"

      - name: "External CT2 Power"
        l: 2
        class: "power"
        state_class: "measurement"
        uom: "W"
        rule: 2
        scale: 1 # out of date docs
        registers: [0x00AB]
        icon: "mdi:transmission-tower"

      - name: "External Power"
        l: 1
        class: "power"
        state_class: "measurement"
        uom: "W"
        rule: 2
        scale: 1 # out of date docs
        registers: [0x00AC]
        icon: "mdi:transmission-tower"

davidrapan, 2025-01-01:
@maconly34, it's there for a reason. You proly checked scale modifier in the options. 😉

davidreher, 2025-01-06:
Sorry, was a bit busy over xmas ... I also updated to 2025.01 already, here are the necessary parts from the debug logs:
```log
2025-01-06 07:20:03.990 ERROR (MainThread) [homeassistant.config_entries] Error setting up entry Inverter for solarman
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/config_entries.py", line 640, in __async_setup_with_context
    result = await component.async_setup_entry(hass, self)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/solarman/__init__.py", line 59, in async_setup_entry
    await async_migrate_entries(hass, config_entry.entry_id, partial(migrate_unique_ids, config.name, config.serial))
  File "/usr/src/homeassistant/homeassistant/helpers/entity_registry.py", line 1636, in async_migrate_entries
    ent_reg.async_update_entity(entry.entity_id, **updates)
    ~~~~~~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/entity_registry.py", line 1179, in async_update_entity
    return self._async_update_entity(
           ~~~~~~~~~~~~~~~~~~~~~~~~~^
        entity_id,
        ^^^^^^^^^^
    ...<21 lines>...
        unit_of_measurement=unit_of_measurement,
        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
    )
    ^
  File "/usr/src/homeassistant/homeassistant/helpers/entity_registry.py", line 1118, in _async_update_entity
    raise ValueError(
    ...<2 lines>...
    )
```

and then I tried to change the config to sofar_hyd-es.yaml and back to sofar_g3hyd.yaml, which resulted in the following output: 
```
2025-01-06 13:50:16.550 DEBUG (MainThread) [custom_components.solarman.config_flow] ConfigFlowHandler.async_get_options_flow: <ConfigEntry entry_id=01J6HE6KWV1C1TCW7KWWF1WJWR version=1 domain=solarman title=Inverter state=ConfigEntryState.SETUP_ERROR unique_id=None>
2025-01-06 13:50:16.551 DEBUG (MainThread) [custom_components.solarman.config_flow] OptionsFlowHandler.__init__: <ConfigEntry entry_id=01J6HE6KWV1C1TCW7KWWF1WJWR version=1 domain=solarman title=Inverter state=ConfigEntryState.SETUP_ERROR unique_id=None>
2025-01-06 13:50:16.551 DEBUG (MainThread) [custom_components.solarman.config_flow] OptionsFlowHandler.async_step_init: user_input: None, current: {'additional_options': {'battery_nominal_voltage': 400}, 'host': '172.16.43.200', 'lookup_file': 'sofar_g3hyd.yaml'}
2025-01-06 13:50:16.555 DEBUG (MainThread) [custom_components.solarman.config_flow] step_user_data_schema: custom_components/solarman/inverter_definitions/: ['Auto', 'afore_2mppt.yaml', 'afore_BNTxxxKTL-2mppt.yaml', 'afore_hybrid.yaml', 'chint_cps-scetl.yaml', 'deye_hybrid.yaml', 'deye_micro.yaml', 'deye_p3.yaml', 'deye_string.yaml', 'invt_xd-tl.yaml', 'kstar_hybrid.yaml', 'megarevo_r-3h.yaml', 'pylontech_Force-H+.yaml', 'pylontech_Force-H.yaml', 'renon_ifl.yaml', 'sofar_g3hyd.yaml', 'sofar_hyd-es.yaml', 'sofar_lsw3.yaml', 'sofar_tlx-g3.yaml', 'sofar_wifikit.yaml', 'solarman_dtsd422-d3.yaml', 'solis_1p-5g.yaml', 'solis_3p-4g+.yaml', 'solis_3p-4g.yaml', 'solis_3p-5g.yaml', 'solis_hybrid.yaml', 'solis_s6-gr1p.yaml', 'swatten_sih-th.yaml', 'tsun_tsol-ms.yaml', 'zcs_azzurro-1ph-tl-v3.yaml', 'zcs_azzurro-hyd-zss-hp.yaml', 'zcs_azzurro-ktl-v3.yaml']
2025-01-06 13:50:16.556 DEBUG (MainThread) [custom_components.solarman.config_flow] step_user_data_schema: data_schema: {'host': <class 'str'>, 'port': All(Coerce(int, msg=None), Range(min=1, max=65535, min_included=True, max_included=True, msg=None), msg=None), 'lookup_file': In(['Auto', 'afore_2mppt.yaml', 'afore_BNTxxxKTL-2mppt.yaml', 'afore_hybrid.yaml', 'chint_cps-scetl.yaml', 'deye_hybrid.yaml', 'deye_micro.yaml', 'deye_p3.yaml', 'deye_string.yaml', 'invt_xd-tl.yaml', 'kstar_hybrid.yaml', 'megarevo_r-3h.yaml', 'pylontech_Force-H+.yaml', 'pylontech_Force-H.yaml', 'renon_ifl.yaml', 'sofar_g3hyd.yaml', 'sofar_hyd-es.yaml', 'sofar_lsw3.yaml', 'sofar_tlx-g3.yaml', 'sofar_wifikit.yaml', 'solarman_dtsd422-d3.yaml', 'solis_1p-5g.yaml', 'solis_3p-4g+.yaml', 'solis_3p-4g.yaml', 'solis_3p-5g.yaml', 'solis_hybrid.yaml', 'solis_s6-gr1p.yaml', 'swatten_sih-th.yaml', 'tsun_tsol-ms.yaml', 'zcs_azzurro-1ph-tl-v3.yaml', 'zcs_azzurro-hyd-zss-hp.yaml', 'zcs_azzurro-ktl-v3.yaml']), 'additional_options': <homeassistant.data_entry_flow.section object at 0x7f52fcaa50>}
2025-01-06 13:50:19.287 WARNING (MainThread) [bosch_thermostat_client.sensors.sensor] Can't update data for Actual supply temperature for HC. Trying uri: /heatingCircuits/hc2/actualSupplyTemperature. Error message: URI /heatingCircuits/hc2/actualSupplyTemperature doesn not exist: 404, message='Not Found', url='http://172.16.43.14/heatingCircuits/hc2/actualSupplyTemperature'
2025-01-06 13:50:25.399 DEBUG (MainThread) [custom_components.solarman.config_flow] OptionsFlowHandler.async_step_init: user_input: {'host': '172.16.43.200', 'port': 8899, 'lookup_file': 'sofar_hyd-es.yaml', 'additional_options': {'battery_nominal_voltage': 400, 'mod': False, 'phase': 3, 'mppt': 4, 'battery_life_cycle_rating': 6000, 'mb_slave_id': 1}}, current: {'additional_options': {'battery_nominal_voltage': 400}, 'host': '172.16.43.200', 'lookup_file': 'sofar_g3hyd.yaml'}
2025-01-06 13:50:25.400 DEBUG (MainThread) [custom_components.solarman.config_flow] validate_connection: {'host': '172.16.43.200', 'port': 8899, 'lookup_file': 'sofar_hyd-es.yaml', 'additional_options': {'battery_nominal_voltage': 400, 'mod': False, 'phase': 3, 'mppt': 4, 'battery_life_cycle_rating': 6000, 'mb_slave_id': 1}}
[... cut ...]
2025-01-06 13:53:20.320 DEBUG (MainThread) [custom_components.solarman.config_flow] validate_connection: {'host': '172.16.43.200', 'port': 8899, 'lookup_file': 'sofar_g3hyd.yaml', 'additional_options': {'battery_nominal_voltage': 400, 'mod': False, 'phase': 3, 'mppt': 4, 'battery_life_cycle_rating': 6000, 'mb_slave_id': 1}}
2025-01-06 13:53:20.320 DEBUG (MainThread) [custom_components.solarman.config_flow] validate_connection: validation passed: {'host': '172.16.43.200', 'port': 8899, 'lookup_file': 'sofar_g3hyd.yaml', 'additional_options': {'battery_nominal_voltage': 400, 'mod': False, 'phase': 3, 'mppt': 4, 'battery_life_cycle_rating': 6000, 'mb_slave_id': 1}}
```

davidrapan, 2025-01-06:
That first part is not full error and in this state I have no chance to find out what's the error is about.

davidreher, 2025-01-06:
You are right, I missed a crucial part in the first log: `ValueError: Unique id 'inverter_2714000940_ambient_temperature_sensor' is already in use by 'sensor.inverter_ambient_temperature'` ... sorry.

davidrapan, 2025-01-06:
That should not happen. You can fix that by removing and re-adding device back.

davidreher, 2025-01-06:
That did solve it for me 🎉

Pstart, 2025-01-08:
@davidreher attachment it is my debug log. Now I have 
Core 2025.1.1
Supervisor 2024.12.3
Operating System 14.1
There are always invalid values:
[home-assistant_2025-01-08T10-45-20.455Z.log](https://github.com/user-attachments/files/18346219/home-assistant_2025-01-08T10-45-20.455Z.log)

ActivePower_PCC_R - 65,496 W
ActivePower_PCC_S - 65,496 W
ActivePower_PCC_T - 65,506 W
ActivePower_PCC_Total - 65,406 W

Pstart, 2025-01-08:
Hi @davidrapan, attached log is good?

Pstart, 2025-01-17:
Hi @davidrapan , all is latest version
Home Assistant

Core 2025.1.2
Supervisor 2024.12.3
Operating System 14.1
Frontend 20250109.0
Solarman 24.12.22

[home-assistant_solarman_2025-01-17T11-25-22.125Z.log](https://github.com/user-attachments/files/18453799/home-assistant_solarman_2025-01-17T11-25-22.125Z.log)

davidrapan, 2025-01-17:
F*ck, there is an error w/ if statement somewhere
```
ValueError: The truth value of an array with more than one element is ambiguous. Use a.any() or a.all()
```
which prevents most of the requests from completion.

**But it has nothing to do w/ the original issue though.**

davidrapan, 2025-01-26:
[fix: ValueError: The truth value of an array with more than one element is ambiguous](https://github.com/davidrapan/ha-solarman/commit/b570403affac408789779f4cd0c736be3af653da) for the issue above.

And can someone test latest version of the code in the repo (requires manual installation) and change row
https://github.com/davidrapan/ha-solarman/blob/b570403affac408789779f4cd0c736be3af653da/custom_components/solarman/parser.py#L209
to
```
            value += (int(temp) & 0xFFFF) << shift
```
and see if it helps w/ reporting of incorrect values?

Thanks!

davidrapan, 2025-01-27:
Hi @githubDante, would you mind help me understand what's going on?

* They introduced Python 3.13 in HA 2024.12
* Reading of signed int (in some occasions) is now producing RuntimeWarning: overflow encountered in scalar multiply/subtract/...
* int(X) beforehand is possibly solving the issue?

_I don't want to just blatantly fix it and call it a day, but rather understand what they did?_

werfpsa, 2025-01-27:
I can only test the latest version next week when back 'home'. Hope someone else can do before....

githubDante, 2025-01-27:
Hi @davidrapan ,

Is it possible the value to be a numpy `uint` ? I see that numpy is imported in `sensors.py` but is not used anywhere. However, this is the only way which produces similar results

```
$ ./sample_overflow.py 
/tmp/./sample_overflow.py:5: RuntimeWarning: overflow encountered in scalar subtract
  y = x - 10
/tmp/./sample_overflow.py:6: RuntimeWarning: overflow encountered in scalar multiply
  y * 10

```
---
```
$ cat sample_overflow.py 
#!/usr/bin/env python3

import numpy as np
x = np.uint16(5)
y = x - 10
y * 10
```

davidrapan, 2025-01-27:
`import numpy` is just some leftover and not used.

Variable in question is just what's returned from pysolarmav5 and umodbus subsequently w/o anything fancy on top of it. This issue is closely tied w/ python upgrade to 3.13 because downgrade of HA (thus downgrade of python) solves it too.

githubDante, 2025-01-27:
Well, if it's easily reproducible, wrap the block in `catch_warnings`  and log the types of the variables when the warning occurs

```python
#!/usr/bin/env python3

import numpy as np
from warnings import catch_warnings

with catch_warnings(record=True) as w:
    x = np.uint16(5)
    y = x - 10
    y * 10
    if w:
        print(f"Warning spawn from x: {type(x)} - y: {type(y)}")
        for wm in w:
            print(f"{wm.category} <{wm.filename}> [{wm.lineno}]:  {wm.message}")
```
---
Another possibility is this to be a bug in the JIT compiler. The warning is seen in 3 of 5 responses with the data returned for register 4368, but the hex data is identical for all 5 of them.
 I can't see it failing (without numpy) even after a million iterations. The state of the function in `parser.py` (when the warning occurs) should look like this if I'm not wrong:
```python
def try_to_overflow():

    maxint = 0
    value = 0
    shift = 0
    temp = 0x0179
    magnitude = False

    for r in range(1):
        maxint <<= 16
        maxint |= 0xFFFF
        value += (temp & 0xFFFF) << shift
        shift += 16

    value = (value - maxint) if not magnitude else -(value & (maxint >> 1))

for i in range(1_000_000):
    try_to_overflow()
```

davidrapan, 2025-01-27:
Thanks @githubDante!

It's easily reproducible, just not for me... so I hope someone here who has this problem can try out https://github.com/davidrapan/ha-solarman/commit/a97ca498de62965dc053d0e4f6a2b6a78fa43364 (w/ the test code you suggested) from the [dev](https://github.com/davidrapan/ha-solarman/tree/dev) branch and tell us the results. 😉

werfpsa, 2025-02-01:
I can test this on Monday. To my understanding only have to just replace the _read_registers_signed in parser.py and not the change in line 209. Correct?

davidrapan, 2025-02-01:
You need to replace **WHOLE** method not just single line and add that import line to the beginning of the file.

werfpsa, 2025-02-01:
Ok. It's clear

werfpsa, 2025-02-03:
[parser.py.docx](https://github.com/user-attachments/files/18640788/parser.py.docx)

[image]

I have updated HA to os 14.2 and core 2025.1.4
After this updated parser.py with the 2 changes (see docx attachment) and reboot HA.
Unfortunately the issue is still there (see picture).

davidrapan, 2025-02-03:
You misunderstood purpose of this change. It’s not to solve the issue but reveal what’s going in the debug log.

werfpsa, 2025-02-03:
Ok. What do you want me to do next?

davidrapan, 2025-02-03:
Exactly the same thing but enable debug log and attach the file...

werfpsa, 2025-02-03:
[home-assistant_solarman_2025-02-03T14-16-30.344Z.log](https://github.com/user-attachments/files/18643472/home-assistant_solarman_2025-02-03T14-16-30.344Z.log)

davidrapan, 2025-02-03:
Thanks!

So the vals are `<class 'numpy.uint16'>` for some reason **ON THEIR INSTALLATION** (proly all those who experiences the issue). 😕

davidrapan, 2025-02-03:
Can you run it again w/ this change? 5 row from the bottom from
```
_LOGGER.warning(f"Warning spawn w/ value2: {type(value2)} ({value2}), value: {type(value)} ({value}), maxint: {type(maxint)} ({maxint}), scale: {type(scale)} ({scale})")
```
to
```
_LOGGER.warning(f"Warning spawn w/ temp: {type(temp)} ({temp}), value2: {type(value2)} ({value2}), value: {type(value)} ({value}), maxint: {type(maxint)} ({maxint}), scale: {type(scale)} ({scale})")
```

werfpsa, 2025-02-03:
This gives a failed setup, will retry: 0

davidrapan, 2025-02-03:
Debug log, there is proly some error in the new code.

werfpsa, 2025-02-03:
Last logged: 5:31:39 PM

ParameterParser.try_parse: data: {(3, 1028): array([ 2, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 42, 0, 32, 0, 0, 0, 0, 0, 33, 0, 0], dtype='>u2'), (3, 1156): array([ 5002, 367, 65520, 0, 65535, 0, 0, 0, 0, 2394, 518, 121, 0, 0, 347, 65510, 0, 0, 0, 0, 2382, 678, 160, 0, 0, 279, 0, 0, 0, 0, 0, 2406, 367, 85, 0, 0, 303, 25, 0, 0, 0, 0, 0, 364], dtype='>u2'), (3, 1284): array([ 0, 0, 0, 5002, 0, 0, 2381, 18, 0, 0, 0, 0, 2371, 0, 2378, 16, 0, 0, 0, 0, 2356, 0, 2401, 17, 0, 0, 0, 0], dtype='>u2'), (3, 1412): array([1929, 0, 0, 1929, 7, 1], dtype='>u2'), (3, 1540): array([ 3944, 64620, 65175, 28, 80, 100, 173, 0, 0, 0, 0, 0, 0, 0], dtype='>u2'), (3, 1668): array([ 0, 1978, 1, 40038, 0, 1030, 1, 17910, 0, 298, 0, 21711, 0, 190, 0, 61599, 0, 1651, 0, 39040, 0, 364, 0, 37930], dtype='>u2'), (3, 2114): array([2650, 120, 2750, 14, 0, 1], dtype='>u2')}, definition: {'name': 'ReactivePower_Output_Total', 'class': 'power', 'state_class': 'measurement', 'uom': 'W', 'scale': 10, 'rule': 2, 'registers': [1158], 'icon': 'mdi:home-lightning-bolt', 'platform': 'sensor', 'key': 'reactivepower_output_total_sensor', 'code': 3} [KeyError: 0]
ParameterParser.try_parse: data: {(3, 1028): array([ 2, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 42, 0, 32, 0, 0, 0, 0, 0, 33, 0, 0], dtype='>u2'), (3, 1156): array([ 5002, 370, 65515, 0, 65535, 0, 0, 0, 0, 2404, 591, 139, 0, 0, 340, 65513, 0, 0, 0, 0, 2381, 645, 151, 0, 0, 276, 9, 0, 0, 0, 0, 2402, 343, 78, 0, 0, 287, 13, 0, 0, 0, 0, 0, 369], dtype='>u2'), (3, 1284): array([ 0, 0, 0, 5002, 0, 0, 2391, 18, 0, 0, 0, 0, 2381, 0, 2376, 16, 0, 0, 0, 0, 2353, 0, 2398, 16, 0, 0, 0, 0], dtype='>u2'), (3, 1412): array([2131, 0, 0, 1853, 6, 1], dtype='>u2'), (3, 1540): array([ 3943, 64590, 65163, 28, 80, 100, 173, 0, 0, 0, 0, 0, 0, 0], dtype='>u2'), (3, 1668): array([ 0, 1978, 1, 40038, 0, 1030, 1, 17910, 0, 298, 0, 21711, 0, 190, 0, 61599, 0, 1651, 0, 39040, 0, 364, 0, 37930], dtype='>u2'), (3, 2114): array([2650, 120, 2750, 14, 0, 1], dtype='>u2')}, definition: {'name': 'ReactivePower_Output_Total', 'class': 'power', 'state_class': 'measurement', 'uom': 'W', 'scale': 10, 'rule': 2, 'registers': [1158], 'icon': 'mdi:home-lightning-bolt', 'platform': 'sensor', 'key': 'reactivepower_output_total_sensor', 'code': 3} [KeyError: 0]
ParameterParser.try_parse: data: {(3, 1028): array([ 2, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 42, 0, 32, 0, 0, 0, 0, 0, 33, 0, 0], dtype='>u2'), (3, 1156): array([ 5001, 371, 65517, 0, 65533, 0, 0, 0, 0, 2409, 651, 154, 0, 0, 346, 65510, 0, 0, 0, 0, 2374, 583, 136, 0, 0, 280, 6, 0, 0, 0, 0, 2401, 351, 80, 0, 0, 292, 17, 0, 0, 0, 0, 0, 374], dtype='>u2'), (3, 1284): array([ 0, 0, 0, 5001, 0, 0, 2396, 17, 0, 0, 0, 0, 2385, 0, 2368, 16, 0, 0, 0, 0, 2345, 0, 2398, 16, 0, 0, 0, 0], dtype='>u2'), (3, 1412): array([1983, 0, 0, 1978, 6, 1], dtype='>u2'), (3, 1540): array([ 3943, 64607, 65170, 28, 80, 100, 173, 0, 0, 0, 0, 0, 0, 0], dtype='>u2'), (3, 1668): array([ 0, 1978, 1, 40038, 0, 1030, 1, 17910, 0, 298, 0, 21711, 0, 190, 0, 61599, 0, 1651, 0, 39040, 0, 364, 0, 37930], dtype='>u2'), (3, 2114): array([2650, 120, 2750, 14, 0, 1], dtype='>u2')}, definition: {'name': 'ReactivePower_Output_Total', 'class': 'power', 'state_class': 'measurement', 'uom': 'W', 'scale': 10, 'rule': 2, 'registers': [1158], 'icon': 'mdi:home-lightning-bolt', 'platform': 'sensor', 'key': 'reactivepower_output_total_sensor', 'code': 3} [KeyError: 0]

werfpsa, 2025-02-03:
Warning spawn w/ temp: <class 'numpy.uint16'> (65520), value2: <class 'numpy.uint16'> (65520), value: <class 'numpy.uint16'> (65386), maxint: <class 'int'> (65535), scale: <class 'int'> (10)
Warning spawn w/ temp: <class 'numpy.uint16'> (65515), value2: <class 'numpy.uint16'> (65515), value: <class 'numpy.uint16'> (65336), maxint: <class 'int'> (65535), scale: <class 'int'> (10)
Warning spawn w/ temp: <class 'numpy.uint16'> (65517), value2: <class 'numpy.uint16'> (65517), value: <class 'numpy.uint16'> (65356), maxint: <class 'int'> (65535), scale: <class 'int'> (10)

davidrapan, 2025-02-03:
Change:
```
_LOGGER.warning(f"Warning spawn w/ data: {type(data)} and {type(data[0])} ({data})")
```
to:
```
_LOGGER.warning(f"Warning spawn w/ data: {type(data)} and {type(data[(3, 1156)][0])}and {type(data[(3, 1156)][2])} ({data})")
```

werfpsa, 2025-02-03:
[home-assistant_solarman_2025-02-03T16-44-02.535Z.log](https://github.com/user-attachments/files/18645363/home-assistant_solarman_2025-02-03T16-44-02.535Z.log)

davidrapan, 2025-02-03:
Thank you @werfpsa, you were very helpful! 👍😉

davidrapan, 2025-02-03:
> @githubDante, why is the type inside an array returned from umodbus response adu suddenly a numpy? What the heck?

* What does this have to do with python update?
* Why it does not happen on my installation?

All that while `parse_response_adu` specifically calls `int(i)`...

_This is hella confusing, I'm starting to hate python again..._ 😆

githubDante, 2025-02-03:
I'm 100% sure that the guys complaining here have an extra `umodbus` library. [See here](https://github.com/tiagocoutinho/async_modbus/blob/77d8dc48992c9d558c69b135ddd36ee6a515db6c/async_modbus/umodbus_numpy.py#L133).

davidrapan, 2025-02-03:
Ah I see...

This is some nasty level bs...

githubDante, 2025-02-03:
Link to the issue in [async_modbus](https://github.com/tiagocoutinho/async_modbus/issues/18).

davidrapan, 2025-02-03:
Yes, I'm also able to reproduce numpy overflow now.

Also, `is not None` remove introduces:
```
ValueError: The truth value of an array with more than one element is ambiguous. Use a.any() or a.all()
```

githubDante, 2025-02-04:
Quick and dirty fix for this [async umodbus patcher](https://gist.github.com/githubDante/c46ee04c3851ddc58c8bab9cf7d2d1de)

- save it to a file or download the raw format directly in your `venv` dir
then start it with
`python3  async_modbus_patch.py`

- in case that the module is not there move the script to your `.homeassistant/custom_components` dir and start it from there

davidrapan, 2025-02-04:
Thanks! I was even thinking of using pymodbus instead umodbus in the past but the structure is very different for it to be an easy transition... pymodbus is already used by core parts of HA and now w/ this issue maybe it will be worth the hustle...

davidrapan, 2025-02-04:
So i tried it: [refactor: Reimplementation using pymodbus](https://github.com/davidrapan/ha-solarman/commit/c471e4bc92db0e3465605b02100485667db47052) in [dev](https://github.com/davidrapan/ha-solarman/tree/dev)

_Just an experiment for now and not yet polished though...._

werfpsa, 2025-02-05:
David,
Can I already test from DEV?

werfpsa, 2025-02-05:
David,
I downloaded all the DEV files. Installed the custom_component files. Deleted the integration and re-installed the integration again. I must miss something as the error is still there. What am I missing?
Regards,
Pedro

davidrapan, 2025-02-05:
How did you downloaded it? Can you show me screenshots of folder structure and also of `pysolarman/pysolarman.py` file?

[32 more replies]

closed 2025-02-06 by davidrapan