davidrapan/ha-solarman #319: Time of use select box

gacopl, 2024-12-12:
### Description

Seems like your approach for time of use switch is not working properly because you merged together time of use register and weekdays register into single entity. Those should be separate, otherwise can give false reading, all i have to do is disable time of use on inverter panel, and integration still shows Enabled (because of the days are still enabled)

### Reproduction Steps

Have time of use enabled
Disable just time of use on inverter panel
Time of use entity stays enabled

### Log

No need easy to reproduce

### Version

24.12.08

### Home Assistant Version

2024.11.6

davidrapan, 2024-12-13:
There is no merging in place because it is in fact single register.

gacopl, 2024-12-13:
You have two entities. Prog time of use enable and use timer the merged one is use timer

When i turn off time of use on inverter i get prog time of used disabled. But use timer stays enabled because it's connected with weekdays also

Solarman stick logger integration has same issue. But he's got only one entity the problematic one

davidrapan, 2024-12-13:
Sure, just making clear that it is all controlled using single register. 😉

davidrapan, 2024-12-13:
[fix: TOU entity stays enabled when disabled - Deye](https://github.com/davidrapan/ha-solarman/commit/05cacfdc02b014d3974e20618c823efb44242ce9) should help.

gacopl, 2024-12-13:
Thanks and how one could enable Time of use on all days except Monday :D

That really should be separate entities IMO anyways thanks for a fix

davidrapan, 2024-12-13:
For even finer control, you can use a service action.

davidrapan, 2024-12-16:
I added some more TOU combinations especially for you [feat: Add some other TOU combinations](https://github.com/davidrapan/ha-solarman/commit/b4059b05842dda146f8fa67f55653c4ab19125de) 😆

_Will add some more on request as I really don't want to separate it into multiple entities. It doesn't feel right to me._ 😉

Until there is something like this https://github.com/home-assistant/architecture/discussions/927 implemented. 😉

closed 2024-12-16 by davidrapan

gacopl, 2024-12-16:
lol thank you that might actually be useful :D
still i prefer kellerza approach :P