davidrapan/ha-solarman #322: Deye generator is flapping between on/off

krasnoukhov, 2024-12-13:
### Description

[image]

### Reproduction Steps

My device is SG0XLP1. It was working fine on ha-solarman 24.11.x

### Log

```
2024-12-13 12:21:26.568 DEBUG (MainThread) [custom_components.solarman.api] [2727278373] Scheduling 2 query requests. ^1065
2024-12-13 12:21:26.568 DEBUG (MainThread) [custom_components.solarman.api] [2727278373] Querying 03 ~ 0079 - 0113 | 0x004F - 0x0071 #035 ...
2024-12-13 12:21:26.569 DEBUG (MainThread) [custom_components.solarman.api] [2727278373] SENT: a5 17 00 10 45 13 00 25 f7 8e a2 02 00 00 00 00 00 00 00 00 00 00 00 00 00 00 01 03 00 4f 00 23 35 c4 3c 15
2024-12-13 12:21:26.802 DEBUG (MainThread) [custom_components.solarman.api] [2727278373] RECD: a5 59 00 10 15 13 4b 25 f7 8e a2 02 01 ca bf 10 02 d1 00 00 00 dd 4a 4b 65 01 03 46 00 00 00 00 00 00 00 00 00 00 00 01 10 2d 00 00 0b 5f 00 00 00 00 05 3b 05 11 00 8a 00 00 07 d0 00 00 00 00 00 00 f2 c5 ff ff 0a 01 00 00 00 00 00 00 00 00 00 00 00 00 01 3b 00 00 00 10 00 00 00 05 00 06 00 00 e9 0d 24 15
2024-12-13 12:21:26.803 DEBUG (MainThread) [custom_components.solarman.api] [2727278373] Querying 03 ~ 0079 - 0113 | 0x004F - 0x0071 #035 succeeded, response: [0, 0, 0, 0, 0, 1, 4141, 0, 2911, 0, 0, 1339, 1297, 138, 0, 2000, 0, 0, 0, 62149, 65535, 2561, 0, 0, 0, 0, 0, 0, 315, 0, 16, 0, 5, 6, 0]
2024-12-13 12:21:26.803 DEBUG (MainThread) [custom_components.solarman.api] [2727278373] Querying 03 ~ 0150 - 0196 | 0x0096 - 0x00C4 #047 ...
2024-12-13 12:21:26.803 DEBUG (MainThread) [custom_components.solarman.api] [2727278373] SENT: a5 17 00 10 45 14 00 25 f7 8e a2 02 00 00 00 00 00 00 00 00 00 00 00 00 00 00 01 03 00 96 00 2f e4 3a b5 15
2024-12-13 12:21:28.055 DEBUG (MainThread) [custom_components.solarman.api] [2727278373] RECD: a5 71 00 10 15 14 4d 25 f7 8e a2 02 01 cb bf 10 02 d2 00 00 00 dd 4a 4b 65 01 03 5e 00 4e 00 00 00 4e 09 11 08 fe 00 00 08 fe 09 11 00 00 00 00 00 00 00 00 00 00 00 00 01 54 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 01 e7 00 00 01 e7 01 e7 00 00 01 e7 01 e7 00 00 00 1c 04 19 14 8a 00 62 01 e7 00 00 00 00 00 00 00 00 02 37 04 38 13 88 13 88 00 00 00 10 00 00 a8 23 4d 15
2024-12-13 12:21:28.056 DEBUG (MainThread) [custom_components.solarman.api] [2727278373] Querying 03 ~ 0150 - 0196 | 0x0096 - 0x00C4 #047 succeeded, response: [78, 0, 78, 2321, 2302, 0, 2302, 2321, 0, 0, 0, 0, 0, 0, 340, 0, 0, 0, 0, 0, 0, 0, 0, 487, 0, 487, 487, 0, 487, 487, 0, 28, 1049, 5258, 98, 487, 0, 0, 0, 0, 567, 1080, 5000, 5000, 0, 16, 0]
2024-12-13 12:21:28.057 DEBUG (MainThread) [custom_components.solarman.parser] do_validate Total Energy Export: 0.0 < 0.1
2024-12-13 12:21:28.057 DEBUG (MainThread) [custom_components.solarman.parser] do_validate Total Production: 0.0 < 0.1
2024-12-13 12:21:28.059 DEBUG (MainThread) [custom_components.solarman.api] [2727278373] Returning 57 new values to the Coordinator. [Previous State: Connected (1)]
2024-12-13 12:21:28.059 DEBUG (MainThread) [custom_components.solarman.coordinator] Finished fetching Deye data in 1.494 seconds (success: True)
2024-12-13 12:21:32.568 DEBUG (MainThread) [custom_components.solarman.api] [2727278373] Scheduling 2 query requests. ^1070
2024-12-13 12:21:32.568 DEBUG (MainThread) [custom_components.solarman.api] [2727278373] Querying 03 ~ 0059 - 0113 | 0x003B - 0x0071 #055 ...
2024-12-13 12:21:32.569 DEBUG (MainThread) [custom_components.solarman.api] [2727278373] SENT: a5 17 00 10 45 15 00 25 f7 8e a2 02 00 00 00 00 00 00 00 00 00 00 00 00 00 00 01 03 00 3b 00 37 75 d1 8b 15
2024-12-13 12:21:32.748 DEBUG (MainThread) [custom_components.solarman.api] [2727278373] RECD: a5 81 00 10 15 15 4e 25 f7 8e a2 02 01 d0 bf 10 02 d7 00 00 00 dd 4a 4b 65 01 03 6e 00 02 ff f9 00 00 00 00 dd 36 ff ff 00 00 01 05 ff db 00 00 00 00 00 02 00 03 20 88 00 00 1b 09 00 00 00 07 00 00 2e 4a 00 00 00 00 00 00 00 00 00 00 00 01 10 2d 00 00 0b 5f 00 00 00 00 05 3b 05 11 00 8a 00 00 07 d0 00 00 00 00 00 00 f2 c5 ff ff 0a 01 00 00 00 00 00 00 00 00 00 00 00 00 01 3b 00 00 00 10 00 00 00 05 00 06 00 00 77 22 63 15
2024-12-13 12:21:32.750 DEBUG (MainThread) [custom_components.solarman.api] [2727278373] Querying 03 ~ 0059 - 0113 | 0x003B - 0x0071 #055 succeeded, response: [2, 65529, 0, 0, 56630, 65535, 0, 261, 65499, 0, 0, 2, 3, 8328, 0, 6921, 0, 7, 0, 11850, 0, 0, 0, 0, 0, 1, 4141, 0, 2911, 0, 0, 1339, 1297, 138, 0, 2000, 0, 0, 0, 62149, 65535, 2561, 0, 0, 0, 0, 0, 0, 315, 0, 16, 0, 5, 6, 0]
2024-12-13 12:21:32.750 DEBUG (MainThread) [custom_components.solarman.api] [2727278373] Querying 03 ~ 0150 - 0196 | 0x0096 - 0x00C4 #047 ...
2024-12-13 12:21:32.751 DEBUG (MainThread) [custom_components.solarman.api] [2727278373] SENT: a5 17 00 10 45 16 00 25 f7 8e a2 02 00 00 00 00 00 00 00 00 00 00 00 00 00 00 01 03 00 96 00 2f e4 3a b7 15
2024-12-13 12:21:32.929 DEBUG (MainThread) [custom_components.solarman.api] [2727278373] RECD: a5 71 00 10 15 16 4f 25 f7 8e a2 02 01 d0 bf 10 02 d7 00 00 00 dd 4a 4b 65 01 03 5e 00 4e 00 00 00 4e 09 0e 08 fd 00 00 08 fd 09 0e 00 00 00 00 00 00 00 00 00 00 00 00 01 54 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 01 cf 00 00 01 cf 01 cf 00 00 01 cf 01 cf 00 00 00 1c 04 19 14 8a 00 62 01 cf 00 00 00 00 00 00 00 00 02 20 04 0c 13 88 13 88 00 00 00 10 00 00 e9 4d eb 15
2024-12-13 12:21:32.930 DEBUG (MainThread) [custom_components.solarman.api] [2727278373] Querying 03 ~ 0150 - 0196 | 0x0096 - 0x00C4 #047 succeeded, response: [78, 0, 78, 2318, 2301, 0, 2301, 2318, 0, 0, 0, 0, 0, 0, 340, 0, 0, 0, 0, 0, 0, 0, 0, 463, 0, 463, 463, 0, 463, 463, 0, 28, 1049, 5258, 98, 463, 0, 0, 0, 0, 544, 1036, 5000, 5000, 0, 16, 0]
2024-12-13 12:21:32.931 DEBUG (MainThread) [custom_components.solarman.parser] do_validate Total Energy: -890.5 < 0.1
2024-12-13 12:21:32.931 DEBUG (MainThread) [custom_components.solarman.parser] do_validate Total Energy Export: 0.0 < 0.1
2024-12-13 12:21:32.931 DEBUG (MainThread) [custom_components.solarman.parser] do_validate Total Production: 0.0 < 0.1
2024-12-13 12:21:32.934 DEBUG (MainThread) [custom_components.solarman.api] [2727278373] Returning 68 new values to the Coordinator. [Previous State: Connected (1)]
2024-12-13 12:21:32.934 DEBUG (MainThread) [custom_components.solarman.coordinator] Finished fetching Deye data in 0.368 seconds (success: True)
2024-12-13 12:21:37.567 DEBUG (MainThread) [custom_components.solarman.api] [2727278373] Scheduling 2 query requests. ^1075
2024-12-13 12:21:37.568 DEBUG (MainThread) [custom_components.solarman.api] [2727278373] Querying 03 ~ 0079 - 0113 | 0x004F - 0x0071 #035 ...
2024-12-13 12:21:37.568 DEBUG (MainThread) [custom_components.solarman.api] [2727278373] SENT: a5 17 00 10 45 17 00 25 f7 8e a2 02 00 00 00 00 00 00 00 00 00 00 00 00 00 00 01 03 00 4f 00 23 35 c4 40 15
2024-12-13 12:21:37.717 DEBUG (MainThread) [custom_components.solarman.api] [2727278373] RECD: a5 59 00 10 15 17 50 25 f7 8e a2 02 01 d5 bf 10 02 dc 00 00 00 dd 4a 4b 65 01 03 46 00 00 00 00 00 00 00 00 00 00 00 01 10 2d 00 00 0b 5f 00 00 00 00 05 3b 05 11 00 8a 00 00 07 d0 00 00 00 00 00 00 f2 c5 ff ff 0a 01 00 00 00 00 00 00 00 00 00 00 00 00 01 3b 00 00 00 10 00 00 00 05 00 06 00 00 e9 0d 43 15
2024-12-13 12:21:37.718 DEBUG (MainThread) [custom_components.solarman.api] [2727278373] Querying 03 ~ 0079 - 0113 | 0x004F - 0x0071 #035 succeeded, response: [0, 0, 0, 0, 0, 1, 4141, 0, 2911, 0, 0, 1339, 1297, 138, 0, 2000, 0, 0, 0, 62149, 65535, 2561, 0, 0, 0, 0, 0, 0, 315, 0, 16, 0, 5, 6, 0]
2024-12-13 12:21:37.718 DEBUG (MainThread) [custom_components.solarman.api] [2727278373] Querying 03 ~ 0150 - 0196 | 0x0096 - 0x00C4 #047 ...
2024-12-13 12:21:37.718 DEBUG (MainThread) [custom_components.solarman.api] [2727278373] SENT: a5 17 00 10 45 18 00 25 f7 8e a2 02 00 00 00 00 00 00 00 00 00 00 00 00 00 00 01 03 00 96 00 2f e4 3a b9 15
2024-12-13 12:21:37.888 DEBUG (MainThread) [custom_components.solarman.api] [2727278373] RECD: a5 71 00 10 15 18 51 25 f7 8e a2 02 01 d5 bf 10 02 dc 00 00 00 dd 4a 4b 65 01 03 5e 00 4e 00 00 00 4e 09 11 08 fd 00 00 08 fd 09 11 00 00 00 00 00 00 00 00 00 00 00 00 01 54 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 01 c8 00 00 01 c8 01 c8 00 00 01 c8 01 c8 00 00 00 1c 04 19 14 8c 00 62 01 c8 00 00 00 00 00 00 00 00 02 1a 03 ff 13 88 13 88 00 00 00 10 00 00 36 f0 b3 15
2024-12-13 12:21:37.889 DEBUG (MainThread) [custom_components.solarman.api] [2727278373] Querying 03 ~ 0150 - 0196 | 0x0096 - 0x00C4 #047 succeeded, response: [78, 0, 78, 2321, 2301, 0, 2301, 2321, 0, 0, 0, 0, 0, 0, 340, 0, 0, 0, 0, 0, 0, 0, 0, 456, 0, 456, 456, 0, 456, 456, 0, 28, 1049, 5260 [... cut ...]
2024-12-13 12:22:39.892 DEBUG (MainThread) [custom_components.solarman.parser] do_validate Total Energy Export: 0.0 < 0.1
2024-12-13 12:22:39.892 DEBUG (MainThread) [custom_components.solarman.parser] do_validate Total Production: 0.0 < 0.1
2024-12-13 12:22:39.896 DEBUG (MainThread) [custom_components.solarman.api] [2727278373] Returning 57 new values to the Coordinator. [Previous State: Connected (1)]
2024-12-13 12:22:39.897 DEBUG (MainThread) [custom_components.solarman.coordinator] Finished fetching Deye data in 0.332 seconds (success: True)

```

### Version

24.12.02

### Home Assistant Version

2024.12.2

davidrapan, 2024-12-13:
Hi @krasnoukhov, could you please record long enough log where is change in the register 195 observable?

krasnoukhov, 2024-12-13:
@davidrapan Here are some logs where I'm sure that value has flapped once. Compare the timestamps on screenshot and in the log. Not sure how to look for 195 in there

[image]

```
2024-12-13 23:31:02.568 DEBUG (MainThread) [custom_components.solarman.api] [2727278373] Scheduling 2 query requests. ^39860
2024-12-13 23:31:02.568 DEBUG (MainThread) [custom_components.solarman.api] [2727278373] Querying 03 ~ 0059 - 0113 | 0x003B - 0x0071 #055 ...
2024-12-13 23:31:02.569 DEBUG (MainThread) [custom_components.solarman.api] [2727278373] SENT: a5 17 00 10 45 1d 00 25 f7 8e a2 02 00 00 00 00 00 00 00 00 00 00 00 00 00 00 01 03 00 3b 00 37 75 d1 93 15
2024-12-13 23:31:02.757 DEBUG (MainThread) [custom_components.solarman.api] [2727278373] RECD: a5 81 00 10 15 1d cb 25 f7 8e a2 02 01 81 5c 11 02 84 15 00 00 15 4b 4b 65 01 03 6e 00 02 ff eb 00 00 00 00 dd 27 ff ff 00 00 01 1c ff cd 00 00 00 00 00 22 00 1e 20 a8 00 00 1b 24 00 00 00 2d 00 00 2e 70 13 81 00 00 00 00 00 00 00 00 00 18 10 45 00 00 0b 76 00 00 00 00 05 a1 05 7a 00 8a 00 00 07 d0 00 00 00 00 00 00 f2 b7 ff ff 0a 01 00 00 00 00 00 00 00 00 00 00 00 00 01 3b 00 00 00 1a 00 00 00 0c 00 06 00 00 bc d6 85 15
2024-12-13 23:31:02.758 DEBUG (MainThread) [custom_components.solarman.api] [2727278373] Querying 03 ~ 0059 - 0113 | 0x003B - 0x0071 #055 succeeded, response: [2, 65515, 0, 0, 56615, 65535, 0, 284, 65485, 0, 0, 34, 30, 8360, 0, 6948, 0, 45, 0, 11888, 4993, 0, 0, 0, 0, 24, 4165, 0, 2934, 0, 0, 1441, 1402, 138, 0, 2000, 0, 0, 0, 62135, 65535, 2561, 0, 0, 0, 0, 0, 0, 315, 0, 26, 0, 12, 6, 0]
2024-12-13 23:31:02.758 DEBUG (MainThread) [custom_components.solarman.api] [2727278373] Querying 03 ~ 0150 - 0196 | 0x0096 - 0x00C4 #047 ...
2024-12-13 23:31:02.759 DEBUG (MainThread) [custom_components.solarman.api] [2727278373] SENT: a5 17 00 10 45 1e 00 25 f7 8e a2 02 00 00 00 00 00 00 00 00 00 00 00 00 00 00 01 03 00 96 00 2f e4 3a bf 15
2024-12-13 23:31:02.926 DEBUG (MainThread) [custom_components.solarman.api] [2727278373] RECD: a5 71 00 10 15 1e cc 25 f7 8e a2 02 01 82 5c 11 02 85 15 00 00 15 4b 4b 65 01 03 5e 09 50 00 00 09 50 09 44 09 35 00 00 09 35 09 44 00 00 00 00 02 e8 00 00 02 64 00 00 02 44 00 00 00 00 06 c6 00 00 06 c6 ff ec 00 00 ff ec f9 5f 00 00 f9 5f 00 25 00 00 00 25 00 25 00 00 00 1f 04 2d 16 22 00 64 00 25 00 00 00 00 00 00 00 00 f9 b1 f4 dc 13 84 13 84 00 01 00 10 00 00 0b 9d 29 15
2024-12-13 23:31:02.927 DEBUG (MainThread) [custom_components.solarman.api] [2727278373] Querying 03 ~ 0150 - 0196 | 0x0096 - 0x00C4 #047 succeeded, response: [2384, 0, 2384, 2372, 2357, 0, 2357, 2372, 0, 0, 744, 0, 612, 0, 580, 0, 0, 1734, 0, 1734, 65516, 0, 65516, 63839, 0, 63839, 37, 0, 37, 37, 0, 31, 1069, 5666, 100, 37, 0, 0, 0, 0, 63921, 62684, 4996, 4996, 1, 16, 0]
2024-12-13 23:31:02.928 DEBUG (MainThread) [custom_components.solarman.parser] do_validate Total Energy: -892.0 < 0.1
2024-12-13 23:31:02.928 DEBUG (MainThread) [custom_components.solarman.parser] do_validate Total Energy Export: 0.0 < 0.1
2024-12-13 23:31:02.928 DEBUG (MainThread) [custom_components.solarman.parser] do_validate Total Production: 0.0 < 0.1
2024-12-13 23:31:02.931 DEBUG (MainThread) [custom_components.solarman.api] [2727278373] Returning 68 new values to the Coordinator. [Previous State: Connected (1)]
2024-12-13 23:31:02.932 DEBUG (MainThread) [custom_components.solarman.coordinator] Finished fetching Deye data in 0.367 seconds (success: True)
2024-12-13 23:31:02.966 DEBUG (MainThread) [custom_components.solarman.api] [2727278373] Scheduling call request.
2024-12-13 23:31:02.966 DEBUG (MainThread) [custom_components.solarman.api] [2727278373] Call 10 ~ 230 | 0x00E6: 112 ...
2024-12-13 23:31:02.966 DEBUG (MainThread) [custom_components.solarman.api] [2727278373] SENT: a5 1a 00 10 45 1f 00 25 f7 8e a2 02 00 00 00 00 00 00 00 00 00 00 00 00 00 00 01 10 00 e6 00 01 02 00 70 b1 b2 a9 15
2024-12-13 23:31:03.046 DEBUG (MainThread) [custom_components.solarman.api] [2727278373] RECD: a5 16 00 10 15 1f cd 25 f7 8e a2 02 01 82 5c 11 02 85 15 00 00 15 4b 4b 65 01 10 00 e6 00 01 e0 3e 27 15
2024-12-13 23:31:03.047 DEBUG (MainThread) [custom_components.solarman.api] [2727278373] Call 10 ~ 230 | 0x00E6: 112 succeeded, response: 1
2024-12-13 23:31:07.568 DEBUG (MainThread) [custom_components.solarman.api] [2727278373] Scheduling 2 query requests. ^39865
2024-12-13 23:31:07.568 DEBUG (MainThread) [custom_components.solarman.api] [2727278373] Querying 03 ~ 0079 - 0113 | 0x004F - 0x0071 #035 ...
2024-12-13 23:31:07.569 DEBUG (MainThread) [custom_components.solarman.api] [2727278373] SENT: a5 17 00 10 45 20 00 25 f7 8e a2 02 00 00 00 00 00 00 00 00 00 00 00 00 00 00 01 03 00 4f 00 23 35 c4 49 15
2024-12-13 23:31:07.718 DEBUG (MainThread) [custom_components.solarman.api] [2727278373] RECD: a5 59 00 10 15 20 ce 25 f7 8e a2 02 01 86 5c 11 02 89 15 00 00 15 4b 4b 65 01 03 46 13 84 00 00 00 00 00 00 00 00 00 18 10 45 00 00 0b 76 00 00 00 00 05 a1 05 7a 00 8a 00 00 07 d0 00 00 00 00 00 00 f2 b7 ff ff 0a 01 00 00 00 00 00 00 00 00 00 00 00 00 01 3b 00 00 00 14 00 00 00 07 00 06 00 00 c0 a4 26 15
2024-12-13 23:31:07.719 DEBUG (MainThread) [custom_components.solarman.api] [2727278373] Querying 03 ~ 0079 - 0113 | 0x004F - 0x0071 #035 succeeded, response: [4996, 0, 0, 0, 0, 24, 4165, 0, 2934, 0, 0, 1441, 1402, 138, 0, 2000, 0, 0, 0, 62135, 65535, 2561, 0, 0, 0, 0, 0, 0, 315, 0, 20, 0, 7, 6, 0]
[... cut ...]
2024-12-13 23:31:50.893 DEBUG (MainThread) [custom_components.solarman.parser] do_validate Total Energy Export: 0.0 < 0.1
2024-12-13 23:31:50.894 DEBUG (MainThread) [custom_components.solarman.parser] do_validate Total Production: 0.0 < 0.1
2024-12-13 23:31:50.899 DEBUG (MainThread) [custom_components.solarman.api] [2727278373] Returning 57 new values to the Coordinator. [Previous State: Connected (1)]
2024-12-13 23:31:50.900 DEBUG (MainThread) [custom_components.solarman.coordinator] Finished fetching Deye data in 0.334 seconds (success: True)
```

davidrapan, 2024-12-13:
> Not sure how to look for 195 in there

It is the second value from the end on this line: `Querying 03 ~ 0150 - 0196 | 0x0096 - 0x00C4 #047 succeeded, response:`

davidrapan, 2024-12-13:
And I think I know what is the issue. 😉

Thanks for your assistance.

davidrapan, 2024-12-14:
Should be fixed in [v24.12.14](https://github.com/davidrapan/ha-solarman/releases/tag/v24.12.14)

krasnoukhov, 2024-12-14:
Ok thanks @davidrapan. It's not flapping anymore. But there must be still something wrong with it...

If I turn on the Generator switch it stays turned on. However it does not look like it actually takes effect:

[image]

Also Generator binary sensor is always On, not matter what switch state. Given that in web UI Generator force is off it all seems wrong

krasnoukhov, 2024-12-14:
I see that binary sensor's register changed from 0x00A6 to 0x00C3 in [this commit](https://github.com/davidrapan/ha-solarman/commit/954f4e2da0eb327fcb00b54db6385c281ca4782e#diff-df7339f92e2b4cfbef4f5b22227332f650a16d991c8c934930f2b24f3da8ffc7R870). Maybe that's the issue with it?

As for non-functional switch I'm not sure, yaml seems correct, so no idea what went wrong since 24.11.x

krasnoukhov, 2024-12-14:
If I enable Gen Force via web UI nothing changes in integration entities - not switch, not binary sensor. So it seems like these entities are not hooked up to a correct state

davidrapan, 2024-12-14:
YAML is according to the documentation, if you think it's not right you need to find out what's wrong exactly.

krasnoukhov, 2024-12-16:
I have smartload mode set to generator.

I see that in [this commit](https://github.com/davidrapan/ha-solarman/commit/cf3c8dc433255365a82d9b288a5701c87ec9322a) you renamed the entity that actually controls the Generator Force (for me at least, that's register 0x0146) to Meter. If I change that select to Generator it actually works as expected. I initially added that entity in #154 as Generator Force.

So I guess I'll be using that when I need to force generator. And I guess for me both Generator binary sensor and Switch are incorrect so I'm not going to be using those.

closed 2024-12-16 by krasnoukhov

davidrapan, 2024-12-16:
The `Meter` entity change is tracked here: https://github.com/davidrapan/ha-solarman/discussions/297#discussioncomment-11498798

It was for two reasons, one mentioned in the comments above and the other because the register controls more than just generator relay.

krasnoukhov, 2024-12-16:
I guess it makes sense overall... I'm sure you did all according to the docs. But unfortunately in real life that Meter entity controls the Gen Force which is in Solarman's web UI. And other current Generator entities are non-functional, at least in my case.

krasnoukhov, 2024-12-16:
In my case changing 234's value does nothing practically. But it seems to persist. So I guess it's stored at least 😄 

As for 146, this configuration works and enables/disables the Gen Force I can see in the web UI:

```
      - name: "Generator"
        platform: switch
        rule: 1
        registers: [0x0146]
        value:
          on: 0x2000
          off: 0x0000
        icon: mdi:generator-stationary
```

I can confirm it in practice with my device and generator I have

davidrapan, 2024-12-16:
And are you sure that those two do not work in tandem maybe?

krasnoukhov, 2024-12-16:
Nope, I have 234 "turned off" at all times. And 146 works regardless

davidrapan, 2024-12-17:
Take a look at [feat: Gen/SL/MI sensors named by selected IO mode](https://github.com/davidrapan/ha-solarman/commit/7e94a4d6b8bceb11e44153faa013d83eaae42ff1)

_Could you please show me (using built-in service action) what's in register 195 when is your Generator On and Off?_

krasnoukhov, 2024-12-17:
Action:

```
action: solarman.read_holding_registers
data:
  device: b0ca68b38bce24520d940266549f663a
  register: 195
  quantity: 1
```

When generator is off:

```
"195": 16
```

When on, via register 146:

```
"195": 272
```

davidrapan, 2024-12-17:
So it looks like bit 8 of 195 could be used for `binary_sensor`. 🤔
```
      - name: "Generator"
        name_lookup: "io_mode_select"
        update_interval: 30
        platform: "binary_sensor"
        device_class: power
        rule: 1
        registers: [0x00C3]
        bit: 8
```

krasnoukhov, 2024-12-17:
Yeah, this works. I replaced the binary sensor for generator with this and it works as expected when I turn generator on/off via register 146

krasnoukhov, 2024-12-17:
> ```
>       - name: "Generator"
>         platform: switch
>         rule: 1
>         registers: [0x0146]
>         value:
>           bit: 13
>         icon: mdi:generator-stationary
> ```

Just tried this and it also works to turn the generator on/off and state of 195 changes as well

davidrapan, 2024-12-17:
So I guess you are not using external meter when disabling it /w that 0x146 bit 0 set to zero did not caused you any issues?

krasnoukhov, 2024-12-17:
Yeah, no external meter, and no, no issues that I noticed. Gen works for a while and then I disable it via the switch, everything's good