davidrapan/ha-solarman #325: BMS values missing from sensor.inverter_battery after upgrade

toruonu, 2024-12-15:
### Description

I was running 24.11.02 (I think, can't remember that preciselyl). Then discovered yesterday that out of the blue solarman has a huge amount of updates and decided to update all the way to 24.12.14. 

However after restarting HA I discovered, that a crucial data attribute is no longer available. I have Deye SG04LP3 and the battery voltage tends to be reported wrong after the discharge ends. So I was using the attributes from sensor.inverter_battery that had the BMS current and BMS voltage. However those are no longer available. Now it contains only the nominal voltage and max cycle count that seem to be given in the configuration. 

When did they disappear and how can I get the actual BMS values now? 

Why I need it is that I have the battery charging set to be between 47V and 58.5V. When it discharges down to 47V it nicely stops discharge and settles in the BMS. However the sensor.inverter_battery_voltage seems to drop relatively fast to somewhere around 36V. However if I go to the inverter and check the battery BMS information from the battery LCD display the packs are nicely at 46-47V region with cells relatively balanced. No clue why Deye misreports the voltage. But the attributes that took it from BMS previously had the right numbers so I used those in all my automations and code. 

### Reproduction Steps

I guess install a prior version and upgrade. 

### Log

Not relevant

### Version

24.12.14

### Home Assistant Version

2024.12.3

amiko24, 2024-12-15:
I have had a similar problem since the last update. I have read out the registers from address 10032, where the data of the individual battery packs are located. Unfortunately, this no longer works. I have extended ayml file, here as an example
`      - name: "Battery 1 Cycle number"
        class: "battery"
        state_class: "measurement"
        uom: " "
        scale: 1
        rule: 1
        registers: [0x2748]
        icon: "mdi:battery"    

      - name: "Battery 2 Cycle number"
        class: "battery"
        state_class: "measurement"
        uom: " "
        scale: 1
        rule: 1
        registers: [0x276E]
        icon: "mdi:battery"       

      - name: "Battery 1 Max Cel V"
        class: "voltage"
        state_class: "measurement"
        uom: "V"
        scale: 0.001
        rule: 1
        registers: [0x2746]
        icon: "mdi:battery"`

How it was before the update
[image]

this is after the update
[image]

davidrapan, 2024-12-15:
Hi @toruonu and @amiko24, I'm aware of broken attributes and I'll fix it soon.

But I'm really not sure what could be causing issues w/ regular reading of registers and w/ this so little information I can't even guess.

_BTW do not use_ `class: "battery"` _more than once per (file) device._ 😉

davidrapan, 2024-12-16:
Attributes are fixed in [v24.12.16](https://github.com/davidrapan/ha-solarman/releases/tag/v24.12.16)

toruonu, 2024-12-16:
I can confirm that the attributes are back.

closed 2024-12-16 by toruonu

amiko24, 2024-12-16:
Hello David, thank you for your integration and for the work you are doing.
What information do you need?
I have modified the deye_p3.yaml, I have added 14 more registers from address 10040. Unfortunately, reading these registers no longer works since the last update. It doesn't work with version 24.12.16 either. Have you possibly limited the number of registers to be read to a value below 10.000?
I would like to read the registers
[image]

amiko24, 2024-12-16:
I have found the problem, now it works. That was my mistake.

davidrapan, 2024-12-16:
What it was then?

amiko24, 2024-12-17:
As I did not want to make any changes to the original deye_p3.yaml file, I copied the file and renamed it. I made the changes in the new file. Apparently unknown file for the integration caused problems. I then made the changes in the original deye_p3.yaml file, after which everything worked as desired.

davidrapan, 2024-12-17:
But you will lose the changes w/ update if you just changed the default file.

amiko24, 2024-12-17:
Yes, I know