davidrapan/ha-solarman #341: Grid Tie Inv meter2

Golempl, 2024-12-23:
### Is your feature request related to a problem?

Would be nice to read this additional meter.

### Describe the solution you'd like

Just add Grid Tie Inv Meter2

### Describe alternatives you've considered

No alternative.

davidrapan, 2024-12-23:
I have no idea what this device is. Do you have a  rtu documentation?

Golempl, 2024-12-23:
This is additional meter when u tick Grid Side INV meter2
It works with normal CT's but also provides generation data from AC coupled inverters. Deye Cloud include this data. I think it's possible but need possibly modebus address
[image]

davidrapan, 2024-12-23:
You should have said it's for Deye. It looked like you are requesting addition of new device. 😉

So you want to be able to configure what's on this page?

Golempl, 2024-12-23:
No configuration is needed. I just want to access the readings from the additional meter connected to the Deye. I have the Deye 12k-SG04LP3-EU, and with the latest firmware (C04B/F150), it is now possible to use an extra meter. It would be great to have the data from this meter in Home Assistant, specifically the kWh value and name, such as: Grid Tie Meter2.

I'm not entirely sure if the following address is correct, but it might be worth checking or consulting Deye support for the correct one:

    536: Daily energy generation by the generator (R, unit: 0.1 kWh)
    537: Low byte of the total energy generated by the generator (R, unit: 0.1 kWh)
    538: High byte of the total energy generated by the generator (R, unit: 0.1 kWh)

davidrapan, 2024-12-23:
Ah I see, but those are already there. It's from generator port and it's called `Total Generation`.

closed 2024-12-25 by davidrapan