davidrapan/ha-solarman #342: deye_micro - Device State Fault, when Inverter is working

V4n1X, 2024-12-24:
### Description

Deye SUN600G3-EU-230
Firmware: MW3_SSL_5408_1.0B

I don't use the additional Relay Box.

Solarman Device State show Fault (Value: 4), when the Inverter is working.
Solarman Device State show Normal (Value: 2), when the Inverter is online, but not working, due to low light conditions.

[image]

[image]

So something seems of in the inverter defintion?
```
      - name: "Device State"
        class: "enum"
        rule: 1
        registers: [0x003B]
        icon: "mdi:state-machine"
        range:
          min: 0
          max: 5
        lookup:
          - key: 0x0000
            value: "Standby"
          - key: 0x0001
            value: "Self-test"
          - key: 0x0002
            value: "Normal"
          - key: 0x0003
            value: "Alarm"
          - key: 0x0004
            value: "Fault"
```

### Reproduction Steps

Inverter working -> Value 4
Inverter online, but not working due to low light -> 2
Other values, I don't have seem in the HA log.

### Log

I dont know what to post.

### Version

24.12.22

### Home Assistant Version

2024.12.5

CrazyUs3r, 2024-12-24:
Hey, do you have a few numbers for the error codes? On your history you can see that everything was fine the day before but what happened in the meantime

CrazyUs3r, 2024-12-24:
I have the same problem with my sun80m, I currently have 5.0C-S because I have had a ticket with deye for almost a year, where they keep saying that when it is produced everything is fine, but I keep writing to support because maybe they will write to me and tell me if it is like that. If you get a solution from deye, you are welcome to forward it to me.

davidrapan, 2024-12-24:
That's a firmware for the stick but what about device firmware? Did you tried to ask deye for an update maybe?

V4n1X, 2024-12-24:
> That's a firmware for the stick but what about device firmware? Did you tried to ask deye for an update maybe?

Yes, but the micro deye its one physical device, so there is not stick attached, I can ask Deye about it too, if they reply on my first email.

[image]

[image]

If I check the inverter on the solarman portal, I do get a DC master firmware version:
[image]

davidrapan, 2024-12-24:
There is a "stick inside" and device still has it's own firmware. 😉

davidrapan, 2024-12-28:
This is rather question for your manufacturer. This integration simply reads registers according to the documentation.

closed 2024-12-28 by davidrapan