davidrapan/ha-solarman #346: solis_3p-5g.yaml is not working for RHI-3P8K-HVES-5G

honikos, 2024-12-26:
### Description

I have solis inverter RHI-3P8K-HVES-5G.
Inverter-Version
090037
Modell
C9

When I choose in initial configuration the profil -  solis_3p-5g.yaml, I get configuration errors.
Following the comment in yaml - file, this should be fitting file
# Solis RHI-3P5~10K-HVES-5G | 5~10kW | 2 MPPT | Three Phase Inverter

If I change in  solis_3p-5g.yaml the inital defaults from

default:
  update_interval: 10
  code: 0x04
  digits: 6

to (from solis_hybrid.yaml)

default:
  update_interval: 10
  code: 0x04
  min_span: -1
  max_size: 100
  digits: 6

I get 60 entities 

whereas when I use the solis_hybrid.yaml file in profile, I get 80 entities, which looks better.
I am not sure if this is an stable solution.

### Reproduction Steps

for RHI-3P8K-HVES-5G

use solis_3p-5g.yaml  file in profile section

### Log

configuration error

### Version

24.12.22

### Home Assistant Core Version

2025.12.5

### Home Assistant Operating System Version

14.1

davidrapan, 2024-12-26:
They both use the same registers and just contains different number of them so if it works w/ the other one just use it. 😉

honikos, 2024-12-26:
You are absolutely right. But shouldn't the wrong file be deleted, cause it doesn't work?

davidrapan, 2024-12-28:
It was added by an user (by proxy) so maybe the file does work for some.

I'm planning "big" merging of files with the same registers in the future so I will leave it for now. 😉

closed 2024-12-28 by davidrapan