davidrapan/ha-solarman #348: Set Inverter Time via Addon

Xnyle, 2024-12-26:
### Is your feature request related to a problem?

Deye is pushing out updates that are potentially bricking inverters (no relays box despite no need / wrong country detected and so on)
So many people block internet access completely to forbid further updates.

In that case the inverter can not get the network time and so daily production can't reset.

### Describe the solution you'd like

But pvsolarman has a way to set the time as well via

"SunXG3RegistersWrite.DeviceTime(inv, datetime.datetime.now())"

Could that be implemented within the addon, maybe with an option switch to enable it?

### Describe alternatives you've considered

-

davidrapan, 2024-12-26:
There is a way you can do it already and it's by using automation and setting Date & Time entity but there are some other ways currently coming (it's already part of latest code in this repo and just not yet released) and it's time synchronization using heartbeat time replies but it won't work for every inverter.

Xnyle, 2024-12-27:
Thanks, but how? this doesn't seem to do anything, time entity not changing when triggered?
```
alias: Set Inverter Datetime
description: ""
triggers:
  - hours: "*"
    minutes: /5
    seconds: "5"
    trigger: time_pattern
conditions:
  - condition: and
    conditions:
      - condition: or
        conditions:
          - condition: time
            after: "06:30:00"
            before: "22:30:59"
  - condition: not
    conditions:
      - condition: state
        entity_id: sensor.inverter_device_state
        state: unavailable
actions:
  - data:
      datetime: "{{ now() }}"
    target:
      entity_id: datetime.inverter_date_time
    action: input_datetime.set_datetime
mode: single
```

davidrapan, 2024-12-27:
Try to run this during debug logging enabled.

Xnyle, 2024-12-28:
Referenced entities datetime.inverter_date_time are missing or not currently available

But thats the entities id name from the details?

davidrapan, 2024-12-28:
Do you have this entity in the list of entities in device detail page?

Xnyle, 2024-12-29:
Yes, in the "Configuration" section

davidrapan, 2024-12-29:
Ah, it should be proly this action
```
action: datetime.set_value
```

CrazyUs3r, 2024-12-29:
Is it necessary to change a setting on the inverter for the heartbeat? I just blocked the internet and it had no date or time. Writing about the action worked though

davidrapan, 2024-12-29:
> Is it necessary to change a setting on the inverter for the heartbeat?

Not sure if it's enough honestly. It needs to be tested more and the packets has to be verified in server mode I guess.

Skarabaen, 2024-12-29:
Deye can upgrade your firmware without your knowledge and currently do firmware upgrades which brick some devices. So I blocked the inverters from connecting the internet like @CrazyUs3r 
Unfortunately it seems that there are hardcoded ntp servers in the firmware, so without internet they can`t update the date and "Today Production" values don`t reset every day but sum up. I used the datetime.xxxxxx_date_time function to set the date manually. But it seems that it only triggers if you go to the webinterface of the inverter and press the restart button. After that the values "Today Production" values are set back to 0.

Can we set the somehow the ntp server with your addon?

Xnyle, 2024-12-29:
I wouldn't advice to grant those devices any internet access at all now or even locally connect via the app (I guess that's how some got their updates forced)

```
actions:
  - data:
      datetime: "{{ now() }}"
    target:
      entity_id: datetime.inverter_date_time
    action: datetime.set_value
```

worked but as mentioned does not reset todays production
But that isn't really a problem as i will reset the next day so in essence it will reset every day except there it some very heavy cloud cover during the day.

davidrapan, 2024-12-29:
> Unfortunately it seems that there are hardcoded ntp servers.
Can we set the somehow the ntp server with your addon?

It doesn't use NTP. And this is an integration not an addon.

So far it seems like devices proly set time during handshake which is done in server mode. This integration doesn't support that, yet. But it's proly not even possible because integration (to my knowledge) can't listen on ports so it would have to be done using an addon.

_Integrations and Addons are for HA two different things._

Skarabaen, 2024-12-30:
> I wouldn't advice to grant those devices any internet access at all now or even locally connect via the app (I guess that's how some got their updates forced)
>
> ```
> actions:
>   - data:
>       datetime: [...]

with this automation the time will get set. I can´t check if the production will be reset, because we had no sun yesterday. But in case of doubt, it is just another automation that restarts the inverter once in the morning

closed 2025-01-20 by davidrapan

coolje, 2025-03-18:
Thank you for the Adon helping me to cloud-free operation of my 300W (single) and 800W (double) inverters. 
The above automatisations failed several times since my balcony inverters (one 300W ond 800W) from deye are not ready all days.
So i added a little bit:
I created first boolean helper that schould indicating if the time already was set on the specific day.
1. In my case i created 2 boolean (switch) helper:  
daylytimeset300wattdeye
daylytimeset800wattdeye
 
2. Than i created an automation that thanges the "flags" to "false" triggert by one hour before sunrise (and homeassist-start)    This will indicate the next automation that the time has not set for the inverters on the actual day.

alias: ResetDaylyinverterResetFlag
description: ""
triggers:
  - trigger: homeassistant
    event: start
  - trigger: sun
    event: sunrise
    offset: "-01:00:00"
conditions: []
actions:
  - action: input_boolean.turn_off
    metadata: {}
    data: {}
    target:
      entity_id:
        - input_boolean.daylytimeset300wattdeye
        - input_boolean.daylytimeset800wattdeye
mode: single

3. Now i created for each inverter a automation that runs when my inverter get connected (with +5 Seconds) but runs only the first time a day (this is enough since there is an internal capacitor backing up the time in the inverter in my case recent long time)
(my inverter named 300W and 800W, the following automation is just the 300W, but i have an identical with 800W naming).
(Please notice: Device_id and entity_id in the next automation have to be set to you individual id-number.)
The automation will set the Flag to "true" preventing the automation to run more than once a time (, so not resetting the dayly production mutliple time to "0")

alias: 300W time synchronization
description: Set device's time to `now()`
triggers:
  - event: start
    trigger: homeassistant
  - type: connected
    device_id: (id set to your own)
    entity_id: (my identity_id set to your own)
    domain: binary_sensor
    trigger: device
    for:
      hours: 0
      minutes: 0
      seconds: 5
conditions:
  - condition: state
    entity_id: input_boolean.daylytimeset300wattdeye
    state: "off"
actions:
  - data:
      datetime: "{{ now() }}"
    target:
      entity_id: datetime.300W_date_time
    action: datetime.set_value
  - action: input_boolean.turn_on
    metadata: {}
    data: {}
    target:
      entity_id: input_boolean.daylytimeset300wattdeye
mode: single

davidrapan, 2025-03-18:
Hi @coolje, please use code blocks for the automations.

coolje, 2025-03-18:
sorry - never used codblocks before  - lesson learned

coolje, 2025-03-18:
I use two automations since i have one 800W inverter serving 2 450W-panels on a south oriented balcony. The second inverter mounted on a north balcony serves one 320W-Panel.
Especially the nort orientated inverter starts on cloudy, foggy, days very late compared to the south one, and not realy related to sunrise. In last January i had days the south produced more than 3kWh wheras the north did only 0,1kWh. 
For example the 800W inverter this mornig startet at 6:58am wheras the north orientated startet at 9:06am. 
Reacting on the first event the inverter is connecting with my wlan aka. homeassist, for me it was easiest to use individual automation on each.mations. 
I first used the example automation but this did not offer me the "real" connecting time to setup the time (--> resetting automatically the stored production of last day).

davidrapan, 2025-03-18:
> I first used the example automation but this did not offer me the "real" connecting time to setup the time

I don't know what this means.

lukaszbalcerzakk, 2026-03-29:
Hi, I have similar issue. I have sofar 11KTLX-g3 inverter, so I use sofar_g3.yaml.
I blocked internet access for this inverter, to avoid updates that may brick the device.
However datetime of the device seems to be off, Today's sensors reset during the date , producing incorrect values. 

[image]

[image]

I'm in +1 timezone. 

I don't see any entity like datetime.inverter_date_time that I can use to reset the date time on my inverter. 
Does it mean such entity doesn't exist / is not exposed for my inverter? Or i get something wrong?

One workaround is to calculate Today's sensor from Total sensor, is this something you would recommend here?