davidrapan/ha-solarman #350: Battery SOH, Today Life Cycles and Total Life Cycles are calculated incorrectly for Deye BOS GM5.1 + BMS HVB750V/100A

iwannatalk, 2024-12-29:
### Description

Hello,

The following sensor values are incorrectly calculated for Deye BOS GM5.1 + BMS HVB750V/100A:
- Battery SOH
- Today Life Cycles
- Total Life Cycles

Im not sure whether it is a bug or a feature request (cause maybe noone had a chance to test it with Deye hith voltage batteries).

I have BMS HVB750V/100A + 8 instances of Deye BOS GM5.1 and was wondering why my battery when charging from (lets say from todays example) from SOC 12% till 90% takes 3.37 lifecycles :) and totally 58 days of work already used 142 lifecycles but i change only over night and discharge during the day (excepting sunny days but now it is a winter and it is not too mamy sunny days). 
Today i have investigated the code and see that those values are calculated but not taken from inverter/bms/battery statistics.
So seems like there is a bug with those calculations.

### Reproduction Steps

Any charge shows wrongly calculated sensor data for:
- Battery SOH
- Today Life Cycles
- Total Life Cycles
For example, charging from 12% to 90% takes 3.37 lifecycles.

### Log

Not sure they are needed, if so - i will attach later.

### Version

24.12.22

### Home Assistant Core Version

2024.12.5

### Home Assistant Operating System Version

14.1

davidrapan, 2024-12-29:
Did you set the nominal voltage in the configuration?

iwannatalk, 2024-12-29:
> Did you set the nominal voltage in the configuration?

No, i thought it is taken from inverter data as this value is shown in stat.
Now i set in configuration 409 v.
Seems like it is more or less correct now. Showing 0.44 instead of 3.37 for today but not sure if it still correct.

iwannatalk, 2024-12-29:
Btw, total battery cycles and soh is available in solarman stat.

davidrapan, 2024-12-29:
What value you have in Device sensor?

iwannatalk, 2024-12-30:
> What value you have in Device sensor?

[image]

But this value changes depending on the SOC.

davidrapan, 2024-12-30:
Can you please show me what value you get when you read register number 224 using Developer tools >> ACTIONS >> Solarman: Read Holding Registers (Modbus Function Code 3)?

iwannatalk, 2024-12-30:
> Can you please show me what value you get when you read register number 224 using Developer tools >> ACTIONS >> Solarman: Read Holding Registers (Modbus Function Code 3)?

[image]

davidrapan, 2024-12-30:
What about 255, 10006 and 10048.

iwannatalk, 2024-12-30:
> What about 255, 10006 and 10048.

0
100
30000

davidrapan, 2024-12-30:
So it's proly at 100% at all times and thus not measured.

closed 2024-12-30 by davidrapan