davidrapan/ha-solarman #351: Not initialising

pravishj, 2024-12-29:
### Description

I run home asssistant in a container when I configure I get the following error.  I think its not able to connect 

### Reproduction Steps

not able to configure 

### Log

[home-assistant_solarman_2024-12-29T06-28-23.837Z.log](https://github.com/user-attachments/files/18269094/home-assistant_solarman_2024-12-29T06-28-23.837Z.log)

### Version

24.12.22

### Home Assistant Core Version

2024.12.5

### Home Assistant Operating System Version

14.1

davidrapan, 2024-12-29:
There is no connection w/ your device.

closed 2024-12-30 by davidrapan

pravishj, 2025-01-01:
thanks @davidrapan I have found out the problem...Mine HYD 6000-EP but when I use auto profile it looks for "HYD3K-6K-ES". As shown in the above log. Thanks for your prompt response. I also take this oppurtunity to Wish you a Happy new year!

davidrapan, 2025-01-01:
I'm glad you were able to resolve the issue and thank you, and I also wish you a happy new year!