davidrapan/ha-solarman #353: After latest update Inverter Grid sendor always On

iwannatalk, 2024-12-30:
### Description

After latest update Inverter Grid sendor always On.
It was working before update for sure as i have HA automation based on this sensor.

[image]

### Reproduction Steps

Turn off grid.
Check Inverter Grid sensor.

### Log

[error_log-2.txt](https://github.com/user-attachments/files/18272170/error_log-2.txt)

### Version

24.12.22

### Home Assistant Core Version

2024.12.5

### Home Assistant Operating System Version

14.1

davidrapan, 2024-12-30:
What was the state of the relay?

iwannatalk, 2024-12-30:
> What was the state of the relay?

At the moment there is an outage at all.
[image]

Other sensors shows correctly, but Grid is on.
It also is reproducible when some phase is absent. It also shows on hut before update it was showing Off.
[image]

davidrapan, 2024-12-30:
Show me the **Relay State**.

iwannatalk, 2024-12-30:
> Show me the **Relay** state in HA (w/ attributes in sensor detail).

Do you mean this one?
[image]

davidrapan, 2024-12-30:
Yes that's the one, it's used for Grid sensor.

iwannatalk, 2024-12-30:
> Yes that's the one, it's used for Grid sensor. The relay has bit for Grid set... there is not much I can really do...
> 
> Can you show history of this Relay sensor? What value was there previously for example with grid off?

Seems its not related to relay status.
Here is an example, there was no voltage at grid L2, the rest of phases were ok. So there was same power supply but with grid off:
[image]
[image]
[image]

davidrapan, 2024-12-30:
Mhmm, "Power Suppy" is a placeholder value for everything except known combination so the real value could be different.

_But it's hard to read history of attributes so I won't bother you w/ that._

But my suspicion is that your inverter simply does not show correct state of the Relay and thus also shows incorrect state of the Grid sensor.

Did you tried to restart it or smthing?

iwannatalk, 2024-12-30:
> Mhmm, "Power Suppy" is a placeholder value for everything except known combination so the real value could be different.
> 
> _But it's hard to read history of attributes so I won't bother you w/ that._
> 
> My suspicion is that your inverter simply does not show correct state of the Relay and thus also shows incorrect state of the Grid sensor.
> 
> Did you tried to restart it or smthing?

I have restarted it couple of times after update to latest version of this hacs. Solarman and deyecloud alway show grid off status properly.
 
This issue started to reproduce right after update so something definatelly changed in the hacs code.

davidrapan, 2024-12-30:
I was talking about restarting your inverter and stick logger cause they are giving out incorrect value.

> This issue started to reproduce right after update so something definatelly changed in the hacs code.

It did not, sorry,

iwannatalk, 2024-12-30:
> I was talking about restarting your inverter and stick logger cause they are giving out incorrect value.
> 
> > This issue started to reproduce right after update so something definatelly changed in the hacs code.
> 
> It didn't, sorry,
> 
> I mean you can try to revert back to version you think works and we will see. 😉

Ye, i understood that you are taking about restarting of inverter. It was restarted couple of times.
Also deye support has updated logger and inverter with latest sf version so it was updated today and after that there was couple of outages :)

Ok, i will play with reverting to previous version to see whether the issue is fixed. But not sure when :)

davidrapan, 2024-12-30:
> Also deye support has updated logger and inverter with latest sf version so it was updated today and after that there was couple of outages :)

You should have said at the beginning. So proly they fuck*d it up. So what version you have now? Show me attributes of Device sensor.

iwannatalk, 2024-12-30:
> > Also deye support has updated logger and inverter with latest sf version so it was updated today and after that there was couple of outages :)
> 
> You should have start w/ this! So proly they fuck*d it up. So what version you have now? Show me attributes of Device sensor.
> 
> _Don't forget to hide serial number._

Issue started at 23rd of Dec, when i have updated the hacs, not today, so this is not related to inverter sw update (performed today).

davidrapan, 2024-12-30:
And I'm telling you that nothing about this changed and sensor shows correct value according what ac relay register number 552 says.

Value: 7 which is 111 in binary, bit0 is inverter relay and is set, bit1 is undefined and is set, bit2 is grid relay and is set.

closed 2025-02-13 by github-actions[bot]