davidrapan/ha-solarman #358: My inverter goes off at night and no data comes from smart meter

nazeg, 2025-01-05:
Hi all, maybe it's not related with this plugin exactly but i just bought Deye 10kw string inverter and it goes off when there's no sun. So i can't get info from smart meter till sunshine. Is  there any settings on the inverter so i can get data 24/7?

davidrapan, 2025-01-07:
Hi @nazeg, I have honestly no idea what do you mean.

nazeg, 2025-01-07:
> Hi @nazeg, I have honestly no idea what do you mean.

As I understand it, on-grid inverters don’t work at night without PV power, so the inverter doesn’t send any data to Home Assistant or Solarpower because it’s in an off state. I’m wondering if there are any settings on this inverter model to address this. Perhaps this explanation makes it clearer:

https://diysolarforum.com/threads/my-deye-inverter-goes-off-at-night-and-no-data-comes-from-smart-meter.97381/#post-1312723

[image]

davidrapan, 2025-01-07:
> As I understand it, on-grid inverters don’t work at night without PV power.

Microinverters does not but hybrid inverters do work at night. 😉

Better tell me the exact model of your device.

nazeg, 2025-01-07:
> > As I understand it, on-grid inverters don’t work at night without PV power.
> 
> Microinverters does not but hybrid inverters do work at night. 😉
> 
> Better tell me the exact model of your device.

My model is not microinverter, it's Deye SUN-10K-G06P3-EU-AM2 10kW string inverter

It's link: https://www.deyeinverter.com/product/three-phase-string-inverter/sun3-4-5-6-7-8-9-10-12-15kg06p3euam2-315kw-three-phase-2-mppt.html

davidrapan, 2025-01-07:
String inverters are not able to draw energy from the grid either. _I think_.

nazeg, 2025-01-07:
> String inverters are not able to draw energy from the grid either. _I think_.

Maybe there's a setting about this but i couldn't find :( I just want to see the other datas at night aswell.

davidrapan, 2025-01-07:
It's hardware limited not through software.

nazeg, 2025-01-07:
> It's hardware limited not through software.

I understand, thanks anyway.

closed 2025-01-07 by davidrapan