davidrapan/ha-solarman #366: Can't connect to lSW3

AlexRux, 2025-01-12:
### Description

[image]
[image]

As pictures and log says. 

What to do?

### Reproduction Steps

Configuring as picture. 
[image]

### Log

2025-01-12 12:48:59.365 ERROR (MainThread) [custom_components.solarman.coordinator] Unexpected error fetching Inverter data
Traceback (most recent call last):
  File "/config/custom_components/solarman/api.py", line 55, in load
    self.device_info = await self.profile.resolve(self.get)
                       ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/solarman/provider.py", line 120, in resolve
    f = lookup_profile(await request(-1, set_request(*AUTODETECTION_REQUEST_DEYE)), self.attributes) if self.auto and request else self.filename
                       ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/solarman/api.py", line 134, in get
    responses[(code, start)] = await self.try_read_write(code, start, quantity, f"Querying {code:02X} ~ {start:04} - {end:04} | 0x{start:04X} - 0x{end:04X} #{quantity:03}")
                               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/solarman/api.py", line 101, in try_read_write
    if (response := await self.read_write(code, start, arg)) and (length := ilen(response)) is None and (expected := arg if code < CODE.WRITE_SINGLE_COIL else 1) and length != expected:
                    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/solarman/api.py", line 78, in read_write
    return await self.modbus.read_holding_registers(start, arg)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/solarman/include/pysolarmanv5/pysolarman.py", line 87, in read_holding_registers
    return await super().read_holding_registers(register_addr, quantity)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/solarman/include/pysolarmanv5/pysolarmanv5_async.py", line 342, in read_holding_registers
    modbus_values = await self._get_modbus_response(mb_request_frame)
                    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/solarman/include/pysolarmanv5/pysolarmanv5_async.py", line 298, in _get_modbus_response
    mb_response_frame = await self._send_receive_modbus_frame(mb_request_frame)
                        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/solarman/include/pysolarmanv5/pysolarmanv5_async.py", line 286, in _send_receive_modbus_frame
    mb_response_frame = self._v5_frame_decoder(v5_response_frame)
  File "/config/custom_components/solarman/include/pysolarmanv5/pysolarmanv5.py", line 264, in _v5_frame_decoder
    raise V5FrameError(f"V5 Modbus EXCEPTION: {err.__name__}")
custom_components.solarman.include.pysolarmanv5.pysolarmanv5.V5FrameError: V5 Modbus EXCEPTION: AcknowledgeError
The above exception was the direct cause of the following exception:
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/update_coordinator.py", line 318, in __wrap_async_setup
    await self._async_setup()
  File "/usr/src/homeassistant/homeassistant/helpers/update_coordinator.py", line 351, in _async_setup
    return await self.setup_method()
           ^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/solarman/api.py", line 58, in load
    raise Exception(f"[{self.config.serial}] Device setup failed. [{format_exception(e)}]") from e
Exception: [2381131926] Device setup failed. [V5FrameError: V5 Modbus EXCEPTION: AcknowledgeError]

### Version

24.12.22

### Home Assistant Core Version

2025.1.2

### Home Assistant Operating System Version

14.1

davidrapan, 2025-01-12:
Hi @AlexRux, you need to select appropriate profile for your device. Look into [Supported Devices](https://github.com/davidrapan/ha-solarman/wiki/Supported-Devices).

_Auto is only for Deye compatible devices._

AlexRux, 2025-01-12:
> Hi @AlexRux, you need to select appropriate profile for your device. Look into [Supported Devices](https://github.com/davidrapan/ha-solarman/wiki/Supported-Devices)
> 
> _Auto is only for Deye compatible devices
My bad. You can close

closed 2025-01-12 by davidrapan