davidrapan/ha-solarman #367: Fuji Microinverter 2.2kW is disconnecting

dcigir, 2025-01-12:
### Description

I have Fuji Microinverter 2.2kW that goes offline (Disconnected) every few minutes.
I also use Fuji String inverter 10kW, with LSW3 logger and that Inverter is working totally fine.
Wifi connection are great as I use Unifi system all over my house. I was using this Microinverter on the previous Solarman Integration from StephanJoubert and I didn't have these issues.

### Reproduction Steps

I just added Inverter to the integration.

### Log

This error originated from a custom integration.

Logger: custom_components.solarman.coordinator
Source: helpers/update_coordinator.py:385
integration: Solarman (documentation, issues)
First occurred: 11:42:44 AM (20 occurrences)
Last logged: 12:57:41 PM

Timeout fetching Solarman 2 data

### Version

24.12.22

### Home Assistant Core Version

core-2025.1.2

### Home Assistant Operating System Version

Home Assistant OS 14.1

davidrapan, 2025-01-12:
Hi @dcigir, you gonna have to enable debug logging to help me investigate anything. 😉

dcigir, 2025-01-12:
Sure, but I don't have any clue how to do that :)

davidrapan, 2025-01-12:
I see several attempts to read data from your microinverter in the log, but it just doesn't get any response. Is microinverter connected to solarman servers?

dcigir, 2025-01-12:
Yes
- [image]

davidrapan, 2025-01-13:
Could you try to disconnect it from them for couple of hours to check if the issues are caused by saturation?

vymak, 2025-01-24:
I had a similar problem with a Sofar HYD 10KTL-3PH inverter, where my **LSW3** logger kept disconnecting for some time. Then I got annoyed and connected the **LSE3** (LAN) logger to the inverter. Since then I have only had one or two connection failures (usually only 15 seconds).

[image]

davidrapan, 2025-01-24:
Hi @vymak, what fw version you are running on your LSE3?

vymak, 2025-01-24:
Hi @davidrapan 

[image]

davidrapan, 2025-01-24:
Great, this is the most stable fw to my knowledge. 👍

_Still wonder what could possibly cause those few disconnects..._

closed 2025-03-03 by dcigir