davidrapan/ha-solarman #369: Time of use changes

titia137, 2025-01-13:
### Description

Hello,
I have a Deye 12kW inverter (model SG04LP3) and use Node-RED for automation.
I'm trying to configure the entire "Time of Use" table. After making the changes, everything looks fine (all the changes I made appear correctly in the Home Assistant entities). (I use time.set_value). However, after a few minutes, the values for "Deye Program 5 Time" and "Deye Program 6 Time" revert to their previous settings. The values for "Deye Program 1-4 Time" and "charging" and "SOC" remain good.

### Reproduction Steps

  Program times 5 and 6 revert to their previous values on their own.

### Log

[ha.log](https://github.com/user-attachments/files/18401691/ha.log)

### Version

24.12.10_2

### Home Assistant Core Version

2025.1.2

### Home Assistant Operating System Version

13.2

davidrapan, 2025-01-15:
Hi @titia137, I have the same device and time entities controlling TOU working just fine for me. In addition it would also make no sense that only some work and other don't.

Make sure you are trying to set correct values:
* Program 4 Time < Program 5 Time
* Program 5 Time < Program 6 Time
* Program 6 Time < Program 1 Time (next day)

titia137, 2025-01-15:
Deye Program 6 Time změněno na 21:00 - this is by itself
20:40:29 - Před 1 minutou
Deye Program 5 Time změněno na 20:05 - this is by itself
20:40:29 - Před 1 minutou
Deye Date & Time změněno na 15. ledna 2025 v 20:40
20:40:29 - Před 1 minutou
Deye Date & Time změněno na 15. ledna 2025 v 20:39
20:39:27 - Před 2 minutami
Deye Date & Time změněno na 15. ledna 2025 v 20:38
20:38:24 - Před 3 minutami
Deye Program 6 Time změněno na 23:55 spuštěno akcí Čas: Nastavit čas - this is by Node-red
20:38:09 - Před 3 minutami - Supervisor
Deye Program 5 Time změněno na 23:50 spuštěno akcí Čas: Nastavit čas - this is by Node-red
20:38:09 - Před 3 minutami - Supervisor
Deye Program 4 Time změněno na 20:00 spuštěno akcí Čas: Nastavit čas - this is by Node-red
20:37:41 - Před 4 minutami - Supervisor
Deye Program 3 Time změněno na 16:00 spuštěno akcí Čas: Nastavit čas - this is by Node-red
20:37:41 - Před 4 minutami - Supervisor

davidrapan, 2025-01-15:
That just means that it was changed in the inverter, not that it was changed by itself (or by integration). 😉

titia137, 2025-01-26:
I tried different settings – directly in HA – and encountered the same issue. When the order was different, only the last values weren’t recorded. So, I added a time delay after setting ten values, and now it works. It seems like the inverter can’t handle receiving all 18 values at once.
Anyway, thank you for your responses and for your work.

closed 2025-01-26 by titia137

davidrapan, 2025-01-26:
I'm really glad you were able to solve it! And thank you for reporting back what the issue was. 😉

martinmildner, 2025-04-12:
hi @titia137,
I'm also working on automating my Deye inverter.
Would you be willing to share the relevant part of your Node-RED flow or the code you’re using for time.set_value?
thanks!!
martin