davidrapan/ha-solarman #377: Invalid unique_id 'inverter_tag' of type int when a string is expected

catalinstroie, 2025-01-27:
### Description

The only errors my HA logs are reporting are relating to the solarman integration:
2025-01-26 20:51:14.905 WARNING (SyncWorker_0) [homeassistant.loader] We found a custom integration dreame_vacuum which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
2025-01-26 20:51:14.951 ERROR (MainThread) [homeassistant.config_entries] Config entry 'Inverter' from integration solarman has an invalid unique_id '[my_id]' of type int when a string is expected, please create a bug report at https://github.com/davidrapan/ha-solarman/issues

### Reproduction Steps

Just install and configure the integration.

### Log

2025-01-26 20:51:14.905 WARNING (SyncWorker_0) [homeassistant.loader] We found a custom integration dreame_vacuum which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
2025-01-26 20:51:14.951 ERROR (MainThread) [homeassistant.config_entries] Config entry 'Inverter' from integration solarman has an invalid unique_id '[my_id]' of type int when a string is expected, please create a bug report at https://github.com/davidrapan/ha-solarman/issues

### Version

24.12.22

### Home Assistant Core Version

2025.1.4

### Home Assistant Operating System Version

14.1

luigicarfagnini, 2025-02-01:
same issue today

davidrapan, 2025-02-01:
New release is not out yet so that's why and it's just a warning, you can ignore it for now. 😉

zollak, 2025-02-08:
This error still exists in the HA Core 2025.2.1 update. I'm using the currently latest Solarman Stick Logger addon with version v24.12.22.

closed 2025-02-09 by davidrapan