davidrapan/ha-solarman #385: HA conflict with Solarman and Deyecloud clouds

Mira141, 2025-02-03:
### Description

With this integration active in HA, it is a big problem to read data and give commands to the inverter using the Solarman and Deyecloud clouds. Usually one of 5-10 attempts is successful, the timeout of each is 2 minutes, so it is unusable and frustrating. When the HA is disconnected from the network, it is reliable, so there is probably a conflict in the WiFi stick of the inverter. Would you be willing to adapt this integration via the MODbus port of the inverter using a simple USB(HA) - RS485(DEYE) converter? Or another suggestion: Where can the HA-inverter communication intensity be set to longer intervals so that the clouds have a chance to communicate?

### Reproduction Steps

Simply compare the inverter's response to read and write through clouds with HA turned off and on with this integration.

### Log

Not needed.

### Version

24.12.22

### Home Assistant Core Version

2025.1.4

### Home Assistant Operating System Version

14.2

davidrapan, 2025-02-03:
Use one or the other. Solarman sticks and underlying protocol are not really good. There's nothing we can do about this, except increase the refresh intervals in the yaml file used, but you'll lose real-time data, which defeats the core purpose of this integration.

Partiolainen, 2025-02-03:
@Mira141 
In my case Solarman App showed battery status as offline
Increasing the interval from 5 to 10 seconds solves the problem

kkmaslowski, 2025-03-25:
@Partiolainen how to increase timeout?

Partiolainen, 2025-03-25:
@kkmaslowski 
I didn't see any options for this, so I edit the configuration file related to my inverter manually (after each update)

Mira141, 2025-03-27:
@kkmaslowski 
I had to manually rewrite all 5 second intervals to 10 sec. in the inverter definition in the yaml file using the HA integrated file editor. It has to be done after every update. Unfortunately, it is not convenient... 10 sec. seems to be good compromise when using Solarman or Deyecloud app for comparing results during debugging my own changes ...

kkmaslowski, 2025-03-27:
@davidrapan maybe change it by default to 10s for LSW logger? We have the same problem. I tried to disable cloud and still got timeouts with 5s.

Partiolainen, 2025-03-27:
After the last update I set it to 7 seconds and it seems to be enough

Mira141, 2025-03-28:
> [@davidrapan](https://github.com/davidrapan) maybe change it by default to 10s for LSW logger? We have the same problem. I tried to disable cloud and still got timeouts with 5s.

I understand, that David has other priorities regarding development and this workaround by editing yaml file is a solution, but no comfortable...

davidrapan, 2025-04-03:
My LSW (w/ cloud connected) is working just fine w/ current update interval.