davidrapan/ha-solarman #386: Wrong response

githubDante, 2025-02-03:
### Description

An unwanted response can be returned [here ](https://github.com/davidrapan/ha-solarman/blob/d5937cd5e259bf1435b9e8706ed421d41de9bed2/custom_components/solarman/inverter.py#L95).  It happens when the `if` statement in the try/except block raises an error and `attempts_left` > 0. It's visible in the log shared in #316 

> Querying 03 ~ 2114 - 2119 | 0x0842 - 0x0847 #006 failed, attempts left: 4 [ValueError: The truth value of an array with more than one element is ambiguous. Use a.any() or a.all()]

The while loop is interrupted after this point as the response variable is not None anymore

### Reproduction Steps

Log examination

### Log

```python
        while attempts_left > 0 and response is None:
            attempts_left -= 1
            try:
                if (response := await self.modbus.read_write(code, start, arg)) is not None and (length := ilen(response)) is not None and (expected := arg if code < CODE.WRITE_SINGLE_COIL else 1) is not None and length != expected:
                    raise Exception(f"[{self.config.serial}] Unexpected response: Invalid length! (Length: {length}, Expected: {expected})")

                _LOGGER.debug(f"[{self.config.serial}] {message} succeeded, response: {response}")
            except Exception as e:
                _LOGGER.debug(f"[{self.config.serial}] {message} failed, attempts left: {attempts_left}{'' if attempts_left > 0 else ', aborting.'} [{format_exception(e)}]")

                if isinstance(e, TimeoutError):
                    await self.endpoint.discover(True)
                if not attempts_left > 0:
                    raise

                await asyncio.sleep(TIMINGS_WAIT_SLEEP)

        return response
```

### Version

git@main

### Home Assistant Core Version

2025<something>

### Home Assistant Operating System Version

?!?

davidrapan, 2025-02-03:
Hi! Are you referring to the log alone or did you run the code yourself? Because I believe that the error was due to missing `is not None` in [v24.12.22](https://github.com/davidrapan/ha-solarman/releases/tag/v24.12.22) (the guy in mentioned post proly doesn't run main):
```
(response := await self.modbus.read_write(code, start, arg)) is not None
```
but should be fixed in main.

githubDante, 2025-02-03:
> Hi! Are you referring to the log alone or did you run the code yourself? Because I believe that the error was due to missing is not None in [v24.12.22](https://github.com/davidrapan/ha-solarman/releases/tag/v24.12.22) 

Judging by what I saw in the log. Proper testing reveal that it doesn't fail like that at the  moment (due to the added is not None). Also the error message is generated specifically with numpy array.

Closing this.

closed 2025-02-03 by githubDante