davidrapan/ha-solarman #395: DhcpServiceInfo a deprecated constant

haysdb, 2025-02-07:
### Description

Saw this in the HA log:

`2025-02-06 23:54:24.771 WARNING (ImportExecutor_0) [homeassistant.components.dhcp] DhcpServiceInfo was used from solarman, this is a deprecated constant which will be removed in HA Core 2026.2. Use homeassistant.helpers.service_info.dhcp.DhcpServiceInfo instead, please report it to the author of the 'solarman' custom integration`

### Reproduction Steps

Saw this in the log

### Log

`2025-02-06 23:54:24.771 WARNING (ImportExecutor_0) [homeassistant.components.dhcp] DhcpServiceInfo was used from solarman, this is a deprecated constant which will be removed in HA Core 2026.2. Use homeassistant.helpers.service_info.dhcp.DhcpServiceInfo instead, please report it to the author of the 'solarman' custom integration`

### Version

24.12.22

### Home Assistant Core Version

2025.2.0

### Home Assistant Operating System Version

14.2

davidrapan, 2025-02-07:
Hi @haysdb, thanks! 👍

davidrapan, 2025-02-07:
[fix: Deprecation of homeassistant.components.dhch.DhcpServiceInfo](https://github.com/davidrapan/ha-solarman/commit/c7a54850ba3a710d114dff2b239d0dd11ed54ef3)

closed 2025-02-09 by davidrapan