davidrapan/ha-solarman #400: Solarman 25.02.10 breaks initialization for Deye SG04LP3 and LSE3 Ethernet logger.

jkolo, 2025-02-09:
### Description

After update Solarman to version 25.02.10, integration can't initializing. I have Deye Hybrid SG04LP3 Inverter and Ethernet logger LSE3. 

### Reproduction Steps

Update Solarman to 25.02.10. 

### Log

2025-02-09 22:07:36.338 DEBUG (MainThread) [custom_components.solarman] async_setup_entry({'created_at': '2024-10-31T23:58:26.896638+00:00', 'data': {'battery_life_cycle_rating': 6000, 'battery_nominal_voltage': 48, 'inverter_host': '10.38.107.18', 'inverter_mb_slave_id': 1, 'inverter_port': 8899, 'inverter_serial': 2110645731, 'lookup_file': 'deye_sg04lp3.yaml', 'name': 'SUN-12K-SG04-LP3'}, 'discovery_keys': {}, 'disabled_by': None, 'domain': 'solarman', 'entry_id': '01JBJG892GNQX5T5R0RYX1VZ2Q', 'minor_version': 6, 'modified_at': '2025-02-09T21:07:23.585065+00:00', 'options': {'host': '10.38.107.18', 'additional_options': {'battery_nominal_voltage': 51, 'mppt': 2, 'pack': 1, 'battery_life_cycle_rating': 8000, 'mod': True}}, 'pref_disable_new_entities': False, 'pref_disable_polling': False, 'source': 'user', 'title': 'SUN-12K-SG04-LP3', 'unique_id': 'solarman_2110645731', 'version': 1})
2025-02-09 22:07:36.339 DEBUG (MainThread) [custom_components.solarman] async_setup_entry: config_entry.runtime_data.async_config_entry_first_refresh
2025-02-09 22:07:36.339 DEBUG (MainThread) [custom_components.solarman.discovery] _discover: Broadcasting on 10.38.107.18
2025-02-09 22:07:36.339 DEBUG (MainThread) [custom_components.solarman.discovery] DiscoveryProtocol: Send to 10.38.107.18
2025-02-09 22:07:36.346 DEBUG (MainThread) [custom_components.solarman.discovery] DiscoveryProtocol: [10.38.107.18, 402A8F2CD64B, 2110645731]
2025-02-09 22:07:36.350 DEBUG (MainThread) [custom_components.solarman.provider] Device autodetection is enabled
2025-02-09 22:07:36.350 DEBUG (MainThread) [custom_components.solarman.device] [2110645731] Scheduling 1 query request. ^-1
2025-02-09 22:07:36.351 DEBUG (MainThread) [custom_components.solarman.device] [2110645731] Querying 03 ❘ 0x03 ~ 0000 - 0022 ❘ 0x0000 - 0x0016 #023 ...
2025-02-09 22:07:36.355 DEBUG (MainThread) [custom_components.solarman.discovery] DiscoveryProtocol: Connection closed
2025-02-09 22:07:36.363 DEBUG (MainThread) [custom_components.solarman.pysolarman.pysolarman] [2110645731] SENT: a5 17 00 10 45 15 00 e3 e5 cd 7d 02 00 00 00 00 00 00 00 00 00 00 00 00 00 00 01 03 00 00 00 17 05 c4 79 15
2025-02-09 22:07:37.402 DEBUG (MainThread) [custom_components.solarman.pysolarman.pysolarman] [2110645731] PROTOCOL_MISMATCH: a5 17 00 10 45 03 00 e3 02
2025-02-09 22:07:50.365 DEBUG (MainThread) [custom_components.solarman.pysolarman.pysolarman] [2110645731] Peer not responding. Closing connection
2025-02-09 22:07:50.366 DEBUG (MainThread) [custom_components.solarman.pysolarman.pysolarman] [2110645731] Connection closed by the remote. Will try to restart the connection
2025-02-09 22:07:50.369 DEBUG (MainThread) [custom_components.solarman.pysolarman.pysolarman] [2110645731] Successful reconnection! Data expected. Will retry the last request
2025-02-09 22:07:51.434 DEBUG (MainThread) [custom_components.solarman.pysolarman.pysolarman] [2110645731] PROTOCOL_MISMATCH: a5 17 00 10 45 03 00 e3 02
2025-02-09 22:07:56.352 INFO (MainThread) [custom_components.solarman.device] [2110645731] Closing connection to 10.38.107.18
2025-02-09 22:07:56.355 DEBUG (MainThread) [custom_components.solarman.coordinator] async_shutdown
2025-02-09 22:07:56.355 INFO (MainThread) [custom_components.solarman.device] [2110645731] Closing connection to 10.38.107.18
2025-02-09 22:08:06.663 DEBUG (MainThread) [custom_components.solarman] async_setup_entry({'created_at': '2024-10-31T23:58:26.896638+00:00', 'data': {'battery_life_cycle_rating': 6000, 'battery_nominal_voltage': 48, 'inverter_host': '10.38.107.18', 'inverter_mb_slave_id': 1, 'inverter_port': 8899, 'inverter_serial': 2110645731, 'lookup_file': 'deye_sg04lp3.yaml', 'name': 'SUN-12K-SG04-LP3'}, 'discovery_keys': {}, 'disabled_by': None, 'domain': 'solarman', 'entry_id': '01JBJG892GNQX5T5R0RYX1VZ2Q', 'minor_version': 6, 'modified_at': '2025-02-09T21:07:23.585065+00:00', 'options': {'host': '10.38.107.18', 'additional_options': {'battery_nominal_voltage': 51, 'mppt': 2, 'pack': 1, 'battery_life_cycle_rating': 8000, 'mod': True}}, 'pref_disable_new_entities': False, 'pref_disable_polling': False, 'source': 'user', 'title': 'SUN-12K-SG04-LP3', 'unique_id': 'solarman_2110645731', 'version': 1})
2025-02-09 22:08:06.664 DEBUG (MainThread) [custom_components.solarman] async_setup_entry: config_entry.runtime_data.async_config_entry_first_refresh
2025-02-09 22:08:06.664 DEBUG (MainThread) [custom_components.solarman.discovery] _discover: Broadcasting on 10.38.107.18
2025-02-09 22:08:06.665 DEBUG (MainThread) [custom_components.solarman.discovery] DiscoveryProtocol: Send to 10.38.107.18
2025-02-09 22:08:06.671 DEBUG (MainThread) [custom_components.solarman.discovery] DiscoveryProtocol: [10.38.107.18, 402A8F2CD64B, 2110645731]
2025-02-09 22:08:06.677 DEBUG (MainThread) [custom_components.solarman.provider] Device autodetection is enabled
2025-02-09 22:08:06.677 DEBUG (MainThread) [custom_components.solarman.device] [2110645731] Scheduling 1 query request. ^-1
2025-02-09 22:08:06.677 DEBUG (MainThread) [custom_components.solarman.device] [2110645731] Querying 03 ❘ 0x03 ~ 0000 - 0022 ❘ 0x0000 - 0x0016 #023 ...
2025-02-09 22:08:06.680 DEBUG (MainThread) [custom_components.solarman.discovery] DiscoveryProtocol: Connection closed
2025-02-09 22:08:06.685 DEBUG (MainThread) [custom_components.solarman.pysolarman.pysolarman] [2110645731] SENT: a5 17 00 10 45 88 00 e3 e5 cd 7d 02 00 00 00 00 00 00 00 00 00 00 00 00 00 00 01 03 00 00 00 17 05 c4 ec 15
2025-02-09 22:08:07.768 DEBUG (MainThread) [custom_components.solarman.pysolarman.pysolarman] [2110645731] PROTOCOL_MISMATCH: a5 17 00 10 45 03 00 e3 02
2025-02-09 22:08:20.686 DEBUG (MainThread) [custom_components.solarman.pysolarman.pysolarman] [2110645731] Peer not responding. Closing connection
2025-02-09 22:08:20.688 DEBUG (MainThread) [custom_components.solarman.pysolarman.pysolarman] [2110645731] Connection closed by the remote. Will try to restart the connection
2025-02-09 22:08:20.692 DEBUG (MainThread) [custom_components.solarman.pysolarman.pysolarman] [2110645731] Successful reconnection! Data expected. Will retry the last request
2025-02-09 22:08:21.768 DEBUG (MainThread) [custom_components.solarman.pysolarman.pysolarman] [2110645731] PROTOCOL_MISMATCH: a5 17 00 10 45 03 00 e3 02
2025-02-09 22:08:26.678 INFO (MainThread) [custom_components.solarman.device] [2110645731] Closing connection to 10.38.107.18
2025-02-09 22:08:26.681 DEBUG (MainThread) [custom_components.solarman.coordinator] async_shutdown
2025-02-09 22:08:26.681 INFO (MainThread) [custom_components.solarman.device] [2110645731] Closing connection to 10.38.107.18

### Version

25.02.10

### Home Assistant Core Version

2025.1.2

### Home Assistant Operating System Version

14.2

davidrapan, 2025-02-09:
Hi @jkolo, thank you for your report!

[fix: Detection of Ethernet Logger](https://github.com/davidrapan/ha-solarman/commit/c1185ed42a054251be4919a80f9dcaf510cf09be)

jkolo, 2025-02-09:
Thank you for your quick response, I look forward to the new version :)

davidrapan, 2025-02-09:
[v25.02.10_01](https://github.com/davidrapan/ha-solarman/releases/tag/v25.02.10_01)

closed 2025-02-09 by davidrapan

jkolo, 2025-02-09:
Unfortunately initialization still failed. 

[image]
```

2025-02-10 00:50:13.168 DEBUG (MainThread) [custom_components.solarman] async_setup_entry({'created_at': '2024-10-31T23:58:26.896638+00:00', 'data': {'battery_life_cycle_rating': 6000, 'battery_nominal_voltage': 48, 'inverter_host': '10.38.107.18', 'inverter_mb_slave_id': 1, 'inverter_port': 8899, 'inverter_serial': 2110645731, 'lookup_file': 'deye_sg04lp3.yaml', 'name': 'SUN-12K-SG04-LP3'}, 'discovery_keys': {}, 'disabled_by': None, 'domain': 'solarman', 'entry_id': '01JBJG892GNQX5T5R0RYX1VZ2Q', 'minor_version': 6, 'modified_at': '2025-02-09T23:17:48.005638+00:00', 'options': {'additional_options': {'battery_life_cycle_rating': 8000, 'battery_nominal_voltage': 51, 'mod': True, 'mppt': 2, 'pack': 1}, 'host': '10.38.107.18'}, 'pref_disable_new_entities': False, 'pref_disable_polling': False, 'source': 'user', 'title': 'SUN-12K-SG04-LP3', 'unique_id': 'solarman_2110645731', 'version': 1})
2025-02-10 00:50:13.168 DEBUG (MainThread) [custom_components.solarman] async_setup_entry: config_entry.runtime_data.async_config_entry_first_refresh
2025-02-10 00:50:13.169 DEBUG (MainThread) [custom_components.solarman.discovery] _discover: Broadcasting on 10.38.107.18
2025-02-10 00:50:13.170 DEBUG (MainThread) [custom_components.solarman.discovery] DiscoveryProtocol: Send to 10.38.107.18
2025-02-10 00:50:13.172 DEBUG (MainThread) [custom_components.solarman.discovery] DiscoveryProtocol: [10.38.107.18, 402A8F2CD64B, 2110645731]
2025-02-10 00:50:13.174 DEBUG (MainThread) [custom_components.solarman.provider] Device autodetection is enabled
2025-02-10 00:50:13.174 DEBUG (MainThread) [custom_components.solarman.device] [2110645731] Scheduling 1 query request. ^-1
2025-02-10 00:50:13.175 DEBUG (MainThread) [custom_components.solarman.device] [2110645731] Querying 03 ❘ 0x03 ~ 0000 - 0022 ❘ 0x0000 - 0x0016 #023 ...
2025-02-10 00:50:13.175 DEBUG (MainThread) [custom_components.solarman.discovery] DiscoveryProtocol: Connection closed
2025-02-10 00:50:13.180 DEBUG (MainThread) [custom_components.solarman.pysolarman.pysolarman] [2110645731] SENT: a5 17 00 10 45 f9 00 e3 e5 cd 7d 02 00 00 00 00 00 00 00 00 00 00 00 00 00 00 01 03 00 00 00 17 05 c4 5d 15
2025-02-10 00:50:14.274 DEBUG (MainThread) [custom_components.solarman.pysolarman.pysolarman] [2110645731] TCP_DETECTED: a5 17 00 10 45 03 00 e3 02
2025-02-10 00:50:14.274 DEBUG (MainThread) [custom_components.solarman.pysolarman.pysolarman] [2110645731] RECD: a5 17 00 10 45 03 00 e3 02
2025-02-10 00:50:14.274 DEBUG (MainThread) [custom_components.solarman.device] [2110645731] Querying 03 ❘ 0x03 ~ 0000 - 0022 ❘ 0x0000 - 0x0016 #023 failed, attempts left: 4 [FrameError: Incorrect control code]
2025-02-10 00:50:14.274 DEBUG (MainThread) [custom_components.solarman.pysolarman.pysolarman] [2110645731] SENT: 3b 68 00 00 00 06 01 03 00 00 00 17
2025-02-10 00:50:14.575 DEBUG (MainThread) [custom_components.solarman.pysolarman.pysolarman] [2110645731] RECD: 3b 68 00 00 00 31 01 03 2e 00 05 00 01 01 04 32 34 30 35 32 33 36 37 31 33 00 00 00 00 00 00 18 07 00 00 00 00 20 06 11 44 00 00 10 01 c0 47 00 00 d4 c0 00 01 ff ff
2025-02-10 00:50:14.576 DEBUG (MainThread) [pymodbus.logging] Processing: 0x3b 0x68 0x0 0x0 0x0 0x31 0x1 0x3 0x2e 0x0 0x5 0x0 0x1 0x1 0x4 0x32 0x34 0x30 0x35 0x32 0x33 0x36 0x37 0x31 0x33 0x0 0x0 0x0 0x0 0x0 0x0 0x18 0x7 0x0 0x0 0x0 0x0 0x20 0x6 0x11 0x44 0x0 0x0 0x10 0x1 0xc0 0x47 0x0 0x0 0xd4 0xc0 0x0 0x1 0xff 0xff
2025-02-10 00:50:14.576 DEBUG (MainThread) [pymodbus.logging] decoded PDU function_code(3 sub -1) -> ReadHoldingRegistersResponse(dev_id=0, transaction_id=0, address=0, count=0, bits=[], registers=[5, 1, 260, 12852, 12341, 12851, 13879, 12595, 0, 0, 0, 6151, 0, 0, 8198, 4420, 0, 4097, 49223, 0, 54464, 1, 65535], status=1) 
2025-02-10 00:50:14.576 DEBUG (MainThread) [pymodbus.logging] Frame advanced, resetting header!!
2025-02-10 00:50:14.576 DEBUG (MainThread) [custom_components.solarman.device] [2110645731] Querying 03 ❘ 0x03 ~ 0000 - 0022 ❘ 0x0000 - 0x0016 #023 succeeded
2025-02-10 00:50:14.576 DEBUG (MainThread) [custom_components.solarman.device] [2110645731] Returning 1 new value. [Previous State: Disconnected (-1)]
2025-02-10 00:50:14.576 DEBUG (MainThread) [custom_components.solarman.device] [2110645731] Scheduling 1 query request. ^-1
2025-02-10 00:50:14.577 DEBUG (MainThread) [custom_components.solarman.device] [2110645731] Querying 03 ❘ 0x03 ~ 10002 - 10003 ❘ 0x2712 - 0x2713 #002 ...
2025-02-10 00:50:14.577 DEBUG (MainThread) [custom_components.solarman.pysolarman.pysolarman] [2110645731] SENT: 55 d5 00 00 00 06 01 03 27 12 00 02
2025-02-10 00:50:14.828 DEBUG (MainThread) [custom_components.solarman.pysolarman.pysolarman] [2110645731] RECD: 55 d5 00 00 00 07 01 03 04 00 03 02 03
2025-02-10 00:50:14.828 DEBUG (MainThread) [pymodbus.logging] Processing: 0x55 0xd5 0x0 0x0 0x0 0x7 0x1 0x3 0x4 0x0 0x3 0x2 0x3
2025-02-10 00:50:14.828 DEBUG (MainThread) [pymodbus.logging] decoded PDU function_code(3 sub -1) -> ReadHoldingRegistersResponse(dev_id=0, transaction_id=0, address=0, count=0, bits=[], registers=[3, 515], status=1) 
2025-02-10 00:50:14.829 DEBUG (MainThread) [pymodbus.logging] Frame advanced, resetting header!!
2025-02-10 00:50:14.829 DEBUG (MainThread) [custom_components.solarman.device] [2110645731] Querying 03 ❘ 0x03 ~ 10002 - 10003 ❘ 0x2712 - 0x2713 #002 succeeded
[... cut ...]
2025-02-10 00:50:22.779 DEBUG (MainThread) [custom_components.solarman.device] [2110645731] Returning 1 new value. [Previous State: Connected (1)]
2025-02-10 00:50:23.782 DEBUG (MainThread) [custom_components.solarman.parser] Defaults for update_interval: 60, code: 3, min_span: 25, max_size: 125, digits: 6
2025-02-10 00:50:23.804 DEBUG (MainThread) [custom_components.solarman.coordinator] async_shutdown
2025-02-10 00:50:23.804 INFO (MainThread) [custom_components.solarman.device] [2110645731] Closing connection to 10.38.107.18
```

davidrapan, 2025-02-10:
Can you show full log? I can't see what's wrong in there...

davidrapan, 2025-02-10:
Hi @jkolo, try [v25.02.10_02](https://github.com/davidrapan/ha-solarman/releases/tag/v25.02.10_02) and show me what's in the debug log please, thank you!

davidrapan, 2025-02-10:
🤯

Could you try to re-add device w/ enabled debug logging anyway so I can take better look at it? It's very strange..

Also, as you have batteries w/ some detailed values exposed through inverter can you help me w/ retrospective and show (few screenshots from HA) me those new sensors and devices? I was implementing this feature blindly because I don't have such batteries.

Thanks!

davidrapan, 2025-02-10:
[v25.02.10_03](https://github.com/davidrapan/ha-solarman/releases/tag/v25.02.10_03) should be finally a fix for your issue!

jkolo, 2025-02-11:
You are too fast for me :-) I have wife, children and lack of time ;-)

Some screenshots for you:

[image]

[image]

davidrapan, 2025-02-11:
Thanks @jkolo! [fix: Incorrect scaling of some battery sensors](https://github.com/davidrapan/ha-solarman/commit/fd92ebd46638084c61ccb1797964a52e9c23d991)