davidrapan/ha-solarman #401: sensor.inverter_generator_power not work since 24.12.16

chomats, 2025-02-10:
### Description

Since the version 24.12.16 the sensor (sensor.inverter_generator_power) is not provided.
It's work fine with the version 24.12.16 but the nexts versions don't provide this sensor.

Profile: deye_hybrid.yaml
Core
2025.2.1
Supervisor
2025.02.0
Operating System
14.2
Interface utilisateur
20250205.0

### Reproduction Steps

Update to 24.12.16 or the last version.

### Log

-

### Version

24.12.20 and next

### Home Assistant Core Version

2025.2.1

### Home Assistant Operating System Version

14.2

chomats, 2025-02-10:
[home-assistant_solarman_2025-02-10T14-12-32.181Z.log](https://github.com/user-attachments/files/18734991/home-assistant_solarman_2025-02-10T14-12-32.181Z.log)

davidrapan, 2025-02-10:
Hi @chomats, this sensor is named according to value selected in the IO Mode.

Options are:
* Generator
* SmartLoad
* Microinverter

Which means you have either **SmartLoad Power** or **Microinverter Power**.

davidrapan, 2025-02-10:
You can merge their history (if you have any and it's desired) according to the guide on the wiki!

closed 2025-02-10 by davidrapan