davidrapan/ha-solarman #402: Sofar HYD 10KTL-3PH - error during initialization

vymak, 2025-02-10:
### Description

After upgrading to version 25.02.10_02, Solarman integration does not initialize. See screenshot for error.

Inverter: **Sofar HYD 10KTL-3PH**

[image]

[image]

### Reproduction Steps

Upgrade to latest version and do HA restart.

### Log

[home-assistant_solarman_2025-02-10T18-14-45.658Z.log](https://github.com/user-attachments/files/18738850/home-assistant_solarman_2025-02-10T18-14-45.658Z.log)

### Version

25.02.10_02

### Home Assistant Core Version

2025.2.1

### Home Assistant Operating System Version

14.2

iwannatalk, 2025-02-10:
I got the same issue with my deye after update.

[image]

davidrapan, 2025-02-10:
Hi @vymak, thank you for your report - [fix: Register address typo in sofar_g3hyd profile](https://github.com/davidrapan/ha-solarman/commit/44455f60194538f00f67fadfd0fbc6c8ed3ee9c3)

@iwannatalk, you it's not the same issue. Open new one w/ attached debug log.

closed 2025-02-10 by davidrapan

davidrapan, 2025-02-10:
[v25.02.10_03](https://github.com/davidrapan/ha-solarman/releases/tag/v25.02.10_03)

vymak, 2025-02-11:
Hi @davidrapan ,
thanks a lot for the quick resolution of the problem. 

After updating to the latest version it works perfectly.