davidrapan/ha-solarman #403: Inverter is not initializing after update to v25.02.10_02

iwannatalk, 2025-02-10:
### Description

Inverter is not initializing after update to v25.02.10_02. The following error is shown:

[image]

### Reproduction Steps

Update Solarman to v25.02.10_02, restart HA.

### Log

[Errorlog.pdf](https://github.com/user-attachments/files/18740163/Errorlog.pdf)

### Version

25.02.10_02

### Home Assistant Core Version

2025.2.1

### Home Assistant Operating System Version

14.2

davidrapan, 2025-02-10:
Hi @iwannatalk, thank you for your report, you were really helpful! - [fix: Typos in scale of additional battery sensors for Deye](https://github.com/davidrapan/ha-solarman/commit/92be65cc873d1a1c211448ab12d04d262f2421ce)

closed 2025-02-10 by davidrapan

davidrapan, 2025-02-10:
[v25.02.10_03](https://github.com/davidrapan/ha-solarman/releases/tag/v25.02.10_03)