davidrapan/ha-solarman #404: Errors after upgrade

Pieter-HA, 2025-02-10:
### Description

After upgrading to the ltest version, I get a number of errors in the log.

### Reproduction Steps

Upgrade to latest version.

### Log

2025-02-10 21:46:44.720 ERROR (MainThread) [homeassistant.config_entries] Config entry 'Inverter' from integration solarman has an invalid unique_id '**redacted**' of type int when a string is expected, please create a bug report at https://github.com/davidrapan/ha-solarman/issues
2025-02-10 21:46:58.678 WARNING (MainThread) [homeassistant.components.sensor] Entity sensor.nexus_battery_remaining_capacity (<class 'custom_components.solarman.sensor.SolarmanSensor'>) is using state class 'measurement' which is impossible considering device class ('energy') it is using; expected None or one of 'total_increasing', 'total'; Please update your configuration if your entity is manually configured, otherwise create a bug report at https://github.com/davidrapan/ha-solarman/issues
2025-02-10 21:47:03.580 WARNING (MainThread) [py.warnings] /config/custom_components/solarman/device.py:102: RuntimeWarning: coroutine 'Solarman.open' was never awaited
  self.modbus.open()
2025-02-10 21:47:15.128 WARNING (MainThread) [py.warnings] /config/custom_components/solarman/device.py:102: RuntimeWarning: coroutine 'Solarman.open' was never awaited
  self.modbus.open()

### Version

25.02.10_02

### Home Assistant Core Version

2025.2.1

### Home Assistant Operating System Version

14.2

davidrapan, 2025-02-10:
Hi @Pieter-HA, first error is solvable proly only by deleting and re-adding device back and the second one is tied to used profile, which one is it?

Pieter-HA, 2025-02-10:
Pylon Force H

I’ll try delete and adding again.

Pieter-HA, 2025-02-10:
Delete and add device again helped a lot. This error remains:

Entity sensor.nexus_battery_remaining_capacity (<class
'custom_components.solarman.sensor.SolarmanSensor'>) is using state class
'measurement' which is impossible considering device class ('energy') it is
using; expected None or one of 'total_increasing', 'total'; Please update
your configuration if your entity is manually configured, otherwise create
a bug report at https://github.com/davidrapan/ha-solarman/issues

Not too hard to fix, I think.

davidrapan, 2025-02-10:
Yes this one is just minor detail but I don't think 'total_increasing' or 'total' are suitable. I guess it should be empty..

Pieter-HA, 2025-02-10:
Agree.

Thanks for your work, much appreciated!

davidrapan, 2025-02-10:
Thanks for your kind words! It should be all good in the latest version!

closed 2025-02-10 by davidrapan

oguzatagan, 2025-02-11:
hi david

i have same issue. what can i do? can i delete this?

[image]

oguzatagan, 2025-02-11:
thanks david. i did delete and error was gone.