davidrapan/ha-solarman #414: Other modules sensors are displayed

mamo-nick, 2025-02-13:
### Description

I have set up 3 modules, but sum sensors of all modules are displayed. (Not cells, cells sensors are OK - displayed only for 3 mod.)

[image]

### Reproduction Steps

inverter_definitions/pylontech_Force-H.yaml

### Log

[image]

### Version

25.02.12_04

### Home Assistant Core Version

2025.2.1

### Home Assistant Operating System Version

14.2

davidrapan, 2025-02-13:
Enable debug logging, restart your HA, let it run for few minutes after restart and then attach the log file here.

mamo-nick, 2025-02-13:
10min log file contains 3k rows.

How can i give it to you?

davidrapan, 2025-02-13:
I can see in the profile that sensors about whole module are not controlled by Number of Battery packs... only cell sensors are.

So do you want me to change it?

mamo-nick, 2025-02-13:
Yes. It would be great.

davidrapan, 2025-02-13:
[refactor: Module sensor reading controllable by Number of Batteries f…](https://github.com/davidrapan/ha-solarman/commit/6b13d793dc574cd4ca071b189a1689d03edb0b13)

closed 2025-02-13 by davidrapan