davidrapan/ha-solarman #417: Update solis_hybrid.yaml to include Storage Control mode/Backup combo

midlakelands, 2025-02-13:
### Is your feature request related to a problem?

Under the Storage Control mode it is possible to have a Battery backup (00x0031) to keep some reserve capacity as well has having a mode such as Self Use on. I have this on my inverter with Self Use and my corresponding key value is 00x0033. I think it would be a good idea to add it to the two registers 43011 and 33121

### Describe the solution you'd like

I suggest adding 

```
          - key: 0x0033
            value: "Backup & Self Use"
```
as I think this would be the most popular combination (might be others but I'm not using them)

### Describe alternatives you've considered

alternative is good old custom.yaml i guess

davidrapan, 2025-02-13:
Hi @midlakelands, sure I can but it rather looks like that **0x0033** is turning on bit which is named as **Optimized Revenue**.

midlakelands, 2025-02-13:
Yeah in the documentation you have linked it's called Optimized Revenue but on the lcd screen on the inverter it calls it self-use (the manual description however sounds a lot like optimized revenue).I'd be happy changing the name to Backup & Optimized Revenue as whatever it's called that's the mode I'll be going with

davidrapan, 2025-02-13:
[feat: Add Optimized Revenue & Backup Storage Control Mode for Solis Hybrid](https://github.com/davidrapan/ha-solarman/commit/e435e380a69ae9a81834ff4b9d52ef04de34469d)

closed 2025-02-13 by davidrapan