davidrapan/ha-solarman #418: Errors in log since latest upate

bert1111, 2025-02-14:
### Description

I just updated to the latest version of the integration and I see these errors in the logs.
When I open the integration page I see connect and deconnected al the time every few minutes.

### Reproduction Steps

Install latest version.

### Log

Deze fout is ontstaan door een aangepaste integratie.

Logger: pymodbus.logging
Bron: custom_components/solarman/pysolarman/pysolarman.py:284
integratie: Solarman (documentatie, problemen)
Eerst voorgekomen: 11:24:22 (4 gebeurtenissen)
Laatst gelogd: 13:27:32

Exception response 137 / 0
Exception response 235 / 0
Exception response 240 / 0
Exception response 224 / 0

[home-assistant_solarman_2025-02-14T12-54-22.490Z.log](https://github.com/user-attachments/files/18799417/home-assistant_solarman_2025-02-14T12-54-22.490Z.log)

### Version

25.02.14

### Home Assistant Core Version

2025.2.3

### Home Assistant Operating System Version

14.2

davidrapan, 2025-02-14:
Hi @bert1111, unfortunately I don't see mentioned errors in provided log.

bert1111, 2025-02-14:
At the top of the log in between the frontend errors. 
That's all I see about those errors and then also this ```
2025-02-14 13:53:50.064 DEBUG (MainThread) [pymodbus.logging] Frame check failed, possible garbage after frame, testing..```

[home-assistant_solarman_2025-02-14T12-54-22.490Z.log](https://github.com/user-attachments/files/18801249/home-assistant_solarman_2025-02-14T12-54-22.490Z.log)

```2025-02-14 11:10:09.291 WARNING (MainThread) [homeassistant.components.websocket_api.http.connection] [281472683256864] from 192.168.0.30 (Mozilla/5.0 (Windows NT 10.0; Win64; x64; rv:135.0) Gecko/20100101 Firefox/135.0): Disconnected: Received close message during auth phase
2025-02-14 11:10:45.371 ERROR (MainThread) [frontend.js.modern.202502100] Failed to format translation for key 'ui.panel.config.tips.join' in language 'nl'. Error: The intl string context variable "mastodon" was not provided to the string "Word lid van de community op onze {forums}, {mastodon}, {bluesky}, {twitter}, {discord}, {blog} of {newsletter}"
2025-02-14 11:24:22.509 ERROR (MainThread) [pymodbus.logging] Exception response 137 / 0
2025-02-14 11:43:49.741 ERROR (MainThread) [frontend.js.modern.202502100] Failed to format translation for key 'ui.panel.config.tips.join' in language 'nl'. Error: The intl string context variable "mastodon" was not provided to the string "Word lid van de community op onze {forums}, {mastodon}, {bluesky}, {twitter}, {discord}, {blog} of {newsletter}"
2025-02-14 11:44:41.484 ERROR (MainThread) [frontend.js.modern.202502100] Failed to format translation for key 'ui.panel.config.tips.join' in language 'nl'. Error: The intl string context variable "mastodon" was not provided to the string "Word lid van de community op onze {forums}, {mastodon}, {bluesky}, {twitter}, {discord}, {blog} of {newsletter}"
2025-02-14 11:48:13.471 ERROR (MainThread) [frontend.js.modern.202502100] Failed to format translation for key 'ui.panel.config.tips.join' in language 'nl'. Error: The intl string context variable "mastodon" was not provided to the string "Word lid van de community op onze {forums}, {mastodon}, {bluesky}, {twitter}, {discord}, {blog} of {newsletter}"
2025-02-14 12:05:22.092 ERROR (MainThread) [pymodbus.logging] Exception response 235 / 0
2025-02-14 12:22:28.482 ERROR (MainThread) [pymodbus.logging] Exception response 240 / 0
2025-02-14 12:57:18.472 ERROR (MainThread) [frontend.js.modern.202502100] Failed to format translation for key 'ui.panel.config.tips.join' in language 'nl'. Error: The intl string context variable "mastodon" was not provided to the string "Word lid van de community op onze {forums}, {mastodon}, {bluesky}, {twitter}, {discord}, {blog} of {newsletter}"
2025-02-14 12:57:41.599 ERROR (MainThread) [frontend.js.modern.202502100] Failed to format translation for key 'ui.panel.config.tips.join' in language 'nl'. Error: The intl string context variable "mastodon" was not provided to the string "Word lid van de community op onze {forums}, {mastodon}, {bluesky}, {twitter}, {discord}, {blog} of {newsletter}"
2025-02-14 12:59:23.425 ERROR (MainThread) [frontend.js.modern.202502100] Failed to format translation for key 'ui.panel.config.tips.join' in language 'nl'. Error: The intl string context variable "mastodon" was not provided to the string "Word lid van de community op onze {forums}, {mastodon}, {bluesky}, {twitter}, {discord}, {blog} of {newsletter}"
2025-02-14 13:27:32.428 ERROR (MainThread) [pymodbus.logging] Exception response 224 / 0```

davidrapan, 2025-02-14:
Sure but I need to see those errors in the context of debug logging.

bert1111, 2025-02-14:
I'll try making a new one tomorrow. It's almost dark here and my inverter will be going offline. 
That log file will be huge because the line "Frame check failed, possible garbage after frame, testing.." is written numerous times a second. Lines and lines of it.

bert1111, 2025-02-14:
I managed to make a new log. An event happened at 18:25:49 when I was logging
The log file is too big to upload I have to shrink it in size.
```2025-02-14 18:18:25.938 DEBUG (MainThread) [custom_components.solarman.pysolarman.pysolarman] [xxxxxxxxxxxx] RECD: a5 2d 00 10 15 96 0e 9c a9 f1 8b 02 01 15 7e 89 04 72 05 00 00 cb fc 25 63 01 04 1a 01 00 00 46 00 00 01 f4 00 c8 02 17 01 cc 07 d0 01 cb 00 00 00 00 12 2a 13 88 10 b3 e6 15
2025-02-14 18:18:25.939 DEBUG (MainThread) [pymodbus.logging] Processing: 0x1 0x4 0x1a 0x1 0x0 0x0 0x46 0x0 0x0 0x1 0xf4 0x0 0xc8 0x2 0x17 0x1 0xcc 0x7 0xd0 0x1 0xcb 0x0 0x0 0x0 0x0 0x12 0x2a 0x13 0x88 0x10 0xb3
2025-02-14 18:18:25.939 DEBUG (MainThread) [pymodbus.logging] decoded PDU function_code(4 sub -1) -> ReadInputRegistersResponse(dev_id=0, transaction_id=0, address=0, count=0, bits=[], registers=[256, 70, 0, 500, 200, 535, 460, 2000, 459, 0, 0, 4650, 5000], status=1) 
2025-02-14 18:18:25.940 DEBUG (MainThread) [pymodbus.logging] Frame advanced, resetting header!!
2025-02-14 18:18:25.940 DEBUG (MainThread) [custom_components.solarman.device] [xxxxxxxxxxxx] Querying 04 ❘ 0x04 ~ 4272 - 4284 ❘ 0x10B0 - 0x10BC #013 succeeded
2025-02-14 18:18:25.943 DEBUG (MainThread) [custom_components.solarman.device] [xxxxxxxxxxxx] Returning 73 new values. [Previous State: Connected (1)]
2025-02-14 18:18:25.943 DEBUG (MainThread) [custom_components.solarman.coordinator] Finished fetching Solarman data in 5.061 seconds (success: True)```

davidrapan, 2025-02-14:
So your inverter is now and then sending back fragmented packets and a lot of rows w/
```
Frame check failed, possible garbage after frame, testing..
```
are generated from the flow which is trying to decode them. When it also prints
```
Exception response
```
just means that it failed miserably.

This very likely happened before (with an older version), but these packets were just immediately dropped. Maybe you could ask your manufacturer for inverter firmware update (if possible) and see if it helps? As this shouldn't be happening (to my knowledge). Maybe @githubDante (Hi there) could say more?

Second option is to remove
```
"loggers": ["pymodbus"],
```
from `manifest.json`. This will detach pymodbus logger and will silence the errors.

bert1111, 2025-02-15:
I can see on the sofar website, there is a new version for the hyd5000es, v4.10. My inverter is on V3.60 but it gave me a lot of grief initially to get it to update, so I'm a bit hesitant, trying to solve one problem introducing another. I never found SOFAR support to be really good.

bert1111, 2025-02-15:
Is this related?
```Logger: homeassistant
Bron: components/sensor/__init__.py:614
Eerst voorgekomen: 14:18:32 (1 gebeurtenissen)
Laatst gelogd: 14:18:32

Error doing job: Task exception was never retrieved (None)
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/update_coordinator.py", line 268, in _handle_refresh_interval
    await self._async_refresh(log_failures=True, scheduled=True)
  File "/usr/src/homeassistant/homeassistant/helpers/update_coordinator.py", line 479, in _async_refresh
    self.async_update_listeners()
    ~~~~~~~~~~~~~~~~~~~~~~~~~~~^^
  File "/usr/src/homeassistant/homeassistant/helpers/update_coordinator.py", line 178, in async_update_listeners
    update_callback()
    ~~~~~~~~~~~~~~~^^
  File "/config/custom_components/solarman/entity.py", line 78, in _handle_coordinator_update
    self.async_write_ha_state()
    ~~~~~~~~~~~~~~~~~~~~~~~~~^^
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 1023, in async_write_ha_state
    self._async_write_ha_state()
    ~~~~~~~~~~~~~~~~~~~~~~~~~~^^
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 1148, in _async_write_ha_state
    self.__async_calculate_state()
    ~~~~~~~~~~~~~~~~~~~~~~~~~~~~^^
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 1085, in __async_calculate_state
    state = self._stringify_state(available)
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 1029, in _stringify_state
    if (state := self.state) is None:
                 ^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/sensor/__init__.py", line 614, in state
    raise ValueError(
    ...<2 lines>...
    )
ValueError: Sensor sensor.solarman_device_state provides state value '', which is not in the list of options provided```

bert1111, 2025-02-16:
Hi @davidrapan,

I just updated the inverter firmware a few minutes ago. And it worked second time. I'll keep you posted on what it does today.

bert1111, 2025-02-19:
also started seeing this since yesterday.
```Deze fout is ontstaan door een aangepaste integratie.

Logger: custom_components.solarman.coordinator
Bron: helpers/update_coordinator.py:412
integratie: Solarman (documentatie, problemen)
Eerst voorgekomen: 18 februari 2025 om 13:57:29 (100 gebeurtenissen)
Laatst gelogd: 07:23:42

Error fetching Solarman data: Invalid modbus response received
Error fetching Solarman data: Modbus exception response: 0x05
Error fetching Solarman data: Incorrect response w/ function code 1 instead of 3 received
Error fetching Solarman data: Modbus exception response: 0x00```

githubDante, 2025-02-19:
> So your inverter is now and then sending back fragmented packets and a lot of rows w/

These packets do not appear to be fragmented.  Some of them are too short to be fragmented, also all of them are properly encapsulated V5 frames. I have no idea why the inverter is sending them though.

> This very likely happened before (with an older version), but these packets were just immediately dropped. Maybe you could ask your manufacturer for inverter firmware update (if possible) and see if [...]

The V5 encapsulation, serial number, sequence numbers and even the V5 checksum are correct, so such frame will pass all checks imposed in pysolarman and will not be dropped silently. It will fail with an invalid CRC  or some modbus error.

davidrapan, 2025-03-02:
> These packets do not appear to be fragmented.

If you look at those incomplete modbus responses within V5 frames and "merge" them you will get full modbus response...

githubDante, 2025-03-02:
If you're talking about the chat e.g after `2025-02-16 11:25:22.666`  probably you are right, but there are other cases like the requests around `2025-02-16 11:24:47.931`  where the logger sends very strange responses.  Something with the firmware of the logger is not right, that's for sure and is not related to the ha-solarman version.

davidrapan, 2025-03-02:
Yup, the responses are not consistent at all.

In some other cases (e.g. #412) inverter/logger even replies w/ different memory blocks (e.g. request is for function code 3 and reply is from function code 4).

@bert1111, what firmware version of the logger this is? Could you also try to downgrade **ha-solarman** to version from last year and record some debug log? I really do wonder how the communication looked like before.

bert1111, 2025-03-02:
My inverter is on V3.6. The stick logger is on LSw3_01_2701_ss_00_01.06.00.06.  I'll try to downgrade tomorrow.

bert1111, 2025-03-08:
Hi, @davidrapan. Do you need more logs from me? Or something I should try?

bert1111, 2025-03-14:
I just updated to the latest version. It still disconnects regularly but it seems the log isn't flooded with the error anymore. 
Now it says
Deze fout is ontstaan door een aangepaste integratie.

Logger: custom_components.solarman.coordinator
Bron: helpers/update_coordinator.py:412
integratie: Solarman (documentatie, problemen)
Eerst voorgekomen: 12:22:49 (2 gebeurtenissen)
Laatst gelogd: 12:24:20

Error fetching Solarman data: Fragmented modbus response received
and this
Deze fout is ontstaan door een aangepaste integratie.

Logger: custom_components.solarman.coordinator
Bron: helpers/update_coordinator.py:386
integratie: Solarman (documentatie, problemen)
Eerst voorgekomen: 12:21:35 (1 gebeurtenissen)
Laatst gelogd: 12:21:35

Timeout fetching Solarman data
If you want me to do something please let me know, I'm happy to help, but for me if it stays like this, I can live with it.

davidrapan, 2025-03-14:
Hi, can you attach the debug log again? Thank you.

bert1111, 2025-03-14:
[home-assistant_solarman_2025-03-14T15-46-03.467Z.log](https://github.com/user-attachments/files/19250382/home-assistant_solarman_2025-03-14T15-46-03.467Z.log)

davidrapan, 2025-03-14:
Try to put into the profile's `default` section `max_size: 50` so it looks like this:
```
default:
  update_interval: 5
  code: 0x03
  max_size: 50
  digits: 6
```

bert1111, 2025-03-14:
I just did that. I'll make a new log tomorrow.

bert1111, 2025-03-15:
[home-assistant_solarman_2025-03-15T14-35-08.959Z.log.zip](https://github.com/user-attachments/files/19262917/home-assistant_solarman_2025-03-15T14-35-08.959Z.log.zip)

bert1111, 2025-03-15:
Now where it first disconnected for a few seconds it now disconnects for minutes. That started after I posted previous log.

bert1111, 2025-03-18:
I'm now on version 25.03.16 but the problem persists. I tried a longer update interval but that doesn't help eighter. Are there other things to try? Or is maybe my logger faulty? At the moment it disconnects every 30 to 70 seconds.

bert1111, 2025-03-18:
Well every time it disconnects it causes the sensors to go unavailable. Leaving gaps in my sensor data.
And this causes problems elsewhere with automations and other things. I don't know when this problem started but I didn't used to have this problem. All the graphs always looked normal and I don't recall seeing all my sensors going offline.

[image]

[image]

bert1111, 2025-03-20:
Hi @davidrapan,
Unfortunately I think you are right. The problem is with the logger or the inverter. I think the logger. Do these break often? When I see the disconnect in HA, when I check the stick at that time the lights are flashing and loses connection. It then comes back after a few minutes to drop connection again after that. So I think I need your advice now. Is it somehow repairable? Or is there a replacement that is better than the solarman logger? Or should I go with an LSE3? I made a ticket with Solarman but they're not really helpful up until now. All they said was give it a static IP, which it already has.

kkmaslowski, 2025-03-24:
I have the same problem with LSW3 and Sofar KTLX 3.3

davidrapan, 2025-03-25:
I need debug log w/ this occuring.

bert1111, 2025-03-25:
[home-assistant_solarman_2025-03-25T13-45-15.553Z.log](https://github.com/user-attachments/files/19449040/home-assistant_solarman_2025-03-25T13-45-15.553Z.log)

davidrapan, 2025-05-04:
Hi, can you try to downgrade integration to v24.12.22 and report back if it changes anything?

PR #552 (2025-05-08): Migration to local umodbus
It turned out that switching to the `pymodbus` library was a very good idea in the end.

Fixes #418
Fixes #430
Fixes #442
Fixes #471
Fixes #526
closed 2025-05-08 by davidrapan