davidrapan/ha-solarman #420: New Fault by pysolarman

CrazyUs3r, 2025-02-14:
### Description

Hey David,

Today with the New Skript i become this error some times

Dieser Fehler stammt von einer benutzerdefinierten Integration

Logger: homeassistant
Quelle: custom_components/solarman/pysolarman/pysolarman.py:169
Integration: Solarman (Dokumentation, Probleme)
Erstmals aufgetreten: 17:27:51 (1 Vorkommnisse)
Zuletzt protokolliert: 17:27:51

Error doing job: Task exception was never retrieved (None)
Traceback (most recent call last):
  File "/config/custom_components/solarman/pysolarman/pysolarman.py", line 169, in _conn_keeper
    data = await self.reader.read(1024)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.13/asyncio/streams.py", line 730, in read
    await self._wait_for_data('read')
  File "/usr/local/lib/python3.13/asyncio/streams.py", line 525, in _wait_for_data
    raise RuntimeError(
        f'{func_name}() called while another coroutine is '
        f'already waiting for incoming data')
RuntimeError: read() called while another coroutine is already waiting for incoming data

### Reproduction Steps

No idee for steps

### Log

No log

### Version

25.02.14

### Home Assistant Core Version

2025.2.1 up .4

### Home Assistant Operating System Version

14.2

davidrapan, 2025-02-14:
Could you try to catch this w/ debug logging enabled, please?

Thanks!

CrazyUs3r, 2025-02-15:
I'll do it too, even if it won't be an easy trace.

CrazyUs3r, 2025-02-15:
Her the log 

Dieser Fehler stammt von einer benutzerdefinierten Integration

Logger: homeassistant
Quelle: custom_components/solarman/pysolarman/pysolarman.py:169
Integration: Solarman (Dokumentation, Probleme)
Erstmals aufgetreten: 11:07:03 (2 Vorkommnisse)
Zuletzt protokolliert: 14:07:48

Error doing job: Task exception was never retrieved (None)
Traceback (most recent call last):
  File "/config/custom_components/solarman/pysolarman/pysolarman.py", line 169, in _conn_keeper
    data = await self.reader.read(1024)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.13/asyncio/streams.py", line 730, in read
    await self._wait_for_data('read')
  File "/usr/local/lib/python3.13/asyncio/streams.py", line 525, in _wait_for_data
    raise RuntimeError(
        f'{func_name}() called while another coroutine is '
        f'already waiting for incoming data')
RuntimeError: read() called while another coroutine is already waiting for incoming data

Dieser Fehler stammt von einer benutzerdefinierten Integration

Logger: custom_components.solarman.pysolarman.pysolarman
Quelle: custom_components/solarman/pysolarman/pysolarman.py:150
Integration: Solarman (Dokumentation, Probleme)
Erstmals aufgetreten: 14:07:48 (1 Vorkommnisse)
Zuletzt protokolliert: 14:07:48

[3907049102] Write error: Connection lost
Traceback (most recent call last):
  File "/config/custom_components/solarman/pysolarman/pysolarman.py", line 150, in _write
    await self.writer.drain()
  File "/usr/local/lib/python3.13/asyncio/streams.py", line 386, in drain
    await self._protocol._drain_helper()
  File "/usr/local/lib/python3.13/asyncio/streams.py", line 166, in _drain_helper
    raise ConnectionResetError('Connection lost')
ConnectionResetError: Connection lost

[error_log.log](https://github.com/user-attachments/files/18810796/error_log.log)

davidrapan, 2025-03-02:
Thanks, I look into it soon!

davidrapan, 2025-03-10:
Fix in [refactor: Merge _open_connection and _reconnect methods](https://github.com/davidrapan/ha-solarman/commit/60c455f2b26928a94ed18203f8f218429a1d2dec)

closed 2025-03-10 by davidrapan