davidrapan/ha-solarman #422: Logger switch (Deye) from LSW-3 to LSE-3 - Change of Serial Number missing

mkoegel, 2025-02-15:
### Description

Hi,

I've just switch from the WiFi Logger to a Ethernet Logger.
In the Configure Dialog I can change the IP Address, but not the Serial# which leads to the following error.

`Failed setup, will retry: [279xxx] Device setup failed. [Invalid: Host 192.168.178.188 has serial number 211xxx but is configured with 279xxx.]
`
[image]

Is there a way to change the Serial Nr without having to remove and readd the inverter?

### Reproduction Steps

Switch WIFI with Ethernet Logger
Configure Solarman Inverter with new IP
Error occurs

### Log

This error originated from a custom integration.

Logger: homeassistant
Source: custom_components/solarman/device.py:68
integration: Solarman (documentation, issues)
First occurred: 10:23:54 AM (38 occurrences)
Last logged: 11:09:35 AM

Error doing job: Task exception was never retrieved (None)
Traceback (most recent call last):
  File "/config/custom_components/solarman/device.py", line 52, in load
    self.endpoint = await EndPointProvider(self.config).discover()
                    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/solarman/provider.py", line 97, in discover
    raise vol.Invalid(f"Host {self.host} has serial number {s} but is configured with {self.serial}.")
voluptuous.error.Invalid: Host 192.168.178.188 has serial number 2110424108 but is configured with 2791239269.

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/config/custom_components/solarman/coordinator.py", line 23, in _async_setup
    return await self.device.load()
           ^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/solarman/device.py", line 60, in load
    raise Exception(f"[{self.config.serial}] Device setup failed. [{format_exception(e)}]") from e
Exception: [2791239269] Device setup failed. [Invalid: Host 192.168.178.188 has serial number 2110424108 but is configured with 2791239269.]

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/update_coordinator.py", line 319, in __wrap_async_setup
    await self._async_setup()
  File "/config/custom_components/solarman/coordinator.py", line 27, in _async_setup
    raise UpdateFailed(e) from e
homeassistant.helpers.update_coordinator.UpdateFailed: [2791239269] Device setup failed. [Invalid: Host 192.168.178.188 has serial number 2110424108 but is configured with 2791239269.]

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/config_entries.py", line 637, in __async_setup_with_context
    result = await component.async_setup_entry(hass, self)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/solarman/__init__.py", line 50, in async_setup_entry
    await config_entry.runtime_data.async_config_entry_first_refresh()
  File "/usr/src/homeassistant/homeassistant/helpers/update_coordinator.py", line 314, in async_config_entry_first_refresh
    raise ex
homeassistant.exceptions.ConfigEntryNotReady: [2791239269] Device setup failed. [Invalid: Host 192.168.178.188 has serial number 2110424108 but is configured with 2791239269.]

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/config/custom_components/solarman/coordinator.py", line 44, in async_shutdown
    await self.device.shutdown()
  File "/config/custom_components/solarman/device.py", line 68, in shutdown
    _LOGGER.info(f"[{self.config.serial}] Closing connection to {self.endpoint.address}")
                                                                 ^^^^^^^^^^^^^^^^^^^^^
AttributeError: 'NoneType' object has no attribute 'address'

### Version

25.02.12

### Home Assistant Core Version

2025.2.1

### Home Assistant Operating System Version

N/A Docker Version

davidrapan, 2025-03-11:
Hi @mkoegel, you will have to remove and re-add device back to be able to change serial number in current version of the integration.

davidrapan, 2025-03-16:
[refactor: Add option to reconfigure serial number](https://github.com/davidrapan/ha-solarman/commit/23595425bb8e66c2ff9d9a9184cce134cfe4e282)

closed 2025-03-16 by davidrapan