davidrapan/ha-solarman #424: energy values not possible and uncorrekt

jandeuss, 2025-02-16:
### Description

Hello,

since the update of the last 2024 version, some entities are repeatedly shown as unknown or have insanely high values ​​that the energy panel in home assistant is not working correctly.

This affects Sofar hyd 10K inverters with the lse-3 ethernet logger.

Perhaps a check can be built in to ensure that the generation values ​​cannot be 2,000,000 W in 5 minutes. or a battery charge / discharge of 150,000 W in 5 minutes.

This should be intercepted.

Thank you

### Reproduction Steps

the problem comes many times a day 

### Log

[image]

### Version

25.02.12

### Home Assistant Core Version

2025.2.4

### Home Assistant Operating System Version

14.2

jandeuss, 2025-02-16:
another example....

[image]

davidrapan, 2025-03-02:
Hi @jandeuss, can you please provide debug log? Thank you.

rromrrom, 2025-03-06:
[image]

[image]

Full DEBUG log in attachment

[home-assistant.log](https://github.com/user-attachments/files/19116605/home-assistant.log)

davidrapan, 2025-03-11:
I'm sorry but it's incomplete log (missing integration debug messages for some reason).

davidrapan, 2025-03-11:
It looks like that the issue is coming from reading of fragmented modbus responses which is proly the same issue as in #426

davidrapan, 2025-03-11:
[fix: Add detection of modbus response fragmentation](https://github.com/davidrapan/ha-solarman/commit/714dbc66c69a343471203267f8a8c64871f51a65)

closed 2025-03-11 by davidrapan

davidrapan, 2025-03-14:
Hi @rromrrom, the log file is missing messages from the integration and only contains debug messages from pymodbus. I need full log to be able to determine what's wrong....

rromrrom, 2025-03-14:
what is logs from the integration and where should I find/enable those?

rromrrom, 2025-03-15:
@davidrapan I will play with a logging settings. Please tell me what particular log example should I see to understand it shows what you are looking for? Thanks.

rromrrom, 2025-03-15:
https://drive.google.com/file/d/1e0jhwl-cqPLx_mXGS2QPORfKNwIcMLGZ/view?usp=sharing

Now log has some more info, please take a look. Thanks.

rromrrom, 2025-03-15:
done:  https://drive.google.com/file/d/16oDX8naxnwl8b0AF_UltQFx0wyT5gFNc/view?usp=sharing

davidrapan, 2025-03-16:
Are you sure you are running the latest version of the integration? This has been fixed in [v25.03.12_1](https://github.com/davidrapan/ha-solarman/releases/tag/v25.03.12_1)

rromrrom, 2025-03-16:
[image]

Yes, I'm sure

davidrapan, 2025-03-16:
No you are not this is [v25.03.12](https://github.com/davidrapan/ha-solarman/releases/tag/v25.03.12) and you need [v25.03.12_1](https://github.com/davidrapan/ha-solarman/releases/tag/v25.03.12_1)

rromrrom, 2025-03-16:
Hm, I just manually re-downloaded latest version. And it is updated to 25.03.12_1 , not sure why it is required manual action.  Anyway, thanks for the help, appreciate that. Will check how it works more.

rromrrom, 2025-03-17:
@davidrapan  25.03.12_1 works in general, but original issue with weird spikes on chart doesn't solved. Periodically I still see some huge numbers.

rromrrom, 2025-03-17:
[image]

rromrrom, 2025-03-17:
[image]

davidrapan, 2025-03-17:
It seems that the problem is not only related to this integration, but others have similar problems with corrupted data... Try incorporating the changes from [fix: Add PV Power validation according to max rated power for Deye](https://github.com/davidrapan/ha-solarman/commit/192857150c17d7f03390a35eba8ba92b6ef3fadd)

rromrrom, 2025-03-17:
I will try that. Thanks.
One more note for the history: I have 2 inverters, same model. One is branded as Deye, another one as TAB, but technically they are same device from same manufacturer. The both have same firmware version.
- First one works without any issues. No weird numbers on charts or other issues.
- Second one - sometimes has connection issues and it is only one who has issues with huge numbers. Integration itself looks fully functional and I see no other issues except than statistics distorsion on energy chart which is coming from one device only.

as-cmi, 2025-03-17:
@rromrrom are your inverters connected on the same modbus-rtu line? 
Check this, maybe it can help: #423 
(I have not yet updated the inverter firmware.. how to do it?)

rromrrom, 2025-03-17:
@as-cmi not sure about how to answer on your question properly. My inverters doesn't use anything like waveshare. They are both connected through wifi/ethernet loggers from deye.

davidrapan, 2025-03-17:
@rromrrom, you can actually connect multiple modbus slave devices even to the stick logger (that's what @as-cmi was referring to), but I'm sure your inverters each have their own logger. 😉

rromrrom, 2025-03-17:
correct, I have 2 separate loggers

rromrrom, 2025-03-18:
Well, I have 25.03.16 and original issue is still happens.

davidrapan, 2025-03-18:
I'll need new debug log. 😉

rromrrom, 2025-03-18:
https://drive.google.com/file/d/1tvISQ8Ligq18QXvnbu6bj5UUsFkqcrPG/view?usp=sharing

davidrapan, 2025-03-18:
And around what ~time any issues occurred?

rromrrom, 2025-03-18:
[image]
UTC+2

rromrrom, 2025-03-18:
[image]

rromrrom, 2025-03-18:
[image]

davidrapan, 2025-03-18:
I can see the issue.

davidrapan, 2025-03-18:
[fix: Add transaction identifier validation for ethernet loggers](https://github.com/davidrapan/ha-solarman/commit/6e89c84a65877aaa213953cd9f01b74398e2ab79)

rromrrom, 2025-03-18:
thanks! testing...

as-cmi, 2025-03-20:
[image]

It seems the problem is not yet solved.. the only version working fine for me was the 24.12.22.

rromrrom, 2025-03-21:
My setup shows zero spikes since upgrade to 25.03.18. 
The only thing is a bit weird I have 2 entities in unkonw state for this invertor, while they are not unknown for another one. But I'm not sure it it is related to this particulat topic.

[image]

rromrrom, 2025-03-21:
@davidrapan Thanks for the fix. Now integration way more better works for me.

davidrapan, 2025-03-21:
Total sensors have validation set to numbers `> 0.1` so whenever is the value lower it rather show Unknown. It's a failsafe because some devices when they restart cycle all sensors trough zeroes which would also generate spikes in the HA. 😉

as-cmi, 2025-03-26:
@coserotondo I have 25.03.18 and I'm experiencing spikes every now and then.. With 24.12.22 worked like a charm.
Others say it's ok, maybe I'm doing something wrong.. I have a wired setup with modbus-rtu to tcp converter.
Now I'm updating the inverter but i don't think it'll make difference.