davidrapan/ha-solarman #426: Upgrade from 24.12.22 to v25.02.14 - Sensors inconsistent values.

coserotondo, 2025-02-16:
### Description

I updated from v24.12.22 (very stable in my case - but dhcp deprecation in latest HA) to v25.02.14 two days ago. Apparently all fine (on top of the logs below) but data seemed to be reliable. Today, looking at Energy Dashboard, I found very “huge” values in solar production, energy imported / exported… and so I realized the those sensors had some 0, unavailable or spikes and so history added the values after 0 (I think it has been interpreted as reset of `total_increasing` sensors. I reverted back to v24.12.22 and did manual adjustments to history, trying lo align with real values.
Using custom inverter definition (based on Sofar Wi-Fi kit).

### Reproduction Steps

Upgrade from 24.12.22 to v25.02.14.

### Log

```
2025-02-16 00:06:15.813 ERROR (MainThread) [homeassistant] Error doing job: Task exception was never retrieved (None)
Traceback (most recent call last):
  File "/config/custom_components/solarman/pysolarman/pysolarman.py", line 169, in _conn_keeper
    data = await self.reader.read(1024)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.13/asyncio/streams.py", line 730, in read
    await self._wait_for_data('read')
  File "/usr/local/lib/python3.13/asyncio/streams.py", line 525, in _wait_for_data
    raise RuntimeError(
        f'{func_name}() called while another coroutine is '
        f'already waiting for incoming data')
RuntimeError: read() called while another coroutine is already waiting for incoming data
2025-02-16 03:06:14.473 ERROR (MainThread) [custom_components.solarman.coordinator] Timeout fetching FV data
```

```
2025-02-16 15:25:28.656 ERROR (MainThread) [homeassistant] Error doing job: Task exception was never retrieved (None)
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/update_coordinator.py", line 268, in _handle_refresh_interval
    await self._async_refresh(log_failures=True, scheduled=True)
  File "/usr/src/homeassistant/homeassistant/helpers/update_coordinator.py", line 479, in _async_refresh
    self.async_update_listeners()
    ~~~~~~~~~~~~~~~~~~~~~~~~~~~^^
  File "/usr/src/homeassistant/homeassistant/helpers/update_coordinator.py", line 178, in async_update_listeners
    update_callback()
    ~~~~~~~~~~~~~~~^^
  File "/config/custom_components/solarman/entity.py", line 78, in _handle_coordinator_update
    self.async_write_ha_state()
    ~~~~~~~~~~~~~~~~~~~~~~~~~^^
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 1023, in async_write_ha_state
    self._async_write_ha_state()
    ~~~~~~~~~~~~~~~~~~~~~~~~~~^^
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 1148, in _async_write_ha_state
    self.__async_calculate_state()
    ~~~~~~~~~~~~~~~~~~~~~~~~~~~~^^
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 1085, in __async_calculate_state
    state = self._stringify_state(available)
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 1029, in _stringify_state
    if (state := self.state) is None:
                 ^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/sensor/__init__.py", line 614, in state
    raise ValueError(
    ...<2 lines>...
    )
ValueError: Sensor sensor.fv_inverter_alert_message provides state value '', which is not in the list of options provided
```

### Version

25.02.14

### Home Assistant Core Version

2025.2.4

### Home Assistant Operating System Version

14.2

AlexRux, 2025-02-21:
I agree

[image]

this has ruined all my data. in HomeAssistant you can't remove bad data. Now i'll have forever this spikes on my graphs.

Gavin-A92, 2025-02-21:
Seeing something similar.......recurring daily but at different times so no real consistency there. 
Still on 2025.02.14 for now so let me know if I can help with anything.... 

`Logger: homeassistant
Source: components/sensor/__init__.py:642
First occurred: 21:29:32 (1 occurrences)
Last logged: 21:29:32

Error doing job: Task exception was never retrieved (None)
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/components/sensor/__init__.py", line 635, in state
    numerical_value = int(value)
ValueError: invalid literal for int() with base 10: ''

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/components/sensor/__init__.py", line 638, in state
    numerical_value = float(value)
ValueError: could not convert string to float: ''

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/update_coordinator.py", line 268, in _handle_refresh_interval
    await self._async_refresh(log_failures=True, scheduled=True)
  File "/usr/src/homeassistant/homeassistant/helpers/update_coordinator.py", line 479, in _async_refresh
    self.async_update_listeners()
    ~~~~~~~~~~~~~~~~~~~~~~~~~~~^^
  File "/usr/src/homeassistant/homeassistant/helpers/update_coordinator.py", line 178, in async_update_listeners
    update_callback()
    ~~~~~~~~~~~~~~~^^
  File "/config/custom_components/solarman/entity.py", line 78, in _handle_coordinator_update
    self.async_write_ha_state()
    ~~~~~~~~~~~~~~~~~~~~~~~~~^^
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 1023, in async_write_ha_state
    self._async_write_ha_state()
    ~~~~~~~~~~~~~~~~~~~~~~~~~~^^
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 1148, in _async_write_ha_state
    self.__async_calculate_state()
    ~~~~~~~~~~~~~~~~~~~~~~~~~~~~^^
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 1085, in __async_calculate_state
    state = self._stringify_state(available)
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 1029, in _stringify_state
    if (state := self.state) is None:
                 ^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/sensor/__init__.py", line 642, in state
    raise ValueError(
    ...<5 lines>...
    ) from err
ValueError: Sensor sensor.solarman_total_charge_discharge_cycles has device class 'None', state class 'total_increasing' unit 'None' and suggested precision 'None' thus indicating it has a numeric value; however, it has the non-numeric value: '' (<class 'str'>)`

heebo1974, 2025-02-26:
I also tried that earlier 24.12.22 version, but I got same kind of results with that.
It is weird that energy dashboard shows larger kWh than the sensor itself, which is probably caused by those 0 spikes.

davidrapan, 2025-03-02:
Hi guys,

I can't look into it w/o debug logs, thanks.

heebo1974, 2025-03-03:
Here is my debug log. It seems to contain more than just Solarman-related things. There wasn't much panel production in the morning, but the essential information is probably there, right?

[home-assistant_solarman_2025-03-03T07-48-39.543Z.log](https://github.com/user-attachments/files/19049885/home-assistant_solarman_2025-03-03T07-48-39.543Z.log)

rromrrom, 2025-03-04:
On my side there is about the same issue. I have 2 invertors, but one of them from time to time produces some huge numbers for "total production" and "total battery charge/discharge" . In logs I see onlyt this:

`ValueError: Sensor sensor.deye_02_device_state provides state value '', which is not in the list of options provided`, which may seems unrelated, but my observation is that statistics incorrect records appears after incorrect state report.

INFO (Recorder) [homeassistant.components.sensor.recorder] Detected new cycle for sensor.deye_02_total_load_consumption, value dropped from 16167973.4 to 769.1, triggered by state with last_updated set to 2025-03-04T18:07:02.030308+00:00

INFO (Recorder) [homeassistant.components.sensor.recorder] Detected new cycle for sensor.deye_02_total_load_consumption, value dropped from 327680000.0 to 769.1, triggered by state with last_updated set to 2025-03-04T17:38:55.840210+00:00

I see that 16167973.4 and 327680000.0 on energy charts which doesn't make any sense.

davidrapan, 2025-03-11:
So it looks like that the issue is coming from fragmented modbus responses generated by your inverters.

davidrapan, 2025-03-11:
Duplicate of #424

closed 2025-03-11 by davidrapan

heebo1974, 2025-03-11:
So in other words, the Solarman integration can't do anything about it  ?

heebo1974, 2025-03-11:
> [fix: Add detection of modbus response fragmentation](https://github.com/davidrapan/ha-solarman/commit/d751509c824e040f00be08277303e36516649217)

Ok, sorry to still bother, but is this going to be in some release near future or do I have to just edit that pysolarman.py in the home assistant ?  Thank you very much! Keep up the good work!

davidrapan, 2025-03-11:
> Ok, sorry to still bother, but is this going to be in some release near future...

[v25.03.12](https://github.com/davidrapan/ha-solarman/releases/tag/v25.03.12)

heebo1974, 2025-03-16:
That particular update didn’t help me. I’m still facing the same issue. Should I open a new ticket?

davidrapan, 2025-03-16:
> Should I open a new ticket?

No.

heebo1974, 2025-03-17:
[home-assistant_solarman_2025-03-17T06-57-36.404Z.log](https://github.com/user-attachments/files/19280827/home-assistant_solarman_2025-03-17T06-57-36.404Z.log)

Here is another log if required. 

Hmm.. I think I have some issues with the inverter. It boots/resets approx every half an hour.
It is very weird.  The Solarman integration can't cause that, can it? Doesn't it just read data from the inverter?

[image]

davidrapan, 2025-03-17:
> Hmm.. I think I have some issues with the inverter. It boots/resets approx every half an hour.

I don't think so but I'm now not sure about anything honestly as these issues are really nuts. You could try disable the intgration for couple of hours and you will see.

And btw, you should use **Total \*** for Energy Dashboard as they are more accurate and less error prone because reset to zero every day can mess things up sometimes. 😉

heebo1974, 2025-03-17:
> I just disabled it and waitng for next reset. 

With solarman disabled the inverter still booted. so atleast it wasn't solarman's fault.

davidrapan, 2025-03-17:
🤯 That's unfortunate.. So hopefully they'll replace it for you soon!