davidrapan/ha-solarman #433: Exceptions despite Logger working

Agents-Hacs, 2025-02-22:
### Description

I Upgraded to 25.02.14, now I got those errors once a min despide the logger working.

```
Error doing job: Task exception was never retrieved (None)
Traceback (most recent call last):
  File "/config/custom_components/solarman/pysolarman/pysolarman.py", line 169, in _conn_keeper
    data = await self.reader.read(1024)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.13/asyncio/streams.py", line 730, in read
    await self._wait_for_data('read')
  File "/usr/local/lib/python3.13/asyncio/streams.py", line 525, in _wait_for_data
    raise RuntimeError(
        f'{func_name}() called while another coroutine is '
        f'already waiting for incoming data')
RuntimeError: read() called while another coroutine is already waiting for incoming data

```

### Reproduction Steps

Use version 25.02.14 on a sun800g3-eu-230

### Log

```
Error doing job: Task exception was never retrieved (None)
Traceback (most recent call last):
  File "/config/custom_components/solarman/pysolarman/pysolarman.py", line 169, in _conn_keeper
    data = await self.reader.read(1024)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.13/asyncio/streams.py", line 730, in read
    await self._wait_for_data('read')
  File "/usr/local/lib/python3.13/asyncio/streams.py", line 525, in _wait_for_data
    raise RuntimeError(
        f'{func_name}() called while another coroutine is '
        f'already waiting for incoming data')
RuntimeError: read() called while another coroutine is already waiting for incoming data

```

### Version

25.02.14

### Home Assistant Core Version

2025.2.5

### Home Assistant Operating System Version

Container

sofkaski, 2025-02-23:
I commented this same thing in [release announcement](https://github.com/davidrapan/ha-solarman/discussions/416#discussioncomment-12190206) discussion.

davidrapan, 2025-03-02:
Hi @Agents-Hacs and @sofkaski,

because it's not happening on any of my installations I will need you guys to provide me w/ a debug log w/ this error happening, thanks!

davidrapan, 2025-03-02:
Duplicate of #420

sofkaski, 2025-03-03:
> Hi [@Agents-Hacs](https://github.com/Agents-Hacs) and [@sofkaski](https://github.com/sofkaski),
> 
> because it's not happening on any of my installations I will need you guys to provide me w/ a debug log w/ this error happening, thanks!

I have had debugging log on for 12 hours now and the error has not surfaced yet. Typically it has appeared at least once in this kind of timeframe.   

There has been the usual timeout errors `2025-03-03 00:37:48.891 ERROR (MainThread) [custom_components.solarman.coordinator] Timeout fetching Inverter data`. These errors have been there always probably due to some sort of congestion or resource leak in the logger stick (NoSocketAvailable. See a lengthy discussion about the issue [https://github.com/StephanJoubert/home_assistant_solarman/issues/523#issue-2150720403](here). Solarman recovers from these errors. 

I wonder, if the error I have seen is actually related to these timeout errors too, but now when debug logging is on it somehow changes timings so that the error does not manifest itself like reported in this issue.  I'll attach one log snippet that cover one timeout error case (from timeout to next successful round).

[solarman-ha-timeout-error-log-2025-03.03.log](https://github.com/user-attachments/files/19050944/solarman-ha-timeout-error-log-2025-03.03.log)

Btw. I noticed that there are couple of validation errors, but have not investigated them further.

davidrapan, 2025-03-03:
> I wonder, if the error I have seen is actually related to these timeout errors too.

I believe so because I have no timeouts on my installation (currently over 3 weeks of persistent connection).

> but now when debug logging is on it somehow changes timings so that the error does not manifest itself

This is certainly possible.

But thanks I'll look into those timeouts and will see. 😉

sofkaski, 2025-03-03:
> Btw. I noticed that there are couple of validation errors, but have not investigated them further.

I quess this is not an issue. The validation failure is `Value: 0.0 of total_generation_sensor validation failed: {'min': 0.1}`. I don't have any generator in my system and therefore the value is 0.  The result is that the state value in the DB is 'unknown' which is about right. 

What is the reasoning of the `min` rule here?  Is it  related to  `state_class: "total_increasing"` of the sensor?

davidrapan, 2025-03-04:
Can you try [refactor: Merge _open_connection and _reconnect methods](https://github.com/davidrapan/ha-solarman/commit/93d33dc323255622ea8a3a1be6d6ed9f032a7cee)?

davidrapan, 2025-03-04:
Hi, can you try [refactor: Merge _open_connection and _reconnect methods](https://github.com/davidrapan/ha-solarman/commit/60c455f2b26928a94ed18203f8f218429a1d2dec)?

sofkaski, 2025-03-04:
> Hi, can you try [refactor: Merge _open_connection and _reconnect methods](https://github.com/davidrapan/ha-solarman/commit/60c455f2b26928a94ed18203f8f218429a1d2dec)?

Sure. I'll do it tomorrow. I have one automation case in  a test run over the night. After that is gone I'll install the new version.

sofkaski, 2025-03-05:
I installed the version upto and including the fix (60c455f). Initially it works. Let's see when we get those timeouts.

sofkaski, 2025-03-05:
The fix has been running now 11 hours. No timeouts yet!

sofkaski, 2025-03-06:
Last night 3  timeouts:

> 2025-03-06 02:18:43.731 ERROR (MainThread) [custom_components.solarman.coordinator] Timeout fetching Inverter data
2025-03-06 04:24:31.730 ERROR (MainThread) [custom_components.solarman.coordinator] Timeout fetching Inverter data
2025-03-06 07:00:17.730 ERROR (MainThread) [custom_components.solarman.coordinator] Timeout fetching Inverter data

I am now running solarman with debugging on.

davidrapan, 2025-03-07:
The log looks about right. Integration is unable to open connection to your device.

sofkaski, 2025-03-07:
Yes. Now there is no extra log writings in addition to the timeout error. The integration recovers from the situation.
The timeout seems to happen a couple of times in a day. The fix seems ok.

I have disabled the Deye cloud by removing server IP addresses from the stick configuration, but it did not have any noticeable impact on timeouts.  I can live with this. The only caveat is that in automations etc. you need to be prepared to get 'Unavailable' as a sensor value during the error period. But in anyways it is a good practice to be prepared for that.

davidrapan, 2025-03-07:
I would suggest you to try investigate what's happening exactly during those disconnects. Wifi stability, if you are able to ping the device, etc.

sofkaski, 2025-03-07:
I did investigate that last year when I was using the integration from Stephan. I think the problem starts by the stick closing an existing connection and then going to a state where the connection fails (with no socket available). The stick does not accept new connections  during about 5 minutes. 

The network is perfect and no glitches there.

davidrapan, 2025-03-07:
This shouldn't be happening and most certainly not for so long..

closed 2025-03-10 by davidrapan