davidrapan/ha-solarman #435: Incorrect values ​​for daily power losses

Summer-of69, 2025-02-23:
### Description

I noticed that there must have been an error in the calculation of the daily power loss since December 2024. I have connected an electric heater to the SmartLoad. The values ​​from this sensor are apparently not taken into account in the calculation and therefore the power loss value is too high. Can this be corrected, please?

### Reproduction Steps

As far as I could find out, the registers 0x029B and 0x029F are relevant for this.

### Log

???

### Version

25.02.14

### Home Assistant Core Version

2025.2.5

### Home Assistant Operating System Version

14.2

Summer-of69, 2025-02-25:
Today i found out that the problem in the „Today Losses" is caused by the register 0x0218 = „Today Generation“ and in the „Total Losses" by the registers 0x0219 & 0x021A = „Total Generation“. The values of ‚Today/Total Generation' are already included in the ‚Today/Total Load Consumption'! Additional consideration leads to incorrect/increased loss values.
I hope my explanations are understandable and the adjustments can be taken into account in one of the next updates.
Please check and thanks in advance!

davidrapan, 2025-03-02:
Hi @Summer-of69, yes it's incorrect to do the calculations w/ **Total/Today's Generation**.

Thanks for reporting this issue.

davidrapan, 2025-03-02:
[fix: Total/Today's Losses calculation for Deye](https://github.com/davidrapan/ha-solarman/commit/440e683264ed1c30b13231562faacc5e13a2cbef)

closed 2025-03-10 by davidrapan

davidrapan, 2025-06-04:
Hi @Summer-of69, can you please inform me about the model of your device? Thank you.

Summer-of69, 2025-06-05:
Hi @davidrapan ! My model is a Deye-SUN-10K-SG04LP3-EU.