davidrapan/ha-solarman #438: Fix for Solarman Integration: JSON Serialization & via_device Error in Home Assistant

Zhyvotovskyy, 2025-03-02:
### Description

Hi everyone,

I recently encountered and resolved two issues with the Solarman integration in Home Assistant. I'm sharing the fix in case someone else faces the same problem.

Issue 1: JSON Serialization Error (serial_number as int instead of string)
Error Message:
Unable to serialize entry to JSON. Bad data found at $.serial_number=73378536860751342469078371227860742960(<class 'int'>)
🔍 Cause:

The serial_number was stored as an integer instead of a string in device_registry.
This caused serialization issues and prevented the device from being properly registered.
✅ Fix:
Modify the function build_device_info() in common.py:

Open /config/custom_components/solarman/common.py.
Find the build_device_info() function.
Update these lines:
device_info["serial_number"] = str(serial)  # Ensure serial_number is a string
device_info["identifiers"] = {(DOMAIN, str(serial))}  # Convert identifiers to string
Issue 2: via_device Referencing a Non-Existing Device
Error Message:
Detected that custom integration 'solarman' calls `device_registry.async_get_or_create`
referencing a non existing `via_device` ('solarman', 2944855237)...

🔍 Cause:

via_device in sensor.py was referencing an incorrect device ID (integer instead of string).
✅ Fix:
Modify the SolarmanNestedSensor class in sensor.py:

Open /config/custom_components/solarman/sensor.py.
Find this line (around line 56):
self.coordinator.device.device_info[device_serial_number]["via_device"] = (DOMAIN, self.coordinator.device.config.serial)

Change it to:
self.coordinator.device.device_info[device_serial_number]["via_device"] = (DOMAIN, str(self.coordinator.device.config.serial))

Final Steps
Replace common.py and sensor.py with the modified versions.
Restart Home Assistant.
Verify that the errors are gone and the Solarman device appears correctly in Settings > Devices & Services.
I hope this helps! Let me know if you have any questions. 🚀

[sensor.txt](https://github.com/user-attachments/files/19042775/sensor.txt)
[common.txt](https://github.com/user-attachments/files/19042774/common.txt)

### Reproduction Steps

[sensor.txt](https://github.com/user-attachments/files/19042778/sensor.txt)
[common.txt](https://github.com/user-attachments/files/19042777/common.txt)

### Log

Регистратор: homeassistant.helpers.device_registry
Источник: helpers/device_registry.py:341
Первое сообщение: 12:27:59 (1 сообщений)
Последнее сообщение: 12:27:59

Unable to serialize entry 14968c1243f5c51fc2e62a4beeaa9b2e to JSON. Bad data found at $.serial_number=73378536860751342469078371227860742960(<class 'int'>, $.identifiers[0]=('solarman', 73378536860751342469078371227860742960)(<class 'tuple'>
Давай спробуємо знайти джерело цієї помилки. вона є лише при перезавантаженні а далі вон не зявляється

### Version

25.02.14

### Home Assistant Core Version

2025.2.5

### Home Assistant Operating System Version

14.2

davidrapan, 2025-03-02:
Hi @Zhyvotovskyy, thanks for reporting this back and w/ fix included!

I will look into it as soon as I can!

davidrapan, 2025-03-02:
So i guess [fix: The identifiers and serial_numbers stored as an integers instead of a strings](https://github.com/davidrapan/ha-solarman/commit/e2003be53fa00d9fd53df5095b1398551425bed3) should do!

I see you are utilizing new `via_device` feature which should create sub devices for connected batteries (Deye compatible) can you confirm that it works correctly and maybe share some screenshots? Thank you very much!

Zhyvotovskyy, 2025-03-03:
I have a *Deye SUN-50K-SG01HP3-EU-BM4 50KW, EU* inverter, which supports
two battery connections. Currently, I only have one *66 kWh* battery.

The inverter seems to generate a *phantom serial number* for the second
battery, or maybe your integration doesn't recognize it correctly. Because
of this, the integration *creates sensors for a second battery that doesn’t
exist*. These extra sensors cannot be hidden and sometimes confuse me when
setting up automations.

I solved this issue by *separating the second battery into a standalone
device and deactivating it*, which cleaned up my setup. However, I was
thinking—if needed, it might also be possible to *separate the first
battery as well*. I don’t think it's strictly necessary, as having it
together with the inverter works fine too.

Regardless of any additional configuration, your integration *always
creates phantom sensors for the second battery*. I didn’t dig too deep into
the code, just found a temporary solution that works for me. I’m not trying
to rewrite your integration, just wanted to share my experience and maybe
help improve it a bit.

Thanks for your work on this integration!

Best regards,

davidrapan, 2025-03-03:
Do these new sensors have any meaningful values?

Zhyvotovskyy, 2025-03-03:
The sensors for the second battery seem to display random values. For
example, "Battery 1 Max Cell Voltage" is 0.014 V, which is clearly
incorrect, and other values also don't make sense. Additionally, some
parameters are marked as "Unknown." It looks like the data is either
corrupted or not being read properly.

davidrapan, 2025-03-03:
You can disable reading of those sensors using device configuration and set **Number of Battery packs** to `0` under **Additional options**.

Zhyvotovskyy, 2025-03-04:
By default, I didn’t change the *Number of Battery Packs* setting and
didn’t notice that it was set to *-1* instead of *0*. After explicitly
setting it to *0*, the second battery was no longer detected, and the error
disappeared.

It looks like the default value was causing the issue.

Best regards,

davidrapan, 2025-03-04:
Can you show me readings using Developer tools >> ACTIONS >> Solarman: Read Holding Registers (Modbus Function Code 3) of **Address**: 10000 w/ **Count**: 100?

Maybe I will be able to came up w/ some detection.

closed 2025-03-10 by davidrapan