davidrapan/ha-solarman #440: Integration unavailable when there is no connection with Inverter (at night)

dcigir, 2025-03-03:
Hi David, I just switched to your integration and as I can see it's working great in terms of connection and reading the data. What I would like to know, is it possible that Integration keep last known data at time when there is no Inverter (after the sun is gone)?

Thanks

Denis

davidrapan, 2025-03-04:
Hi @dcigir, simple answer is no.

And it's because there is no way to differentiate between disconnected state and offline state when there is no sunlight.

dcigir, 2025-03-04:
Thanks, I was afraid of that :)

davidrapan, 2025-03-04:
You could do small change to the code to work around that but I personally wouldn't do that as you would risk messing up your data within HA.

I know that it's not pleasant to see dashboards messed up during nights (w/o any meassures in place to counter that behavior). But this problems needs to be solved from HA's side so I'd rather suggest you to try pressure HA's developers to finally add some mechanism for Unavailable and Unknown states into frontend part of HA. 😉

dcigir, 2025-03-04:
I understand, but do you know how was this been solved with Stephan Joubert integration? As I remember, there was everything ok.